Louisvdw/dbus-serialbattery #159: CCCM dosn‘t work properly

testpaul999, 2022-07-27:
**Describe the bug**
Under parameter I can see 6.25A current
But at the main page of Venus OS I see only -4A current with -109W 
**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...'
SerialBattery
3. Click on '....'
Parameter
4. Scroll down to '....'
see the expected current
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

[image]

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi3b
 - Firmware Version 2.90 - 14

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JKBMS
 - Cells: 8
 - Interface: USB-RS485

**Additional context**
Add any other context about the problem here.

Knedox, 2022-07-27:
Serialbattery is just setting the max limit. 
This works on my side. (For example max 5A at 99%)
If its less than this setting, ask your inverter to pull more current

Louisvdw, 2022-07-28:
Knedox is correct. This does look like it is working correct.
If the Discharge Current Limit(DCL) is 6.25A it means when you see a discharge (the negative A and W value) then that current should be less than 6.25A, which it is as 4.3A. 
The DCL value will change if your SOC goes low. Between 10-20% this will be 1/4 of your normal value which is where the 6.25A value is coming from.
https://github.com/Louisvdw/dbus-serialbattery/wiki/Features#charge-current-control-management

testpaul999, 2022-07-28:
@Louisvdw 
But why we have the delta of 2A? Between the current 6.25A and the real 4.3A?
Regards

Louisvdw, 2022-07-28:
Hi @testpaul999 
You have two different things here. The one is a limit where we say the Maximum you should be using it 6.25A
The other is the actual usage that is being used by your system. If your inverter is only drawing 4.3A that is because that is what is being used. If you switch on more lights for instance that will increase. If we said the system must draw 6.25A all the time while only 4.3A is used, then the 1.95A difference will only be wasted as there is nothing using that power and your battery will be depleted faster. 
You want the system to only supply what is needed and nothing more.

testpaul999, 2022-07-28:
Hi,

but look at this 
[image]

The need for more power is there.

Louisvdw, 2022-07-28:
OK , so if I understand what you want to say in pictures is that you expect that the battery delivers 6.5A as the limit is 6.5A, but the picture only show usage from battery of 4.3A with the rest coming from the grid. Is that what you are trying to say?

So in a perfect situation that would be the case they are the same always, but 1) the inverter can only slowly ramp up/down the power usage and 2) the measurements might not be the same acuracy everywhere. 
That is not something the driver can do anything about. The driver only read the values that your BMS supply. The Victron system then has to dicide how to utilise that.
And if the BMS values is a bit off then it might need a bit of a calibration. (that does not seem to be a problem as the values match within a few Watts)

testpaul999, 2022-07-28:
> OK , so if I understand what you want to say in pictures is that you expect that the battery delivers 6.5A as the limit is 6.5A, but the picture only show usage from battery of 4.3A with the rest coming from the grid. Is that what you are trying to say?

:-) sure that’s what I mean.
> So in a perfect situation that would be the case they are the same always, but 1) the inverter can only slowly ramp up/down the power usage and 2) the measurements might not be the same acuracy [...]

hmmm, ok I’m not happy with that but thx for explanation.
1) the inverter can only slowly ramp up/down the power usage
not a problem with watts/s this will shown the whole time until the SoC under 20%

closed 2022-07-28 by testpaul999