Louisvdw/dbus-serialbattery #160: Daly BMS cell voltage is not requested

SamuelBrucksch, 2022-07-28:
Just found it in the code:
https://github.com/Louisvdw/dbus-serialbattery/blob/206a9d22419b8735bb93038a6486cae0a24e427a/etc/dbus-serialbattery/daly.py#L66-L68

Why is the cell voltage disabled for daly? Can i just enable it or is it disabled for some reason?

SamuelBrucksch, 2022-07-28:
I enabled it myself. It works, but not all cells are shown.

Just did some logging and on a 16S system i get this log:
```
@4000000062e2c32233f06cc4 INFO:SerialBattery:cell: 0, startPos: 4, frame: 0
@4000000062e2c32233fd49bc INFO:SerialBattery:cell: 3, startPos: 16, frame: 1
@4000000062e2c322340982a4 INFO:SerialBattery:cell: 6, startPos: 28, frame: 2
```
And it only shows voltages for cells 1-9, 10-16 are not read...

SamuelBrucksch, 2022-07-28:
Ok i found the issue. This is a rather hacky and improper solution i think, so i leave making it more beautiful to you

The problem was, that you can't rely on the correct order of incoming bytes:
```
\xa5\x01\x95\x08 \x01 \r\x16\r\x14\r\r`\x02
\xa5\x01\x95\x08 \x02 \r\x18\r\x1c\r"`"
\xa5\x01\x95\x08 \x03 \r\x18\r\x1a\r\r`\x0c
\xa5\x01\x95\x08 \x04 \r\x0c\r\r\r\x10`\xf7
\xa5\x01\x95\x08 \x05 \r\x0c\r\x0b\r\x0c`\xf2
\xa5\x01\x95\x08 \x06 \r\x0c\r\x0b\r\x0c`\xf3
```

So in this case the 3rd iteration had 0x08 as frame, as the previous one was one longer, however the correct frame would have been 0x03. So the better solution here would be to rather explicitly check for `\xa5\x01\x95\x08` as frame start and then call the unpack function.

My logic now looks like this:
```
    def read_cells_volts(self, ser):
        if self.cell_count is not None:
            buffer = bytearray(self.cellvolt_buffer)
            buffer[1] = self.command_address[0]   # Always serial 40 or 80
            buffer[2] = self.command_cell_volts[0]

            maxFrame = (int(self.cell_count / 3) + 1)
            lenFixed = (maxFrame * 12)

            cells_volts_data = read_serialport_data(ser, buffer, self.LENGTH_POS, self.LENGTH_CHECK, lenFixed)
            if cells_volts_data is False:
                logger.warning("read_cells_volts")
                return False

            frameCell = [0, 0, 0]
            lowMin = (MIN_CELL_VOLTAGE / 2)
            cellnum = 0
            frame = 0
            while frame >= 0 and frame < maxFrame and cellnum < self.cell_count:
                startPos = ((frame * 12) + 4)
                #logger.debug('cell: ' + str(cellnum) + ', startPos: ' + str(startPos) + ', frame: ' + str(frame))
                if frame > 0 and frame < 16:
                    startPos += 1
                newFrame, frameCell[0], frameCell[1], frameCell[2], reserved = unpack_from('>bhhhb', cells_volts_data, startPos)
                while newFrame != frame + 1:
                  startPos += 1
                  newFrame, frameCell[0], frameCell[1], frameCell[2], reserved = unpack_from('>bhhhb', cells_volts_data, startPos)
                frame = newFrame

                for idx in range(3):
                    if len(self.cells) == cellnum:
                        self.cells.append(Cell(True))
                    self.cells[cellnum].voltage = None if frameCell[idx] < lowMin else (frameCell[idx] / 1000)
                    cellnum += 1

        return True
```
So it's a bit better than yours, as i now check the frame to be actually the next frame, however it is still a bit ugly, as the identifiers before are not checked at all.

It seems to work though on a 16s setup.

Before:
```
@4000000062e2e44d24ed02b4 INFO:SerialBattery:cell: 0, startPos: 4, frame: 0, cell count: 16
@4000000062e2e44d24f61304 INFO:SerialBattery:1: 3350, 2: 3346, 3: 3340
@4000000062e2e44d25017514 INFO:SerialBattery:cell: 3, startPos: 16, frame: 1, cell count: 16
@4000000062e2e44d250d63c4 INFO:SerialBattery:1: 3350, 2: 3355, 3: 3362
@4000000062e2e44d251dbb5c INFO:SerialBattery:cell: 6, startPos: 28, frame: 2, cell count: 16
@4000000062e2e44d252817b4 INFO:SerialBattery:1: 781, 2: 5901, 3: 6669
```

After:
```
@4000000062e2e80222f70eb4 INFO:SerialBattery:cell: 0, startPos: 4, frame: 0, cell count: 16
@4000000062e2e8022300e254 INFO:SerialBattery:frame: 1, 1: 3350, 2: 3346, 3: 3340
@4000000062e2e802230b69a4 INFO:SerialBattery:cell: 3, startPos: 16, frame: 1, cell count: 16
@4000000062e2e8022314df84 INFO:SerialBattery:frame: 2, 1: 3350, 2: 3355, 3: 3361
@4000000062e2e802231ef58c INFO:SerialBattery:cell: 6, startPos: 28, frame: 2, cell count: 16
@4000000062e2e80223285fb4 INFO:SerialBattery:frame: 3, 1: 3351, 2: 3354, 3: 3340
@4000000062e2e80223324eac INFO:SerialBattery:cell: 9, startPos: 40, frame: 3, cell count: 16
@4000000062e2e802233db4a4 INFO:SerialBattery:frame: 4, 1: 3340, 2: 3340, 3: 3344
@4000000062e2e802234891e4 INFO:SerialBattery:cell: 12, startPos: 52, frame: 4, cell count: 16
@4000000062e2e8022352984c INFO:SerialBattery:frame: 5, 1: 3340, 2: 3338, 3: 3339
@4000000062e2e802235cdd34 INFO:SerialBattery:cell: 15, startPos: 64, frame: 5, cell count: 16
@4000000062e2e80223669964 INFO:SerialBattery:frame: 6, 1: 3340, 2: 3338, 3: 3339
```

I'm not that much of a python programmer, so i think you might be quicker to fix this than i am, so my proposal would be something like 

```
if data[idx| == \xa5 and data[idx + 1] == \x01 and data [idx + 2] == \x95 and data[idx + 4] == \x08
  newFrame, frameCell[0], frameCell[1], frameCell[2], reserved = unpack_from('>bhhhb', cells_volts_data, startPos)
  idx += 12 (or whatever the minimum value here is)
else
  idx++
  ...

```

SamuelBrucksch, 2022-07-29:
This also requires a check, if the data that should be unpackes actually exists from the startposition, else an error like this will show:
@4000000062e3810b297ff084 struct.error: unpack_from requires a buffer of at least 87 bytes for unpacking 8 bytes at offset 79 (actual buffer size is 86)

SamuelBrucksch, 2022-07-29:
Ok here finally the right solution in python:
````
            while bufIdx < len(cells_volts_data) - 4: # we at least need 4 bytes to extract the identifiers
                b1, b2, b3, b4 = unpack_from('>BBBB', cells_volts_data, bufIdx)
                logger.info(b1)
                if b1 == 0xA5 and b2 == 0x01 and b3 == 0x95 and b4 == 0x08:
                  frame, frameCell[0], frameCell[1], frameCell[2] = unpack_from('>Bhhh', cells_volts_data, bufIdx + 4)
                  #logger.info('frame: ' + str(frame) + ", 1: " + str(frameCell[0]))
                  for idx in range(3):
                    if len(self.cells) == cellnum:
                        self.cells.append(Cell(True))
                    self.cells[cellnum].voltage = None if frameCell[idx] < lowMin else (frameCell[idx] / 1000)
                    cellnum += 1
                bufIdx += 1
```

Louisvdw, 2022-07-29:
Hi @SamuelBrucksch 
Thanks for looking into the issue and creating a PR. 
Yes, the problem was with the cell values not always being read correctly and then the min/max cell values was a mess. So that was disabled, and as I don't own a Daly BMS myself it was a bit hard to look into to fix.

closed 2022-07-29 by Louisvdw