Louisvdw/dbus-serialbattery #161: Max Battery Current of 50A is dislayed as 25A CCL value

geromd, 2022-07-28:
**Describe the bug**
The Victron remote shell show a fixed max charge current (CCL) of 25A:
[image]

The fixed value of 50A in the script is not considered and chages are not shown in the remote shell:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Open Victron remote Shell
2. Go to folder Serial Batterie
3. Go to folder Parameters
4. Compare CCL value with entry MAX_BATTERY_CURRENT = 50.0 in utils.py

**Expected behavior**
Value of CCL and MAX_BATTERY_CURRENT = 50.0 should be the same

**VenusOS (please complete the following information):**
 - Device type: Multiplus II GX (2629)
 - Firmware Version 481

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: Daly BMS
 - Cells: 16S
 - Interface: USB-RS485

geromd, 2022-07-28:
After writing my report th value in the remote shell was actualized by themselfe to 50A. However, there was never and entered value of 25A before for the MAX_BATTERY_CURRENT in the utils.py

Louisvdw, 2022-07-29:
https://github.com/Louisvdw/dbus-serialbattery/wiki/Features#charge-current-control-management

closed 2022-08-04 by Louisvdw