Louisvdw/dbus-serialbattery #167: Daly BMS via can bus

SamuelBrucksch, 2022-07-30:
Hi,

Some of the Daly BMS support a connection via can bus. That should be considerably faster than the serial connection. The protocol is also open.

So i was wondering if - technically - this could be implemented into this library as well. I could work on it and try it myself as I have such a Daly BMS with can bus, i just want to know if there could be any problems I did not see yet.

Each BMS has its own implementation of the communication, so a new BMS type Daly_CAN could be added that gets the data via can bus, however some logic needs to detect/search for the can bus too, right?

As the cerbo for example already has VE.Bus hardware, do you by chance know, if that could be used directly? Then a RJ45 to Daly cable would be enough and I could start working on this...

SamuelBrucksch, 2022-07-31:
Hm looks like CAN on daly is a gamble as noone really knows if they have can or not. Read from some guys, that only RS485 is available but not can bus, although it is printed on the case and also marked on the cables...

geromd, 2022-07-31:
@SamuelBrucksch  Yes, some Daly BMS also have CAN-Bus. Before I decided for the RS485 way I already connected the CAN Port with RJ45 to the Vicron BMS CAN. Due that communication packages were counted, I know that the CAN-Connection was working, but the Victron did not recognizied the Daly BMS on CAN dur to missing driver.

SamuelBrucksch, 2022-08-01:
How did you know can was working if you could not see anything except counting packages? 

As soon as you connect, Venus OS sends out messages on CAN that fit to the selected CAN profile. Those should be counted on TX.

Did OU also see messages on RX?

Venus OS has some CAN CLI tools like can send, candump that you can use to send messages and see the traffic. When I tried it, i could only see the outgoing messages but could not get any reply from the Daly even when I sent requests to it via cansend.

SamuelBrucksch, 2022-08-01:
Ok can now connect to Daly BMS via canbus. Here a sample for cells voltage:
```
  candump can0
  can0  18950140   [0] 
  can0  18954001   [8]  01 0D 7D 0D 7E 0D 7E 58
  can0  18954001   [8]  02 0D 7E 0D 7F 0D 7E 58
  can0  18954001   [8]  03 0D 7E 0D 7F 0D 7E 58
  can0  18954001   [8]  04 0D 7E 0D 7E 0D 7E 58
  can0  18954001   [8]  05 0D 7E 0D 7F 0D 7F 58
  can0  18954001   [8]  06 0D 7F 0D 7F 0D 7F 58
```

Can needs to be set to 250000, requests look like this:
```
cansend can0 18950140#
```

So the request ID is 18 + Data ID + BMS ID (01) + Response ID (40)

SamuelBrucksch, 2022-08-01:
Looking good, first try with python-can lib:
```
INFO:SerialBattery:Starting dbus-serialbattery
INFO:SerialBattery:dbus-serialbattery v0.12
INFO:SerialBattery:No Port needed
INFO:SerialBattery:Testing DalyCAN
18944001: bytearray(b'\x10\x01\x00\x00\x02\x00\x00\x00')
INFO:SerialBattery:DalyBMS 16 cells
INFO:SerialBattery:Connection established to DalyCAN
INFO:SerialBattery:DeviceInstance = 2
INFO:SerialBattery:com.victronenergy.battery.USB9
INFO:SerialBattery:Battery connected to dbus from /dev/tty/USB9
INFO:SerialBattery:DeviceInstance = 2
INFO:SerialBattery:Changed DeviceInstance = 2
INFO:SerialBattery:read_soc_data
18904001: bytearray(b'\x02\x12\x00\x00t\x9b\x03O')
INFO:SerialBattery:voltage: 530, current: -14.9, soc: 847
```

All data requested is from can bus

SamuelBrucksch, 2022-08-01:
Oh yeah:
[image]

It's fully working now. Needs some cleanup, but will create a PR soon, so we can check if everything is ok. Adoption of Daly to Daly CAN was actually quite easy as the protocol behind is still the same.

geromd, 2022-08-01:
Wow, this would really be great if it can be added to the driver, to work with the CAN-Bus.

SamuelBrucksch, 2022-08-01:
https://github.com/Louisvdw/dbus-serialbattery/pull/169

SamuelBrucksch, 2022-08-02:
Cell voltages from last 6h:
[image]
Seems to be stable and also the values look very nice now. No weird values anymore.

pau1phi11ips, 2022-08-30:
@SamuelBrucksch That looks good! What are you using to generate the cell voltages graph?

SamuelBrucksch, 2022-08-30:
I use openhab which has Grafana integrated for visualization of the values that openhab gets. Requires influxdb additionally to that:
https://community.openhab.org/t/influxdb-grafana-persistence-and-graphing/13761

pau1phi11ips, 2022-08-31:
Thanks, it gave me a head start. Found the Victron Grafana docker and it was super simple to install that and get the stats.

https://github.com/victronenergy/venus-docker-grafana

[image]

RedCross0815, 2022-09-16:
Sorry for my stupididi but i need help. Iam reading and trying for hours now to get the daly bms via can to run. 

I get these meassage via can bus, as in the thread before:
[image]

But how do i get the bms into the device List like this?

[image]

I did all the installation shown in the video and the text, but i wont work...

SamuelBrucksch, 2022-09-18:
You need two terminals to test if the CAN connection works. One for candump and one fro cansnd. Your cansend snippet does not show any Daly messages. Also you need to execute the dbus-serialbattery code manually, as there is no CAN auto detection yet.

mr-manuel, 2023-04-14:
What is the current state of this?