Louisvdw/dbus-serialbattery #171: SoC is falling

dan11hh, 2022-08-02:
**Describe the bug**
I don‘t know if it is a bug, but at two installed systems the SoC shown by serialbattery falls. 
mit works around three weeks perfekt. Voltage was around 3,45V the SoC was 99%. 
now it just shows 90%, at the other (same) system it is at 60%. 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**VenusOS (please complete the following information):**
 - Device type:  Venus OS Raspberry PI at MP 2 5000
 - Firmware Version [e.g. 267]

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JK BMS EVE 280k
 - Cells: 16
 - Interface: USB-RS485
 -

**Additional context**
Add any other context about the problem here.

Louisvdw, 2022-08-03:
This would not be a bug in the driver, but rather that the driver is now showing you what is happening inside your battery. 
The SOC value is what your BMS is publishing. The driver is just publishing that the the Victron system.

This can be influence by the formulas and logic that your BMS use.  It could be that there are some cells not up to voltage that is keeping the SOC value lower. Dig a bit into the min/max cell values and check if you see anything strange. The graphs in the advance section in VRM give a lot of great insight into your battery.

dan11hh, 2022-08-03:
That is my "problem". The Battery Voltages are nearly matching the whole time. The greatest difference which the VRM showed to me is 0,04V while Loading with 65A.

Louisvdw, 2022-08-03:
Are you using large currents or are your system only using small amounts at a time. 
These BMS use a shunt to measure the current flow and while this is one of the best methods to use it can sometimes struggle to measure small amounts of current. So if you have a 100A BMS but are most of the time using say 1A then it could be that your BMS is not reading that correctly. A shunt has to be calibrated for either large currents or small. Or you need 2 to be able to read small and large currents, which increase the cost so no body builds BMS like that. 

When you are seeing the 0.04V cell difference, is that when your battery is at the highest SOC? What are your voltages at that time? Can you post these 3 graphs from your system for the last full day that shows the problem
[image]

dan11hh, 2022-08-03:
You're the man! Thanks for your massiv help. The battery is now up for 40 days. I was last week in holiday so the battery wasn't used that much, if that matters...
I just to make sure that i can add  second 16s pack parallel to the existing one. (same config)
[image]
[image]
[image]
[image]

Today (first day back) i got a max difference from 0,08V, but just while i load with 65A and just a short time. If i disable loading in the bms the cells will be all back nearby.

Louisvdw, 2022-08-04:
Your system is looking good. All the values is as I would expect them

dan11hh, 2022-08-04:
Hmmm thanks again for you answer. i have no idea how to recalibrate the SoC. Did you have any further tipps/ideas?

Louisvdw, 2022-08-04:
What settings do you have for your JKBMS?

dropax, 2022-08-17:
To recalibrate SoC in the JKBMS the following procedure works for me:

- discharge to something below 70% (not shure if there is a distict threshold)
- alter the Ah setting, at this point the BMS guestimates a new SoC (you loose cycle count and cycle Ah, take notes if you want to track these values) 
- it looks like you're "missing" 9%, therefore the new SoC should be at least these 9% above the old value 
- recharge to full, SoC will rise and stay at 99% until "true" full charge is reached

You'll see 100% only for the brief moment when the BMS' Max Cell Voltage is hit and the charge-FET gets disabled.

closed 2023-04-14 by mr-manuel