Louisvdw/dbus-serialbattery #177: Wrong Battery Voltage @ VRM and Remote Console Overview

towatai, 2022-08-08:
I use a Daly 8S 150A BMS and using the newest prerelease 12b2.

The new cell voltages view displays the correct battery voltage. However, in the Remote Console overview as well as in the Victron VRM Portal the voltage is displayed incorrectly. The real voltage is ~27.58V, which is correctly read out in the new sub-item, but the Remote Console overview and the VRM show 27.50V. Attached are a few screenshots.
The whole thing is read out via Daly UART cable and the VenusOS runs on a Raspberry Pi 3 b+ with VenusOS 2.89.

It would be great if the correct voltage can also be displayed in the VRM and the Remote Console overview.

[image]
[image]
[image]

centro-max, 2022-08-08:
I use 3 different Daly 16S BMS with OS 2.89 and the 12b2 release but i do not see the different cell voltage page? What must i do to see the page?
[image]

towatai, 2022-08-08:
> I use 3 different Daly 16S BMS with OS 2.89 and the 12b2 release but i do not see the different cell voltage page? What must i do to see the page? [image]

qml files are installed by running "sh /data/etc/dbus-serialbattery/installqml.sh"
WARNING! - by running the "installqml.sh" is the existing file "/opt/victronenergy/gui/qml/PageBattery.qml" overwritten

https://github.com/Louisvdw/dbus-serialbattery/pull/87

centro-max, 2022-08-08:
[image]

Thanks! 👍

towatai, 2022-08-08:
I have at least managed to get the exact values displayed in the Remote Console. Replace the line in the file "PageBattery.qml":

property VBusItem dcVoltage: VBusItem { bind: service.path("/Dc/0/Voltage") }

by

property VBusItem dcVoltage: VBusItem { bind: service.path("/Voltages/Sum") }

I still fail to get the voltage displayed in the VRM portal.
I also tried to replace "/Dc/0/Voltage" with "/Voltages/Sum" in dbushelper.py. Then on the VRM overview page the voltage is displayed correct but the power display does not work and the Serial Battery menu item in the remote console has completely disappeared ;-)

Louisvdw, 2022-08-10:
The Battery voltage is a measurement over the battery (over the +pole and -pole).
The driver use the battery voltage published by the BMS and while this should be close (as in your case) the cell voltages normally do not perfectly match the battery voltage.

towatai, 2022-08-10:
Thank you very much for your quick answer.
Instead of the battery voltage provided by the BMS, would it be a big effort to pick up the SUM voltage of all cells (which works great in the Remote Control Center) and display it in the VRM? Unfortunately I am a complete newbie in Python and my programming skills are rather limited to one or the other (although already quite complex) Powershell script. I am willing, eager to learn and would also add the appropriate code by hand instead of screaming for a new release.

Louisvdw, 2022-08-10:
Officially the driver will never do that as it has to work the same for all users and also for all BMS types. 
But if you want to make a custom change yourself you can - you will need to change your daly.py to update the self.voltage with the total of the cell voltages. 

However this will be wrong most of the time. 
Why? Well the Daly protocol is not the best which means we have some limits in reading the amount data. To get around that we update the important data each polling cycle (1sec) like the battery voltage, while some not so important data like the cell voltages is only updated every other polling cycle (2 -3 seconds). Your cell voltages will be old, while your battery voltage will have the correct stuff in them.

towatai, 2022-08-11:
Thank you Louis,
have tried something around but just do not come to the associated code. I just had the idea if it would not be possible to use the battery voltage measured by the used Smartsolar as a source for the display of the battery voltage in the VRM Dashboard? Except for minimal deviations, this is accurate. Unfortunately, I have not yet found a way to display the voltage measured by the Smartsolar on the dashboard. You have to switch to the Advanced tab every time.
Call me petty and it's probably a 1st world problem but I'd rather have a display that varies by 0.0x V than one that can only jump up or down in 0.10 increments ;-)

closed 2023-04-14 by mr-manuel