Louisvdw/dbus-serialbattery #187: CVL & CCL & DCL Fixed not matching BMS settings

NunoRgP, 2022-08-21:
**Describe the bug**
SerialBattery/Parameters seem to be oscillating between certain values :

CVL :  55.2 V
CCL : 50 A
DCL : 15A / 30 A

On the JBD BMS settings I am using 50A as maximum charge current and 100 A as discharge current as well as on the utils.py

I selected the SMART Shunt to be used as the battery monitor but it seems the cerboGX is taking the above values on serial battery as the system limits . At the  DCL parameter ( 30A or 15A ) the Inverter throttles down . 

I wand to use the dbus-serialbattery as only a monitoring tool as I currently have 2 batteries in parallel . So I would prefer to use the tool with DVCC off but as soon as I turn it off the inverter gives me the low voltage error and turns on passthrough . 

Is there a way to make these values fixed ? Is this expected behaviour ? 

**Expected behavior**
Have the SerialBattery/Parameters CVL & CCL & DCL as fixed values 

**Screenshots**
[image]
[image]
[image]
[image]
[image]

**VenusOS (please complete the following information):**
 - Device type:  Cerbo GX
 - Firmware Version [e.g. 262]

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JBD

 - Cells: 16
 - Interface: USB-RS485

**Additional context**

Add any other context about the problem here.

NunoRgP, 2022-08-21:
Ok I managed to change the settings and it seems they are fixed now, I had to do the CCM_Enable = False . 
After placing it as False the cerboGX is now using the max battery charge current and discharge current from the utils.py  . 

I am only missing now where is the CVL of 55.2 coming from ?

Louisvdw, 2022-08-22:
Correct. CCCM dynamicly changes your charge current from the SOC value. It does use the battery SOC value not the GX SOC as it is a battery feature.

For the CVL it gets calculated from your cell count. See CVL in [features](https://github.com/Louisvdw/dbus-serialbattery/wiki/Features#features)

closed 2022-09-12 by Louisvdw