Louisvdw/dbus-serialbattery #188: Cannot disable driver on Venus GX

TassieFlash, 2022-08-23:
Installed your driver on a new Venus GX  (Firmware version 2.89)

Decided that I did not like the integration into my existing Victron system.

Running the command...   sh /data/etc/dbus-serialbattery/disabledriver.sh ,to disable the driver fails to work.

I am getting error codes on the Victron mppt 150/60 controller and on the inverter, and cannot resume normal use.

[image]

Louisvdw, 2022-08-23:
Once you installed a BMS/Battery your MPPTs are set by the GX system to be connected to the BMS. When the connection to the battery is lost your MPPT will go into this safety mode and you need to tell your MPPT it is safe to work without the BMS.

See https://www.victronenergy.com/live/mppt-error-codes#err_67_-_bms_connection_lost on how to configure the MPPT back to stand alone mode.

closed 2022-08-23 by Louisvdw

TassieFlash, 2022-08-23:
Hi Louis, I guess there is a similar procedure to rid the Victron RS 48/6000 inverter of its error codes too.

I'll trawl the Victron site for that one. 

Thanks.