Louisvdw/dbus-serialbattery #193: Daly bms sleep mode causing incorrect data in MP2 gx webinterface

LukasBenner, 2022-09-05:
**Describe the bug**
The battery is fully loaded and the daly bms stops charging.
Shortly after it goes into sleep mode and doesn't respond on the uart interface.
In the WebInterface of the MP2 GX and on the VRM Portal wrong values are shown.
e.g the current is displayed with 20 amps but is 0 in real.

The driver log ("/var/log/dbus-serialbattery.ttyUSB0/current") shows "No reply - returning"

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Stop charging so the daly bms goes into sleep mode.
2. Open the log from above.

**Expected behavior**
I'd expect the web interface of the MP2 GX to show the correct data.
I'd expect the software to differentiate between the daly bms being in sleep mode 
and the daly bms not responding.

**VenusOS (please complete the following information):**
 - Device type MP2 GX
 - Firmware Version v2.90

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: Daly
 - Cells: 16
 - Interface: USB-RS485

**Additional context**
Add any other context about the problem here.

LukasBenner, 2022-11-12:
The same behavior can be observed when completely disconnecting the BMS.
This is kind of a critical bug because the soc regulates the parameters for charging and discharging.
Although the Daly BMS will cut the battery when fully charged or discharged, the allowed current for charging and discharging is not updated.

I just connected the BMS to a cerbogx of a friend of mine to test the software. I'd expect my installation to stop charging while the BMS is not connected. Unfortunately it just keeps charging. Went from 61% to 69% percent without any error or something.

mr-manuel, 2023-05-01:
@transistorgit maybe you know, if this is still an issue?

transistorgit, 2023-05-01:
Yes, I think so. There is no means of state handling employed. If the bms stops answering, the last values will be there forever. 
It should be easy to solve with a timeout in refresh_data. If there isn't any new data for 1 minute, set AllowToCharge/Discharge to false. 

How is it done in other bms? Wonder if it shouldn't be a base class function?

mr-manuel, 2023-05-02:
Theoretically there should already be a check, but it seems that it's not working.

https://github.com/Louisvdw/dbus-serialbattery/blob/fce00b998a929fcf7d0b75b2f4b11de4823d366a/etc/dbus-serialbattery/dbushelper.py#L322-L350

transistorgit, 2023-05-03:
just did a real test with my dalys with branch jkbms_ble.
In both cases, unplugging USB or unplugging the serial connector at the BMS, the remote console shows correctly "Not connected" after a minute.
Until 1 minute is over the values stay frozen, that can be optimized.

transistorgit, 2023-05-03:
created pull request https://github.com/mr-manuel/venus-os_dbus-serialbattery/pull/3 
is that right? or should it go into luisvdw's repo?

mr-manuel, 2023-05-04:
It doesn‘t matter in which repo :-) it will get merged in both ways. I test it today with a JKBMS.

mr-manuel, 2023-05-04:
@transistorgit I did some testing and I have this behaviour: After 10 seconds the BMS is shown as `Not connected` and all values are `--`. In this case the charging/discharging is not blocked, since there is no BMS connected. This can be good or bad depending on the cause.

Would it make more sense to keep the driver "connected", but blocking charge/discharge and eventually let the user select/change the behaviour by config file? In this case the driver/system has to be restarted, if a BMS is removed from the device.

What are your thoughts?

transistorgit, 2023-05-04:
yes, I had quite some error handling to do until I had fixed all Type errors. just very probable that I misses one that is now triggered with your BMS. Can you send me the log line with the line no where is crashed?

If all exceptions are handled, the behaviour should be like I wrote. And I had also no problem with redetecting the device then. This is in my view the cleanest solution.

mr-manuel, 2023-05-04:
Ok, in my case somehow `ttyUSB0` gets blocked from the driver. When I disconnect from the USB port `ttyUSB0` and connect it again, before the driver exited, then the devices is recognized as `ttyUSB1`. If I reconnect it again after the driver exited, then it's recognized as `ttyUSB0` again.

I made some more changes in the `jkbms_ble` branch in my fork, that are merged later today with Louisvdw `jkbms_ble` branch:
* Added: Alert is triggered, when BMS communication is lost
* Added: Block charge/discharge when BMS communication is lost. Can be enabled trough the config file

mr-manuel, 2023-05-04:
We wrote nearly at the same time :P I already fixed the other errors, so none left. If the adapter does not change port, all is working as expected. In my eyes we can close this now. What do you think?

closed 2023-05-04 by mr-manuel