Louisvdw/dbus-serialbattery #197: Renogy Smart Lithium iron phosphate with self-heating function cannot be detected

janderholm, 2022-09-07:
**Describe the bug**
Renogy Smart Lithium iron phosphate with self-heating function cannot be detected.

**To Reproduce**
Connect the battery and restart the victron gateway. It is never found.

**Expected behavior**
It is stated that Renogy batteries work.

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi 3
 - Firmware Version: 2.80

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: Renogy Smart Lithium Iron Phosphate 12v 100Ah battery w/ self-heating function.
 - Cells: 4
 - Interface: USB-RS485

**Additional context**
This battery can be connected to successfully using the following module:
https://github.com/rosswarren/renogymodbus
I don't know if this battery is unsupported or if there're bugs in the integration, but I'll try to look into the issue. Perhaps this battery with heating is responds slightly differently.

janderholm, 2022-09-10:
I've managed to get it connected by changing address from 0 -> 247 and disabling the firmware version in the current Renogy integration.

But there are a few readings that are incorrect. Amp hours seem to be multiplied by 1000. How correct is the original implementation?

For example the firmware parsing code is broken because it tries to make a float out of "b'01'.b'31'" caused by an incorrect string conversion, so I don't think anyone have tried to use the integration on a current Venus OS (since python3 at least).

I'd like fix things but can only do that so they work on my batteries. So I'd need someone to test it on the original batteries as well.

Louisvdw, 2022-09-10:
Hi @janderholm 
I created a branch for you https://github.com/Louisvdw/dbus-serialbattery/tree/renogy-fixes
You can check it out and make your changes. When you are done create a Pull Request and I can make a beta build from that that other Renogy users can test and give us feedback.

Louisvdw, 2022-09-21:
It would be great if anyone can give the new beta a test that include changes for this issue ad give us feedback.
https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.12b4

gambleben, 2022-09-21:
I recently forked the code and made some changes to work with Python 3 and the latest release of Venus, but haven't merged those changes back in.  Looks like you beat me to it, but I'll review my code and do some testing to see if there's anything else that should be fixed right away.

I also want to share the documentation for the Renogy modbus that I found: 
[Lithium.Iron.Battery.BMS.Modbus.Protocol.V1.7.zh-CN.en.1.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/9619809/Lithium.Iron.Battery.BMS.Modbus.Protocol.V1.7.zh-CN.en.1.pdf)

When the Renogy bluetooth module is plugged in, it assigns new modbus addresses to any batteries that are plugged in.  I use that, so when I first developed the code, I used the default first address that it assigns.  I'd really like to help get multiple batteries working correctly and help implement the most relevant alarms.