Louisvdw/dbus-serialbattery #198: JK BMS B1A8S20P not found/error

ibraheemhlaiyil, 2022-09-07:
**VenusOS (please complete the following information):**
 - Device type: RPi 3B+
 - Firmware Version: 2.89

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JK B1A8S20P
 - Cells: 4
 - Interface: TTL to USB via Tronix FTDI (only RX/TX pins connected)

1x MPPT and 1x Phoenix also connected to RPi using Victron VE.Direct to USB cables

tail /data/log/dbus-serialbattery.ttyUSB2/current output:

@40000000631917042930b744 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@40000000631917042937a0cc INFO:SerialBattery:Testing Ant
@400000006319170439240f34 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000631917043936a504 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000063191704393d28fc INFO:SerialBattery:Testing Daly
@400000006319170507946c0c ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@400000006319170507a70d94 INFO:SerialBattery:Testing Daly
@40000000631917050fdf4d14 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@40000000631917050ff21d7c INFO:SerialBattery:Testing Jkbms
@40000000631917150d720634 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:855/65281]
@40000000631917150d852c8c INFO:SerialBattery:Testing Renogy
@400000006319171514bcd0a4 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@400000006319171514c45e3c INFO:SerialBattery:Testing Revov
@40000000631917151960066c WARNING:SerialBattery:Modbus Address: 0
@40000000631917151968f394 WARNING:SerialBattery:Modbus Type   : 0
@4000000063191715197065d4 WARNING:SerialBattery:Modbus Command: 0
@40000000631917151977766c WARNING:SerialBattery:Modbus PackLen: 0
@4000000063191715197f1b74 WARNING:SerialBattery:Modbus Packet : [00:00:00:00]
@40000000631917151984ff44 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@40000000631917153766849c INFO:SerialBattery:Testing LltJbd
@40000000631917160674a42c ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@4000000063191716067ce18c INFO:SerialBattery:Testing Ant
@4000000063191716165c3864 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063191716166f37ac ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000631917161675d6fc INFO:SerialBattery:Testing Daly
@40000000631917161f5b178c ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@40000000631917161f6d3ffc INFO:SerialBattery:Testing Daly

When I edit the baud to 9600 in serialbattery.py the output is:

@40000000631916ad163d69ac INFO:SerialBattery:Starting dbus-serialbattery
@40000000631916ad1645417c INFO:SerialBattery:dbus-serialbattery v0.12
@40000000631916ad165dfd84 INFO:SerialBattery:Testing LltJbd
@40000000631916ad205a14e4 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@40000000631916ad20610254 INFO:SerialBattery:Testing Ant
@40000000631916ad30544aa4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000631916ad30665f8c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000631916ad306d2204 INFO:SerialBattery:Testing Daly
@40000000631916ad3a4d396c ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@40000000631916ad3a63da64 INFO:SerialBattery:Testing Daly
@40000000631916ae0834c6d4 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@40000000631916ae0849be04 INFO:SerialBattery:Testing Jkbms
@40000000631916ae0d32befc Traceback (most recent call last):
@40000000631916ae0d32d66c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 115, in <module>
@40000000631916ae0d32eddc     main()
@40000000631916ae0d32f1c4   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 85, in main
@40000000631916ae0d33054c     battery = get_battery_type(port)
@40000000631916ae0d331104   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 64, in get_battery_type
@40000000631916ae0d34f17c     if test.test_connection() is True:
@40000000631916ae0d35011c   File "/opt/victronenergy/dbus-serialbattery/jkbms.py", line 25, in test_connection
@40000000631916ae0d3514a4     return self.read_status_data()
@40000000631916ae0d351c74   File "/opt/victronenergy/dbus-serialbattery/jkbms.py", line 60, in read_status_data
@40000000631916ae0d352ffc     status_data = self.read_serial_data_jkbms(self.command_status)
@40000000631916ae0d3704bc   File "/opt/victronenergy/dbus-serialbattery/jkbms.py", line 149, in read_serial_data_jkbms
@40000000631916ae0d371844     terminal = unpack_from('>L', data[4:])[0]
@40000000631916ae0d3727e4 struct.error: unpack_from requires a buffer of at least 4 bytes for unpacking 4 bytes at offset 0 (actual buffer size is 0)

FYI I have commented out Sinowealth in serialbattery.py as that is selected and installed erroneously.

ibraheemhlaiyil, 2022-09-08:
Tried both v0.12 and v0.12b2

Louisvdw, 2022-09-08:
Hi @ibraheemhlaiyil 
The JKBMS use a fast baud rate so interferrance is more common that with other slower rates.
I do see that you connected "TTL to USB via Tronix FTDI (only RX/TX pins connected)" and I suspect this will be your problem. TTL needs to be grounded, so you need to connect the ground wire as well.

ibraheemhlaiyil, 2022-09-08:
Using a ground wire didn’t work at all, it only started doing anything without a ground wire connected. Will play some more.

ibraheemhlaiyil, 2022-09-08:
OK progress, re-wired and soldered up a stronger shielded and shorter connection.

@400000006319ca98264a353c INFO:SerialBattery:Testing Jkbms
@400000006319ca982d3d4a8c INFO:SerialBattery:Connection established to Jkbms
@400000006319ca983125e6cc INFO:SerialBattery:DeviceInstance = 2
@400000006319ca98312fba6c INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
@400000006319ca9832181e9c INFO:SerialBattery:Battery connected to dbus from /dev/ttyUSB1
@400000006319ca98321ed174 INFO:SerialBattery:=== Settings ===
@400000006319ca9832260d04 INFO:SerialBattery:> Connection voltage 12.62V | current -0.0A | SOC 0%
@400000006319ca98322c815c INFO:SerialBattery:> Cell count 4 | cells populated 4
@400000006319ca983232da5c INFO:SerialBattery:> CCL Charge 50.0A | DCL Discharge 60.0A
@400000006319ca983239623c INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.45V
@400000006319ca98332888e4 INFO:SerialBattery:DeviceInstance = 2
@400000006319ca983331954c INFO:SerialBattery:Changed DeviceInstance = 2
@400000006319ca9a06aa719c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006319ca9a07d9d97c Traceback (most recent call last):
@400000006319ca9a07d9f0ec   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 176, in publish_battery
@400000006319ca9a07da085c     self.publish_dbus()
@400000006319ca9a07da102c   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 254, in publish_dbus
@400000006319ca9a07da23b4     self._dbusservice['/%s/Diff'%pathbase] = self.battery.get_max_cell_voltage() - self.battery.get_min_cell_voltage()
@400000006319ca9a07da3b24 TypeError: unsupported operand type(s) for -: 'NoneType' and 'NoneType'
@400000006319caba001a1d8c INFO:SerialBattery:Starting dbus-serialbattery
@400000006319caba00215534 INFO:SerialBattery:dbus-serialbattery v0.12b2

ibraheemhlaiyil, 2022-09-08:
And now getting this after a reboot, puzzled:

@400000006319cd4d0b5246fc INFO:SerialBattery:Testing Jkbms
@400000006319cd4d1b386434 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006319cd4d1b4d18fc INFO:SerialBattery:Testing Renogy
@400000006319cd4d2b6a1514 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006319cd4d2b80fc5c INFO:SerialBattery:Testing Revov
@400000006319cd4d3b6b5784 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006319cd4d3b817b7c ERROR:SerialBattery:read_serial_data_revov::Serial Data is Bad
@400000006319cd4e1dde04dc ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB1

Louisvdw, 2022-09-08:
As far as I have seen all the JKBMS use the same protocol and the driver was able to talk with it. That is where the connection values come from. It seems to get an error when trying to get the min and max cell voltages. These are either not populated yet or their values have a problem.
It is in the publish cell data section which you can easily disable by changing the setting on line 56 in utils.py to
>BATTERY_CELL_DATA_FORMAT = 0

Louisvdw, 2022-09-08:
Also try and insulate your wires from any interferrance as much as possible. 
Even wrapping the comms wire between the BMS and your GX in foil helps a lot.

ibraheemhlaiyil, 2022-09-08:
Thanks @Louisvdw 

I can see this in the earlier log though:  MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.45V, so must have been able to parse those values correctly.

But now I'm not getting anything after a reboot. I'll try and strengthen the connections as much as I can.

ibraheemhlaiyil, 2022-09-08:
OK I've found the issue - loose connection at the BMS. I think I'm going to need to solder directly to the board removing the pin header. But at least I've got to the bottom of it. Thank you :)

closed 2022-09-08 by ibraheemhlaiyil