Louisvdw/dbus-serialbattery #199: ANT BMS via Bluetooth

equi2k, 2022-09-08:
Hey and thanks for your work!

Is it possible to connect the ANT BMS via Bluetooth?
In my setup without victron i have connected the ANT BMS via Bluetooth to my Raspberry Pi. Nodered reads the data and send it to my MQTT Broker - i think its not a big deal. Please feel free to contact me, if you need the protcoll / script which i use.

regards,
Christoph

TimGFoley, 2022-12-10:
Did you make any progress with this @equi2k ?   Would be great for the newer Renogy batteries where they removed the RS485 ports.

AncoDiara, 2022-12-14:
Hello Louis,
What can we do to help? I'm also very much interested to see your script be extended with bluetooth readout instead of coms cables. I'm an experienced BMS builder and tester would like to share ideas around safety policies when bluetooth connected batteries disappear momentarily. Which happens more likely than the bms's with comms cable. But 99.9999% of the time this should not be a problem as the system is in a 'safe' voltage zone of the battery. Thus system should continue to function and hope for the battery to become visible again. etc. Also I would like to see active balancers readout on Cerbo GX from JKbms for example or other nice active balancers.

mr-manuel, 2023-04-20:
You can check https://github.com/Louisvdw/dbus-serialbattery/discussions/485. A lot of things changed with the last version, so now it should also be easiert to add bluetooth support for other BMS.