Louisvdw/dbus-serialbattery #213: Driver no longer starts

centro-max, 2022-09-19:
**Describe the bug**
I installed Victron version 2.90 and can no longer activate the driver.
Is there a known issue here?
I've installed the driver multiple times with no error messages.

**Screenshots**
/data/log/serial-starter/current:
...
@400000006328a6f70d2acbd4 *** CCGX booted (0) ***
@400000006328a6f927dfe89c *** starting serial-starter ***
@400000006328a6f935aab36c serstart starting
@400000006328a6f93655dbd4 INFO: loading config file /etc/venus/serial-starter.conf
@400000006328a70e2aeb1d2c INFO: loading config file /data/conf/serial-starter.d
@400000006328a70f29f4bedc INFO: Create daemontools service vedirect-interface.ttyUSB0
@400000006328a711209796fc INFO: Create daemontools service dbus-cgwacs.ttyUSB1
@400000006328a7122e71b794 INFO: Create daemontools service mk2-dbus.ttyUSB2
@400000006328a715316925a4 INFO: Start service vedirect-interface.ttyUSB0 once
@400000006328a7172a392e64 INFO: Start service dbus-cgwacs.ttyUSB1 once
@400000006328a718393d3c84 INFO: Start service mk2-dbus.ttyUSB2
@400000006328a73627be9944 INFO: Create daemontools service gps-dbus.ttyUSB1
@400000006328a73c3177db44 INFO: Start service gps-dbus.ttyUSB1 once
@400000006328a74606facfd4 INFO: Create daemontools service vedirect-interface.ttyUSB1
@400000006328a74c099233f4 INFO: Start service vedirect-interface.ttyUSB1 once

**VenusOS (please complete the following information):**
 - Raspberry Pi 4
 - Firmware Version 2.90

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: 3 * Daly
 - Cells: 16
 - Interface: USB-RS485 via USB-HUB

Louisvdw, 2022-09-19:
@centro-max which version of the driver do you have? If you are not on v0.12 upgrade to at least that one and do a reboot.
From VenusOS 2.80 the firmware had some changes which changed how we needed to install the driver.

centro-max, 2022-09-19:
I have installed all available Versions, 0.12 and 0.12b3, nothing works. I can not explain, I also downgraded to Version 2.89 from a backup SD, a version which worked bevor well, the same result.
I disabled all 3 batteries because I remodeled, and after i switched them on the driver was gone. The bluetooth app is working fine, so the daly bms works well. could there be a problem with the USB-plug?

Louisvdw, 2022-09-19:
I don't see serial-starter trying to start the serialbattery on USB1, so that tells me it thinks serialbattery is not installed.
Run this command using v0.12b3 and reboot to see if serial-starter picks it up. 
> sh /data/etc/dbus-serialbattery/reinstalllocal.sh

centro-max, 2022-09-19:
I found the problem, one USB 3.0 Port of the raspberry pi 4 seems not to work any more. and the 4-port usb-hub in which all 3 daly-bms-usb were pluged in is defect also. i can't explain why the usb port and also the usb hub broke.

Louisvdw, 2022-09-20:
That is strange, but at least you know what the issue is now. Perhaps it might be worth while upgrading to isolated cables for your 3x BMS just to avoid any power spikes affecting your devices

closed 2022-09-20 by Louisvdw