Louisvdw/dbus-serialbattery #218: upgrade VenusOS 2.89 to 2.90 -> ERROR: no service template for dbus-serialbattery

dominikde, 2022-09-25:
You have a working dbus-serialbattery installation and upgrade your VenusOS on Cerbo from 2.89 to 2.9, dbus-serialbattery stops working:

root@einstein:~# tail -f /data/log/serial-starter/current
@4000000063303b520f814e94 ERROR: no service template for dbus-serialbattery

 - Device type: Cerbo
 - Firmware Version 2.89 to 2.9

 - BMS/Battery type: JKBMS
 - Cells: 2 x 16
 - Interface: usb-rs485

When you reboot in 2.89 all is working again. How to fix this?

dominikde, 2022-09-26:
On 2.9 the mount point / is read only:

root@einstein:/opt/victronenergy# touch test
touch: test: Read-only file system
root@einstein:/opt/victronenergy# mount 
/dev/mmcblk1p2 on / type ext4 (ro,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=465376k,nr_inodes=116344,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /var/volatile type tmpfs (rw,relatime)
/dev/mmcblk1p5 on /data type ext4 (rw,noatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /service type tmpfs (rw,nosuid,nodev,mode=755)
overlay on /var/lib type overlay (rw,relatime,lowerdir=/var/lib,upperdir=/var/volatile/lib,workdir=/var/volatile/.lib-work)
root@einstein:/opt/victronenergy# 

So the install script cannot reinstall it in /opt/victronenergy:

oot@einstein:~# sh /data/etc/dbus-serialbattery/reinstalllocal.sh 
mkdir: can't create directory '/opt/victronenergy/dbus-serialbattery': Read-only file system
mkdir: can't create directory '/opt/victronenergy/service-templates/dbus-serialbattery': Read-only file system
cp: can't create '/opt/victronenergy/dbus-serialbattery/LICENSE': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/README.md': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/ant.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/battery.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/daly.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/dbushelper.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/disabledriver.sh': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/ecs.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/installlocal.sh': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/installqml.sh': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/installrelease.sh': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/jkbms.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/lifepower.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/lltjbd.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/minimalmodbus.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/mnb.py': No such file or directory
cp: omitting directory '/data/etc/dbus-serialbattery/qml'
cp: can't create '/opt/victronenergy/dbus-serialbattery/reinstalllocal.sh': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/renogy.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/revov.py': No such file or directory
cp: omitting directory '/data/etc/dbus-serialbattery/service'
cp: can't create '/opt/victronenergy/dbus-serialbattery/sinowealth.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/start-serialbattery.sh': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/util_max17853.py': No such file or directory
cp: can't create '/opt/victronenergy/dbus-serialbattery/utils.py': No such file or directory
cp: can't create directory '/opt/victronenergy/service-templates/dbus-serialbattery/log': No such file or directory
cp: can't create '/opt/victronenergy/service-templates/dbus-serialbattery/run': No such file or directory

dominikde, 2022-09-26:
on 2.89 the mount / is read write:

/dev/mmcblk1p3 on / type ext4 (rw,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=465376k,nr_inodes=116344,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /var/volatile type tmpfs (rw,relatime)
/dev/mmcblk1p5 on /data type ext4 (rw,noatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
overlay on /service type overlay (rw,relatime,lowerdir=/opt/victronenergy/service,upperdir=/run/overlays/service,workdir=/run/overlays/work/service)
root@einstein:~#

dominikde, 2022-09-26:
Ok, with the following command you can change to read/write FS:
/opt/victronenergy/swupdate-scripts/remount-rw.sh

closed 2022-09-26 by dominikde

Louisvdw, 2022-09-26:
Thanks for the feedback. I'll do some test with upgrades to 2.90 this week and adjust the script to include the remound