Louisvdw/dbus-serialbattery #233: Create Charge Current Control Management parameters

Louisvdw, 2022-10-02:
To be able to make the CCCM parameters adjustable, we need to create parameters for them in utils.py that users can adjust.

For Charge limits we need 
4x Charge SOC % parameters that can be used like:
- 100% -> ChargeSOC1 -> ChargeSOC2  -> ChargeSOC3  -> ChargeSOC4 -> below

3x ChargeLimits that can be used like:
- ChargeLimitFull -> ChargeLimitDiv1  -> ChargeLimitDiv2  -> no limit

For DisCharge limits we need 
4x DisCharge SOC % parameters that can be used like:
- 0% -> DischargeSOC1 -> DischargeSOC2  -> DischargeSOC3  -> DischargeSOC4 -> above

3x DisChargeLimits that can be used like:
- DischargeLimitLow -> DischargeLimitDiv1  -> DischargeLimitDiv2  -> no limit

WaldemarFech, 2022-10-04:
That would be nice.
But i'm no fried of hard borders ... so i've added some to your code for my purpose to make the limits more linear. Maybe i can take some work - if you like my thoughts.
Here is a quick and dirty 60 Minutes Code. Of course it's not finished - next: DischargeCurrentLimit

--------------
The output of CCL would be in my actual configuration then:

Mode:  Wally
SOC is:  -5% - CCL: 150.00A
...
SOC is:   0% - CCL: 150.00A
SOC is:   1% - CCL: 150.00A
SOC is:   2% - CCL: 150.00A
...
SOC is:  79% - CCL: 150.00A
SOC is:  **80**% - CCL: **123.00A**
SOC is:  81% - CCL: 118.20A
SOC is:  82% - CCL: 113.40A
SOC is:  83% - CCL: 108.60A
SOC is:  84% - CCL: 103.80A
SOC is:  85% - CCL: 99.00A
SOC is:  86% - CCL: 94.20A
SOC is:  87% - CCL: 89.40A
SOC is:  88% - CCL: 84.60A
SOC is:  89% - CCL: 79.80A
SOC is:  **90**% - CCL: **75.00A**
SOC is:  91% - CCL: 63.75A
SOC is:  92% - CCL: 52.50A
SOC is:  93% - CCL: 41.25A
SOC is:  94% - CCL: 30.00A
SOC is:  **95**% - CCL: **18.75A**
SOC is:  96% - CCL: 14.17A
SOC is:  97% - CCL: 9.58A
SOC is:  98% - CCL: 5.00A
SOC is:  99% - CCL: 3.00A
SOC is: **100**% - CCL: **1.00A**
SOC is: 101% - CCL: 1.00A
SOC is: 102% - CCL: 1.00A
SOC is: 103% - CCL: 1.00A
SOC is: 104% - CCL: 1.00A

in utils.py:
# battery Current limits
MAX_BATTERY_CURRENT = **150.0**
MAX_BATTERY_DISCHARGE_CURRENT = 150.0
# Charge current control management enable (True/False). 
CCCM_ENABLE = True
#CCCM_MODE = "Normal"   # "Normal" / "Wally"
                        # "Normal" -> old-style-limitation of ChargeCurrentLimit (CCL) in 4 Steps depending on StateOfCharge (SOC):
                        #  0 <=SOC <= 91 -> ChargeCurrentLimit: MAX_BATTERY_CURRENT
                        # 91 < SOC <= 95 -> ChargeCurrentLimit: MAX_BATTERY_CURRENT / 2
                        # 95 < SOC <= 98 -> ChargeCurrentLimit: MAX_BATTERY_CURRENT / 4
                        # 98 < SOC <= 100-> ChargeCurrentLimit: 5A - but... SOC > 99 -> don't allow charging anymore!
**CCCM_MODE = "Wally"**     # change limitation more linear depending on SOC - with configurable setpoints
CCCM_WALLY_SOC_1 = **80**                   # Use MAX_BATTERY_CURRENT between 0% and 80% SOC
CCCM_WALLY_CCL_1 = **123**  # Lower slowly Charge Current from MAX_BATTERY_CURRENT to MAX_BATTERY_CURRENT/2 between 80% and 90% SOC
CCCM_WALLY_SOC_2 = **90**
CCCM_WALLY_CCL_2 = **MAX_BATTERY_CURRENT / 2**
CCCM_WALLY_SOC_3 = **95**
CCCM_WALLY_CCL_3 = **MAX_BATTERY_CURRENT / 8**
CCCM_WALLY_SOC_4 = **98**
CCCM_WALLY_CCL_4 = **5**
CCCM_WALLY_SOC_5 = **100**
CCCM_WALLY_CCL_5 = **1**                    # Limit will not fall under this value

in battery.py:
    def manage_charge_current(self):
        if CCCM_MODE == "Wally":
            return self.manage_charge_current_WALLY()
        else:
            return self.manage_charge_current_ORIGINAL()

    @staticmethod
    def mapRange(value, inMin, inMax, outMin, outMax):
        return outMin + (((value - inMin) / (inMax - inMin)) * (outMax - outMin))

    def manage_charge_current_WALLY(self):
        # If disabled make sure the default values are set and then exit
        if (not CCCM_ENABLE):
            self.control_charge_current = self.max_battery_current
            self.control_discharge_current = self.max_battery_discharge_current
            self.control_allow_charge = True
            return

        # Start with the current values

        # Change depending on the SOC values
        if self.soc is None:
            # Prevent serialbattery from terminating on error
            return False

        soc = self.soc
        if soc < CCCM_WALLY_SOC_1:
            ccl = MAX_BATTERY_CURRENT
        elif CCCM_WALLY_SOC_1 <= soc < CCCM_WALLY_SOC_2:
            ccl = self.mapRange(soc, CCCM_WALLY_SOC_1, CCCM_WALLY_SOC_2, CCCM_WALLY_CCL_1, CCCM_WALLY_CCL_2)
        elif CCCM_WALLY_SOC_2 <= soc < CCCM_WALLY_SOC_3:
            ccl = self.mapRange(soc, CCCM_WALLY_SOC_2, CCCM_WALLY_SOC_3, CCCM_WALLY_CCL_2, CCCM_WALLY_CCL_3)
        elif CCCM_WALLY_SOC_3 <= soc < CCCM_WALLY_SOC_4:
            ccl = self.mapRange(soc, CCCM_WALLY_SOC_3, CCCM_WALLY_SOC_4, CCCM_WALLY_CCL_3, CCCM_WALLY_CCL_4)
        elif CCCM_WALLY_SOC_4 <= soc < CCCM_WALLY_SOC_5:
            ccl = self.mapRange(soc, CCCM_WALLY_SOC_4, CCCM_WALLY_SOC_5, CCCM_WALLY_CCL_4, CCCM_WALLY_CCL_5)
        else:
            ccl = CCCM_WALLY_CCL_5

        self.control_charge_current = ccl

        # Dischange depending on the SOC values
        if self.soc <= 10:
            self.control_discharge_current = 5
        elif 10 < self.soc <= 20:
            self.control_discharge_current = self.max_battery_discharge_current / 4
        elif 20 < self.soc <= 30:
            self.control_discharge_current = self.max_battery_discharge_current / 2
        else:
            self.control_discharge_current = self.max_battery_discharge_current

    def manage_charge_current_ORIGINAL(self):
        **_# here is your old code ..._**

WaldemarFech, 2022-10-04:
Additional Comment:
I am planing in reducing / rising Charge Current/Voltage Limitations depending on the maxCellVoltage, delta of minCellVoltage and maxCellVoltage, and the batterytemperature.

Because of this i've changed the get_temp() method, too:

bat.temp1=23.5 - bat.temp2=57.8 -> bat.get_temp()=57.8

### in battery.py

    def get_temp(self):
        if self.temp1 is not None and self.temp2 is not None:
            **return round(max(self.temp1, self.temp2), 2)                    # use the max of both -> worst case**
            #return round((float(self.temp1) + float(self.temp2)) / 2, 2)   # original
        if self.temp1 is not None and self.temp2 is None:
            return round(float(self.temp1) , 2)
        if self.temp1 is None and self.temp2 is not None:
            return round(float(self.temp2) , 2)
        else:
            return None

Louisvdw, 2022-10-04:
@WaldemarFech  It's an intresting idea to even out the charge current values. I think others might find that useful. Do you want to create a PR for that?

pos-ei-don, 2022-10-05:
I'd love this as well, as I currently have a runner cell and the bms begins to Block charging if this cell voltage is too High. To give the balancer a chance, I need to slow down charging in this case.

Louisvdw, 2022-10-06:
There is no private messages, but this is fine to discuss here.
Once you have made a Fork you will be able to make changes to you version. On the top right you will see a Sync Fork and Contribute button that you can use. Sync will take the latest from the master repository (mine) and sync anything new to your version. You will have to do this before you do the contribute, and fix any issues (only if the incoming updates has changes to the same lines you have changed). 
Once it is all up to date you can use the Contribute button to open a PR (Pull Request) which will compare the changes for review. That is it. Very easy :) 

[image]

Louisvdw, 2022-10-10:
fix in #238 
(not the WALLY version yet)

WaldemarFech, 2022-11-15:
**Update:**
I've tested and thinked a lot xD -> Result: Changing CCL / DCL depending on SoC is a little bit stupid.
**Spoiler:
I've got a working Solution.**

**The trivial problem is:**
The State of Charge (SoC) is inaccurate. The Multiplus calculates something different than the JK-BMS. And this is also different to the Victron Shunt. (I've done a lot of testing with all of them)
This means: While looking at the SoC, you don't charge or discharge the battery completely. OR: If you try to fine adjust the parameters depending on SoC, then it's just a question of time, when a Battery-High / -Low alert shuts down your Battery - in worst case in a loop every 10 seconds or so.

**The more detailed problem is:**
When we use your (Louis) recommended Battery-Settings, then it's okay for most People.
(3.45V per Cell is a realy realy reaaaaly good value! I recommend it also for beginners (Andy from off-grid-garage have done also lot's of tests, as you know, and convinced me, too).
But it can happen that when we've got unbalanced cells, or maybe adjusted too much, or in the most cases: the (in my case JK-) BMS shows up a state of Charge of 83%, even if the batteries voltage is already at 3.45V*16 = 55.2V - The charge-current wouldn't be decreased and suddenly one cell goes over 3.65V -> high voltage alert -> bms shuts the battery offline -> bad.
OR (in the last days in Germany)
The Battery is low. The BMS shows less than 1% - your driver disables the discharge, even if the voltage is above 2.7V*16 = 43.2.  Also here it could happen, that the cells are not good balanced and one cell drops under 2.6V and the BMS disconnects the battery.

### The idea for a solution
We just have to watch the cells - if one goes over a critical voltage, then we reduce the charge current. Same strategy for discharging.
My strategy:  Setup the ABSOLUTE maximum values for the voltage - if one cell drifts away, then we reduce the (dis-/)charge current anyway.
Charge with maximum current (like constant current)
If one cell reaches the first threshold, then begin reducing the charge-current - linear between two thresholds.
If one cell reaches the highest threshold, then the charge current will be set to 0 and allow_charge will be disabled.
This simulates an absorption time - future request: configurable absorption **time**

```
#Set Steps to reduce charging current. The current will be changed linear between those steps
    CELL_VOLTAGES_WHILE_CHARGING    = [3.55, 3.50, 3.45, 3.30]  # first value must be the highest
    MAX_CHARGE_CURRENT              = [   0,    2,  100,  200]

    CELL_VOLTAGES_WHILE_DISCHARGING = [2.70, 2.80, 2.90, 3.10]  # first value must be the lowest
    MAX_DISCHARGE_CURRENT           = [   0,    5,  100,  200]
```

**My Fork**
My code is based on your v0.14beta1. I am testing it right now in two houses and was testing it in a simulation, too. It seems to work well. If i can figure out how, then i start a pull-request.
Regards
Waldemar Fech

pos-ei-don, 2022-11-15:
Great. Seems to be the solution I need aswell.

WaldemarFech, 2022-11-15:
You can already test it with following ssh commands:

```
rm venus-data.tar.gz
wget -O venus-data.tar.gz https://github.com/WaldemarFech/dbus-serialbattery/releases/download/v0.14beta1_WFech/venus-data.tar.gz
tar -zxf ./venus-data.tar.gz -C /data
sh /data/etc/dbus-serialbattery/reinstalllocal.sh
reboot
```

mascheihei, 2022-11-16:
Hi Waldemar,

that's exactly what was in my mind and started to code (but your version is much better). Running your solution with my JK-BMS (test environment) very smooth at the moment. 

regards
Markus

pos-ei-don, 2022-11-16:
Can we add temperature aswell to reduce the current?

pos-ei-don, 2022-11-16:
I'd use safe values. For my setup, everything between 12-38° can use Max current. Above that i'd reduse this to 20A. Below 0 i'd reduce to 0A.
But i use Lifepo4 cells. Other cells have different values.

This has the possibility to reduce hot wires, reduce the possibility of fire and so on. I'd really live this!

Keepin mind that a bms Might have more than one temp sensor. JKBMS has 2. I use on between the cells and one at the fuse.

WaldemarFech, 2022-11-16:
That's a good idea!
In my case: I know, that there is something wrong IN my battery-switch - it gets very hot, but nothing is melting ^^ ... so i placed one temp-sensor there to monitor the temperature in the victron system. @Louisvdw idea was to middle the values, if there are two sensors (as in the JK-BMS). I knew, that my battery is inside and will never get too cold, but maybe too hot (at the switch). So i changed his Code to use always the max-value of both.

But when i now read your 0°C example, then we maybe need such a similar setting. Why not. I'll give it a try and change my pull-request / place another.

The first idea would be something like this:
```

    #Temperature current control management enable (True/False).
    TCCM_ENABLE = True
    #Set Steps to reduce battery-current, depending on the temperature. Current will be changed linear between the steps
    TEMPERATURE_LIMITS  = [55, 38,  36,  13, 12, 0]     # first value must be the highest
    MAX_CURRENT         = [ 0, 20, 200, 200, 50, 0]
```
And then the code would split the array in half, take the max-value (of two sensors) for the left half and the min-value for the right. What do you all think about this approach?

But if we read the datasheet for the EVE-Cells (i got 18s 280Ah), then we see differenzt limits for charging / discharging. I can also create two/four separate arrays for charging / discharging.
[image]

pos-ei-don, 2022-11-16:
My braker stays quiet cool, the fuse rises to 38° when charging with 50 A or more. So that will be my Limit.
Thank you for this great Patch.

WaldemarFech, 2022-11-16:
**Done** - but not published yet - comming soon.

Here are two results of my test-environment
The first one is about the cell-voltage. The second print is the behavior with high and low temperature. My code does the calculation two times - for min_temp and for max_temp and chooses the minimum of both. Of course also the minimum of the results after the cell-voltage-depending result and the temperature-depending result.

These are my test-settings (if you want to compare, you should open a new tab xD):
````
if LIMITATION_MODE == "WaldemarFech":
    # Charge current control management referring to cell-voltage enable (True/False).
    CCCM_CV_ENABLE = True
    # Discharge current control management referring to cell-voltage enable (True/False).
    DCCM_CV_ENABLE = True
    # Charge current control management referring to cell-voltage enable (True/False).
    CCCM_T_ENABLE = True
    # Charge current control management referring to cell-voltage enable (True/False).
    DCCM_T_ENABLE = True

    # Set Steps to reduce battery current. The current will be changed linear between those steps
    CELL_VOLTAGES_WHILE_CHARGING         = [3.55, 3.50, 3.45, 3.30]     # first value must be the highest
    MAX_CHARGE_CURRENT_CV                = [   0,    2,  100,  200]

    CELL_VOLTAGES_WHILE_DISCHARGING      = [2.70, 2.80, 2.90, 3.10]     # first value must be the lowest
    MAX_DISCHARGE_CURRENT_CV             = [   0,    5,  100,  200]

    TEMPERATURE_LIMITS_WHILE_CHARGING    = [55, 40,  35,   5,  2, 0]    # first value must be the highest
    MAX_CHARGE_CURRENT_T                 = [ 0, 28, 200, 200, 28, 0]

    TEMPERATURE_LIMITS_WHILE_DISCHARGING = [55, 40,  35,   5,  0, -20]  # first value must be the highest
    MAX_DISCHARGE_CURRENT_T              = [ 0, 28, 200, 200, 28,   0]
````

**Influence of the cell voltage on the CCL / DCL**
```
Starte Test
Mode:  WaldemarFech
c1.volt is: 2.65V - CCL: 200.00A - DCL:   0.00A
c1.volt is: 2.66V - CCL: 200.00A - DCL:   0.00A
c1.volt is: 2.67V - CCL: 200.00A - DCL:   0.00A
c1.volt is: 2.68V - CCL: 200.00A - DCL:   0.00A
c1.volt is: 2.69V - CCL: 200.00A - DCL:   0.00A
c1.volt is: 2.70V - CCL: 200.00A - DCL:   0.00A
c1.volt is: 2.71V - CCL: 200.00A - DCL:   0.50A
c1.volt is: 2.72V - CCL: 200.00A - DCL:   1.00A
c1.volt is: 2.73V - CCL: 200.00A - DCL:   1.50A
c1.volt is: 2.74V - CCL: 200.00A - DCL:   2.00A
c1.volt is: 2.75V - CCL: 200.00A - DCL:   2.50A
c1.volt is: 2.76V - CCL: 200.00A - DCL:   3.00A
c1.volt is: 2.77V - CCL: 200.00A - DCL:   3.50A
c1.volt is: 2.78V - CCL: 200.00A - DCL:   4.00A
c1.volt is: 2.79V - CCL: 200.00A - DCL:   4.50A
c1.volt is: 2.80V - CCL: 200.00A - DCL:   5.00A
c1.volt is: 2.81V - CCL: 200.00A - DCL:  14.50A
c1.volt is: 2.82V - CCL: 200.00A - DCL:  24.00A
c1.volt is: 2.83V - CCL: 200.00A - DCL:  33.50A
c1.volt is: 2.84V - CCL: 200.00A - DCL:  43.00A
c1.volt is: 2.85V - CCL: 200.00A - DCL:  52.50A
c1.volt is: 2.86V - CCL: 200.00A - DCL:  62.00A
c1.volt is: 2.87V - CCL: 200.00A - DCL:  71.50A
c1.volt is: 2.88V - CCL: 200.00A - DCL:  81.00A
c1.volt is: 2.89V - CCL: 200.00A - DCL:  90.50A
c1.volt is: 2.90V - CCL: 200.00A - DCL: 100.00A
c1.volt is: 2.91V - CCL: 200.00A - DCL: 105.00A
c1.volt is: 2.92V - CCL: 200.00A - DCL: 110.00A
c1.volt is: 2.93V - CCL: 200.00A - DCL: 115.00A
c1.volt is: 2.94V - CCL: 200.00A - DCL: 120.00A
c1.volt is: 2.95V - CCL: 200.00A - DCL: 125.00A
c1.volt is: 2.96V - CCL: 200.00A - DCL: 130.00A
c1.volt is: 2.97V - CCL: 200.00A - DCL: 135.00A
c1.volt is: 2.98V - CCL: 200.00A - DCL: 140.00A
c1.volt is: 2.99V - CCL: 200.00A - DCL: 145.00A
[... 121 lines cut ...]
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 36°C -> bat.get_temp()=30.50°C - CCL: 165.60A - DCL: 165.60A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 37°C -> bat.get_temp()=31.00°C - CCL: 131.20A - DCL: 131.20A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 38°C -> bat.get_temp()=31.50°C - CCL:  96.80A - DCL:  96.80A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 39°C -> bat.get_temp()=32.00°C - CCL:  62.40A - DCL:  62.40A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 40°C -> bat.get_temp()=32.50°C - CCL:  28.00A - DCL:  28.00A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 41°C -> bat.get_temp()=33.00°C - CCL:  26.13A - DCL:  26.13A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 42°C -> bat.get_temp()=33.50°C - CCL:  24.27A - DCL:  24.27A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 43°C -> bat.get_temp()=34.00°C - CCL:  22.40A - DCL:  22.40A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 44°C -> bat.get_temp()=34.50°C - CCL:  20.53A - DCL:  20.53A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 45°C -> bat.get_temp()=35.00°C - CCL:  18.67A - DCL:  18.67A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 46°C -> bat.get_temp()=35.50°C - CCL:  16.80A - DCL:  16.80A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 47°C -> bat.get_temp()=36.00°C - CCL:  14.93A - DCL:  14.93A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 48°C -> bat.get_temp()=36.50°C - CCL:  13.07A - DCL:  13.07A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 49°C -> bat.get_temp()=37.00°C - CCL:  11.20A - DCL:  11.20A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 50°C -> bat.get_temp()=37.50°C - CCL:   9.33A - DCL:   9.33A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 51°C -> bat.get_temp()=38.00°C - CCL:   7.47A - DCL:   7.47A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 52°C -> bat.get_temp()=38.50°C - CCL:   5.60A - DCL:   5.60A
[... cut ...]
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 55°C -> bat.get_temp()=40.00°C - CCL:   0.00A - DCL:   0.00A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 56°C -> bat.get_temp()=40.50°C - CCL:   0.00A - DCL:   0.00A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 57°C -> bat.get_temp()=41.00°C - CCL:   0.00A - DCL:   0.00A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 58°C -> bat.get_temp()=41.50°C - CCL:   0.00A - DCL:   0.00A
c1.voltage=3.20V - bat.temp1= 25°C - bat.temp2= 59°C -> bat.get_temp()=42.00°C - CCL:   0.00A - DCL:   0.00A
````

pos-ei-don, 2022-11-17:
I think it works really gread. I can ommit to block charging/discharging with this when a my badest cell (i have a runner cell in my system!) voltage is too high or too low.
I use this to slow down charging to help the balancers (i have two of them) to fix the cell-voltage-peak. with this enabled, i can charge the battery much better! great feature!!!

WaldemarFech, 2022-11-17:
Here are some insights how it worked yesterday:

**Min/Max Cell-Voltage over 4h when battery gets full and the corresponding CCL / DCL in this time**
[image][image]

**Here is a detailed view what happened within one hour - Min/Max Cell-Voltage**
The Driver is lowering the CCL as wanted. I used this settings:
```
    CELL_VOLTAGES_WHILE_CHARGING         = [3.55, 3.50, 3.45, 3.30]     # first value must be the highest
    MAX_CHARGE_CURRENT_CV                = [   0,    2,  100,  200]
```
[image][image]

**Normal, but interesting to see, how the Voltage rises while the current goes down and how fast the cell-voltage rises**
[image][image]

**And what still f*cks me up:  left - JK-BMS-SOC. right - Victron Smart Shunt 500A** 
If we would SoC for CCL/DCL- we would have a problem... I am thinking about manipulating the serial-battery-SoC in the Driver and calculating it "better" (?!) haha. Maybe with a proportional factor what the JK says, what the smartShunt says and some kind of offset when one cell reaches the 100% -> then the SoC should definitly be 100%!!!
[image][image]

pos-ei-don, 2022-11-17:
I thought about that aswell.... I thing the averwge cell voltage and the maximum/minimum are good values to estimate the soc.

Louisvdw, 2022-11-18:
If you compare the SOC they will never be the same. You should not expect that to be the case as each one will be using different logic, input/sensors (and hardware) to measure that. Within 5% of each other is actually not bad.

pos-ei-don, 2022-11-18:
I dont need "the same result". What I'd like is not getting a soc of 70% when a cell is already over 3.5v.. The jk bms soc will wait until this cell gets to 3.6v and change the doc from 70% to 100% immediately. 
I'd like to stop charging a bit earlier. With this Patch I can do this.

WaldemarFech, 2022-11-18:
>The jk bms soc will wait until this cell gets to 3.6v and change the doc from 70% to 100% immediately.

Well, it seems like, that mine was setting it to 100% first, when it shuts down because of high cell voltage:
[image]

I think i'll set the 100%-Voltage in the shunt to 61.8V (i got 18S -> 3.4333 V / Cell avg)
[image][image]

I want to figure out if this 100% SoC will disable anything (i hope not. it should just be a visualisation)
[image]

pos-ei-don, 2022-11-18:
I think I saw that it does not disable anything, but the victron charger change to Absorbtion or Float Mode. I am not shure but i think it is supposed to be like this.

Louisvdw, 2022-11-18:
This is what makes BMS so nice, but is also sometimes hard to understand to users. When it jumpt from 90% to 100% because one cell reach the overvoltage, that is what we want to see. The Smartshunt does not know the values of each cell, so it cannot make adjustments like that - it can only count the energy passing through it (and look at the total battery voltage).

A State Of Charge of 100% means you have reached the limit of what energy you can put into the battery. The reason in this case is that at least 1 cell has reach it's charge limit. 

There are 2 possible reasons for a jump to happen.
1) either some cells are not ballanced and so reach full capacity before others. 
2) your capacity settings for the cells do not match
If you solve the issues you should also not see jumps.

If your battery reach 100% then nothing changes in your Victron system. It will only stop charging when the charge voltage has been reached, or if the battery (the serialbattery driver in this case) tells it to limit charging.

pos-ei-don, 2022-11-18:
There is a third reason.
One of my cells degradeed more than others. It has now 45Ah less capacity then All other cells. So this cell is always the first one having both, low and High voltage.
But i dont mind, it works still..... 😉

Louisvdw, 2022-11-18:
I count that under option 2. Your battery capacity is now 45Ah less that what it was. Some BMS like the JBD will adjust it's capacity from what it sees happening in the cells, but again the Smurtshunt would not know about that so you need to manually lower it's capacity setting. 
As cells get older it is normal that they would loose capacity like this. The difference between the original capacity and the actual capacity would be the State Of Health (SOH) that some BMS types reports.

I also have a set of cells like this (actually the reason why I wrote the driver in the first place). If you lower your charge voltage you would also get more energy stored as the voltage jumps are over a lower voltage and there is then more time for the balancer to work and the BMS to add energy to the rest of the cells. For me if I set by system to to charge up to 3.39V per cell then all is happy.

pos-ei-don, 2022-11-19:
Just a theoretical question: would it be possible in the victron System to add a configuration Page where one can Set the uper and lower desired cell voltages?
I know this is possible with node Red. 
The Main reason I ask is that i normally want to set "safe" values, but when I am running out of Power I might change them to some wider Limits. 
Another question: as I understood, the proposed solution woudl be a new "meta battery plugin" for victron, where I can sum multiple batteries and Decide, which soc  and which current settings from other batteries should be used and displayed.

Louisvdw, 2022-11-19:
The Graphical User Interface is what I am currently working on. So no more edit of utils.py by users. https://github.com/Louisvdw/dbus-serialbattery/pull/252
The summary battery is #8 
The choice which SOC is to be used is already built into Venus 2.92.

pos-ei-don, 2022-11-20:
> **Done** - but not published yet - comming soon.
>
> Here are two results of my test-environment The first one is about the cell-voltage. The second print is the behavior with high and low temperature. [...]

Testing this works great!
But i See A LOT OF change to the CCCM.
Should t this be equilized a little bit?
Or Might it be possible to not calculate this value linear? 
I'd prefer steps anyway ;-)

WaldemarFech, 2022-11-21:
easy.
You can do something like this
eg:

cccm_cellvolts and amps:
[..., 3.30, 3.31, 3.40, 3.41, 3.50, 3.51]
[...,  200,  100,  100,    25,    25,      0]

the big range between 3.31 and 3.40 would be constant and then you have your hard step.

the code is made to be dynamic with the amount of values you wish. just watch out to have the same numbers of values in both arrays

Louisvdw, 2022-11-21:
We'll have the option to do steps or curved CCCM values. 
I'm spillting the nice work Waldemar has done so that you can choose stepped or curved and also if you want to apply the limits by SOC, cell voltage or temprature. 

The arrays looks to be a bit harder to implement in GUI settings, so that might take some thinking.

pos-ei-don, 2022-11-21:
> easy. You can do something like this eg:
> 
> cccm_cellvolts and amps: [..., 3.30, 3.31, 3.40, 3.41, 3.50, 3.51] [..., 200, 100, 100, 25, 25, 0]

Great, i didn't want to crash the code ;-)
I found out that if set too narrow, the charging will alternate between 25 and 200A because the cell voltage drops immediately when the charger reduces from 200A to 25A.
I'm testing with some "threshold" now.

closed 2022-12-14 by Louisvdw