Louisvdw/dbus-serialbattery #239: Renogy randomly disconnects (v0.13)

TimGFoley, 2022-10-06:
While watching the Device List in the remote console, the Serialbattery will randomly show disconnected, it will reconnect after a few moments (less than a minute, will time next it happens).

Will continue to monitor to see if I can find a cause or pattern.

---

Venus GX (Victron OEM)
Firmware Version v2.91

Battery:
Renogy Smart Lithium 100AH
Cells: 4
Interface: Renogy USB to RS485 OEM Cable (no other devices connected to battery)

Copy of current log when it happens here:

@40000000633e00a2209ddc74   File "/opt/victronenergy/dbus-serialbattery/renogy.py", line 148, in read_temp_data
@40000000633e00a22147a164     self.temp2 = unpack('>H',temp2)[0] / 10
@40000000633e00a22147c48c TypeError: a bytes-like object is required, not 'bool'
@40000000633e00d0198ecefc INFO:SerialBattery:Starting dbus-serialbattery
@40000000633e00d019acea04 INFO:SerialBattery:dbus-serialbattery v0.13
@40000000633e00d019ed90cc INFO:SerialBattery:Testing LltJbd
@40000000633e00d02b33518c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e00d02b5c3a34 INFO:SerialBattery:Testing Ant
@40000000633e00d10416787c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e00d1043c194c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000633e00d1044cede4 INFO:SerialBattery:Testing Daly
@40000000633e00d1153c8fc4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e00d1156950cc INFO:SerialBattery:Testing Daly
@40000000633e00d1276486ac ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e00d127a642cc INFO:SerialBattery:Testing Jkbms
@40000000633e00d215a6f9f4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e00d2168d0494 INFO:SerialBattery:Testing Sinowealth
@40000000633e00d22f7066cc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e00d22fa211a4 INFO:SerialBattery:Testing Lifepower
@40000000633e00d31d1390bc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e00d31fdec67c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000633e00d320d2272c INFO:SerialBattery:Testing Renogy
@40000000633e00d33603d7bc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e00d3362f44ec INFO:SerialBattery:Testing Renogy
@40000000633e00d40e02efdc INFO:SerialBattery:RENOGY RBT100LFP12S-G1
@40000000633e00d4229877dc INFO:SerialBattery:Connection established to Renogy
@40000000633e00d425b5e094 INFO:SerialBattery:Battery connected to dbus from /dev/ttyUSB2
@40000000633e00d4261ad83c INFO:SerialBattery:=== Settings ===
@40000000633e00d4262d52b4 INFO:SerialBattery:> Connection voltage NoneV | current NoneA | SOC None%
@40000000633e00d4263c3b1c INFO:SerialBattery:> Cell count 4 | cells populated 4
@40000000633e00d42eefb57c INFO:SerialBattery:> CCL Charge NoneA | DCL Discharge NoneA
@40000000633e00d42f6cd4bc INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.45V
@40000000633e00d5165b424c INFO:SerialBattery:DeviceInstance = 1
@40000000633e00d516968a14 INFO:SerialBattery:com.victronenergy.battery.ttyUSB2
@40000000633e00db1c22fecc ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@40000000633e00f234326764 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e00f234e4e2cc Traceback (most recent call last):
@40000000633e00f234e530ec   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 163, in publish_battery
@40000000633e00f234e54474     success = self.battery.refresh_data()
@40000000633e00f234e5502c   File "/opt/victronenergy/dbus-serialbattery/renogy.py", line 72, in refresh_data
@40000000633e00f234e55fcc     result = result and self.read_temp_data()
@40000000633e00f234e56b84   File "/opt/victronenergy/dbus-serialbattery/renogy.py", line 148, in read_temp_data
@40000000633e00f234e72cbc     self.temp2 = unpack('>H',temp2)[0] / 10
@40000000633e00f234e74044 TypeError: a bytes-like object is required, not 'bool'
@40000000633e011c15538a94 INFO:SerialBattery:Starting dbus-serialbattery
@40000000633e011c156783dc INFO:SerialBattery:dbus-serialbattery v0.13
@40000000633e011c15ac7834 INFO:SerialBattery:Testing LltJbd
@40000000633e011c26067964 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e011c2631272c INFO:SerialBattery:Testing Ant
@40000000633e011c3779b6ac ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e011c37aa7724 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000633e011c37d5d0cc INFO:SerialBattery:Testing Daly
@40000000633e011d0dc569dc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e011d0dedf0dc INFO:SerialBattery:Testing Daly
@40000000633e011d21f8e064 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e011d2228702c INFO:SerialBattery:Testing Jkbms
@40000000633e011d32cda35c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e011d32fbac84 INFO:SerialBattery:Testing Sinowealth
@40000000633e011e07e9cb84 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e011e0810a0ec INFO:SerialBattery:Testing Lifepower
@40000000633e011e1c0852ac ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e011e1c2d014c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000633e011e1c3dba8c INFO:SerialBattery:Testing Renogy
@40000000633e011e2daf26fc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000633e011e2e2d0d74 INFO:SerialBattery:Testing Renogy
@40000000633e011f043314b4 INFO:SerialBattery:RENOGY RBT100LFP12S-G1
@40000000633e011f0d933e1c INFO:SerialBattery:Connection established to Renogy
@40000000633e011f0da6f4fc INFO:SerialBattery:Battery connected to dbus from /dev/ttyUSB2
@40000000633e011f0e197964 INFO:SerialBattery:=== Settings ===
@40000000633e011f0e5eac3c INFO:SerialBattery:> Connection voltage NoneV | current NoneA | SOC None%
@40000000633e011f0e859cfc INFO:SerialBattery:> Cell count 4 | cells populated 4
@40000000633e011f0ed89b14 INFO:SerialBattery:> CCL Charge NoneA | DCL Discharge NoneA
@40000000633e011f10b8c4f4 INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.45V
@40000000633e011f2214d4a4 INFO:SerialBattery:DeviceInstance = 1
@40000000633e011f22882bfc INFO:SerialBattery:com.victronenergy.battery.ttyUSB2

Louisvdw, 2022-12-12:
Hi Tim

The protocol doc is in the old ticket #12 

Most of these cheaper BMS only have 1,2 or 3 temp sensors, and not a sensor for each cell. So which cell that min/max temp belongs too is not known, and so also not in their protocol.

TimGFoley, 2023-01-05:
I just updated to 14.3, no change however.

I also just added a second renogy battery but serialbattery only finds the one on the bus, possibly the driver does not yet support multiple batteries or this newer battery (self heated smart) has a different ID.

@4000000063b651681a37fb94 INFO:SerialBattery:Testing Renogy
@4000000063b651682c8e70d4 INFO:SerialBattery:RENOGY RBT100LFP12S-G1
@4000000063b65168347e54e4 INFO:SerialBattery:Connection established to Renogy
@4000000063b6516834916f84 INFO:SerialBattery:Battery Renogy connected to dbus from /dev/ttyUSB1
@4000000063b6516834a05404 INFO:SerialBattery:=== Settings ===
@4000000063b6516834b3e3d4 INFO:SerialBattery:> Connection voltage NoneV | current NoneA | SOC None%
@4000000063b6516834c41c2c INFO:SerialBattery:> Cell count 4 | cells populated 4
@4000000063b6516834d33f2c INFO:SerialBattery:> CCCM SOC True | DCCM SOC True
@4000000063b6516834e16444 INFO:SerialBattery:> CCCM CV True | DCCM CV True
@4000000063b6516834ef5a7c INFO:SerialBattery:> CCCM T True | DCCM T True
@4000000063b6516834fe566c INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.45V
@4000000063b65169064ad50c INFO:SerialBattery:DeviceInstance = 2
@4000000063b65169068c71ec INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
@4000000063b652db07bf7b7c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063b652db2ee2e054 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
@4000000063b652db2f833b1c Traceback (most recent call last):
@4000000063b652db2f836de4   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 193, in publish_battery
@4000000063b652db2f83816c     success = self.battery.refresh_data()
@4000000063b652db2f838d24   File "/opt/victronenergy/dbus-serialbattery/renogy.py", line 68, in refresh_data
@4000000063b652db2f839cc4     result = result and self.read_temp_data()
@4000000063b652db2f83a87c   File "/opt/victronenergy/dbus-serialbattery/renogy.py", line 144, in read_temp_data
@4000000063b652db2f8565cc     self.temp2 = unpack('>H',temp2)[0] / 10
@4000000063b652db2f85756c TypeError: a bytes-like object is required, not 'bool'
@4000000063b6530a06a122cc INFO:SerialBattery:Starting dbus-serialbattery
@4000000063b6530a06b60a5c INFO:SerialBattery:dbus-serialbattery v0.14.3
@4000000063b6530a0725fa9c INFO:SerialBattery:Testing LltJbd
@4000000063b6530a187ee59c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063b6530a18a7e99c INFO:SerialBattery:Testing Ant
@4000000063b6530a2d06f07c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063b6530a2d2c6e24 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000063b6530a2d3cf884 INFO:SerialBattery:Testing Daly
@4000000063b6530b04dcf4fc ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063b6530b05127064 INFO:SerialBattery:Testing Daly
@4000000063b6530b16b32264 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063b6530b16d99df4 INFO:SerialBattery:Testing Jkbms
@4000000063b6530b28351164 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063b6530b28729764 INFO:SerialBattery:Testing Lifepower
@4000000063b6530c01b399ac ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063b6530c01dfba8c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000063b6530c01f0930c INFO:SerialBattery:Testing Renogy
@4000000063b6530c1391c9dc INFO:SerialBattery:RENOGY RBT100LFP12S-G1
@4000000063b6530c1cb2f644 INFO:SerialBattery:Connection established to Renogy
@4000000063b6530c1ded00cc INFO:SerialBattery:Battery Renogy connected to dbus from /dev/ttyUSB1
@4000000063b6530c1e512584 INFO:SerialBattery:=== Settings ===
@4000000063b6530c1e8d9dfc INFO:SerialBattery:> Connection voltage NoneV | current NoneA | SOC None%
@4000000063b6530c1f31adfc INFO:SerialBattery:> Cell count 4 | cells populated 4
@4000000063b6530c1f7e7e0c INFO:SerialBattery:> CCCM SOC True | DCCM SOC True
@4000000063b6530c1f8fd38c INFO:SerialBattery:> CCCM CV True | DCCM CV True
@4000000063b6530c1f9ec7ac INFO:SerialBattery:> CCCM T True | DCCM T True
@4000000063b6530c1fae021c INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.45V
@4000000063b6530c2ce654d4 INFO:SerialBattery:DeviceInstance = 2
@4000000063b6530c2d29a34c INFO:SerialBattery:com.victronenergy.battery.ttyUSB1

TimGFoley, 2023-01-05:
Ah now I see why, the current implementation of the code assumes that the RS485 address of a BMS device is unique to a manufacturer.  There is no routine to scan all RS485 addresses and check for a battery BMS type at each address.

For example it assumes Renogy would be:

{"bms": "Renogy", "address": b"\x30"}, #x30=48 dec
{"bms": "Renogy", "address": b"\xF7"}, #xF7=247 dec

Yet a Renogy battery could be address 01-247 (x01-xF7).

TimGFoley, 2023-01-05:
Added some more debug to various modules and was able to find the addresses of both my batteries.  One is address 0 and the other is address 1.  This took me a while to find as the debug code was not spitting out which RS485 address it found so I added some logger info for that:

@4000000063b6fc893060f4d4 INFO:SerialBattery:-------------------------------------------------------------------------------
@4000000063b6fc8930725ddc INFO:SerialBattery:Starting dbus-serialbattery
@4000000063b6fc8930832aa4 INFO:SerialBattery:dbus-serialbattery v0.14.3.timtest
@4000000063b6fc89309395c4 INFO:SerialBattery:Battery to be tested is:
@4000000063b6fc8930aa9c4c INFO:SerialBattery:{'bms': 'Renogy', 'address': b'1'}
@4000000063b6fc8930badc74 INFO:SerialBattery:Testing Renogy
@4000000063b6fc8930e46544 INFO:SerialBattery:Address = b'1'
@4000000063b6fc8a076cbbe4 INFO:SerialBattery:RENOGY RBT100LFP12SH-G1
@4000000063b6fc8a0fcb1164 INFO:SerialBattery:Connection established to Renogy
@4000000063b6fc8a0fdd2a34 INFO:SerialBattery:Battery Renogy connected to dbus from /dev/ttyUSB1
@4000000063b6fc8a0feba53c INFO:SerialBattery:=== Settings ===

TimGFoley, 2023-01-07:
Found the newer Renogy Protocol spec and have been testing with some other code snippets from another author, now things are making a little more sense.

When scanning addresses x01-xF7, I find my two batteries at 0x48 and 0x49.  It's not perfect yet but at least I can most of the registers from multiple batteries on the RS485 bus with that code.

Register                 Address   Value               Binary
----------------------------------------------------------------------------------------------
cell_count               0x1388    4                   00000000 00000100
cellvoltage_1            0x1389    3.30 V              00000000 00100001
cellvoltage_2            0x138a    3.30 V              00000000 00100001
cellvoltage_3            0x138b    3.30 V              00000000 00100001
cellvoltage_4            0x138c    3.30 V              00000000 00100001
temp_count               0x1399    4                   00000000 00000100
celltemp_1               0x139a    16.00 °C            00000000 10100000
celltemp_2               0x139b    16.00 °C            00000000 10100000
celltemp_3               0x139c    16.00 °C            00000000 10100000
celltemp_4               0x139d    16.00 °C            00000000 10100000
BMS Board Temp           0x13ab    0.00 °C             00000000 00000000
Num Env Temp Sensors     0x13ac    0                   00000000 00000000
Env Temp Sensor 1        0x13ad    25.00 °C            00000000 11111010
Env Temp Sensor 2        0x13ae    25.00 °C            00000000 11111010
Num Htr Temp Sensors     0x13af    0                   00000000 00000000
Htr Temp Sensor 1        0x13b0    25.00 °C            00000000 11111010
Htr Temp Sensor 2        0x13b1    25.00 °C            00000000 11111010
Current                  0x13b2    0.11 A              00000000 00001011
Voltage                  0x13b3    13.50 V             00000000 10000111
Remaining Capacity       0x13b4    97.17 Ah            00000000 00000001 01111011 10001111
Total Capacity           0x13b6    100.00 Ah           00000000 00000001 10000110 10100000
Cycle Number             0x13b8    0                   00000000 00000000
Charge Voltage Limit     0x13b9    14.80 V             00000000 10010100
Discharge Voltage Limit  0x13ba    10.00 V             00000000 01100100
Charge Current Limit     0x13bb    50.00 A             00010011 10001000
Discharge Current Limit  0x13bc    555.36 A            11011000 11110000
Serial Num               0x13f6    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ    11111111 11111111 11111111 11111...
Manufacture Version      0x13fe    0100                00110000 00110001 00110000 00110000
Main Line Version        0x13ff    0001                00110000 00110000 00110000 00110001
Comms Protocl Version    0x1401    19                  00110001 00111001
Battery Name             0x1402    RBT100LFP12SH-G1    01010010 01000010 01010100 00110...
Software Version         0x140a    0131                00110000 00110001 00110011 00110001
Manufacturer             0x140c    RENOGY              01010010 01000101 01001110 01001...

This one may be off due to the older protocol version, it's my other battery at address 0x49
Register                 Address   Value               Binary
----------------------------------------------------------------------------------------------
cell_count               0x1388    4                   00000000 00000100
cellvoltage_1            0x1389    3.30 V              00000000 00100001
cellvoltage_2            0x138a    3.30 V              00000000 00100001
cellvoltage_3            0x138b    3.30 V              00000000 00100001
cellvoltage_4            0x138c    3.30 V              00000000 00100001
temp_count               0x1399    3                   00000000 00000011
celltemp_1               0x139a    16.00 °C            00000000 10100000
BMS Board Temp           0x13ab    0.00 °C             00000000 00000000
Num Env Temp Sensors     0x13ac    1                   00000000 00000001
Env Temp Sensor 1        0x13ad    18.00 °C            00000000 10110100
Env Temp Sensor 2        0x13ae    0.00 °C             00000000 00000000
Num Htr Temp Sensors     0x13af    1                   00000000 00000001
Htr Temp Sensor 1        0x13b0    17.00 °C            00000000 10101010
Htr Temp Sensor 2        0x13b1    0.00 °C             00000000 00000000
Current                  0x13b2    327.58 A            11111111 11110110
Voltage                  0x13b3    13.40 V             00000000 10000110
Remaining Capacity       0x13b4    93.66 Ah            00000000 00000001 01101101 11011000
Total Capacity           0x13b6    97.75 Ah            00000000 00000001 01111101 11010110
Cycle Number             0x13b8    225                 00000000 11100001
Charge Voltage Limit     0x13b9    14.80 V             00000000 10010100
Discharge Voltage Limit  0x13ba    10.00 V             00000000 01100100
Charge Current Limit     0x13bb    50.00 A             00010011 10001000
Discharge Current Limit  0x13bc    555.36 A            11011000 11110000
Serial Num               0x13f6    PPTAH01050C08270    01010000 01010000 01010100 01000...
Manufacture Version      0x13fe    0100                00110000 00110001 00110000 00110000
Main Line Version        0x13ff    0001                00110000 00110000 00110000 00110001
Comms Protocl Version    0x1401    02                  00110000 00110010
Battery Name             0x1402    RBT100LFP12S-G1     01010010 01000010 01010100 00110...
Software Version         0x140a    0014                00110000 00110000 00110001 00110100
Manufacturer             0x140c    RENOGY              01010010 01000101 01001110 01001...

TimGFoley, 2023-01-07:
[Lithium.Iron.Battery.BMS.Modbus.Protocol.V1.7.zh-CN.en.1.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/10367317/Lithium.Iron.Battery.BMS.Modbus.Protocol.V1.7.zh-CN.en.1.pdf)

TimGFoley, 2023-01-08:
So judging by the original error where it tried to unpack a Boolean variable, I suspect the code tried to read temp2 which does not exist on this battery, so the read function returned false and the unpack of course failed since it expect a byte string and not a bool.

Simply adding a check to see if reading temp2 returns false should cover it (we already had a check for temp1).

I'll improve the code by reading how many Environmental sensors are in the battery and then only populate temp1 or temp2 based on this, while adding checks to ensure we exit if any temp value read returns a false.

Something along this lines:

 def read_temp_data(self):
        #OLD temp1 = self.read_serial_data_renogy(self.command_bms_temp1)
        #OLD temp2 = self.read_serial_data_renogy(self.command_bms_temp2)
        
        # Check to see how many Enviromental Temp Sensors this battery has, it may have none.
        num_env_temps = self.read_serial_data_renogy(self.command_env_temp_count)
        logger.info('Number of Enviromental Sensors = %s', num_env_temps)

        if (num_env_temps == 0):
            return False

        if (num_env_temps == 1):
            temp1 = self.read_serial_data_renogy(self.command_env_temp1)

        if temp1 is False:
            return False
        else:
            self.temp1 = unpack(">H", temp1)[0] / 10
            logger.info('temp1 = %s °C', temp1)

        if (num_env_temps == 2):
            temp2 = self.read_serial_data_renogy(self.command_env_temp2)

        if temp2 is False:
            return False
        else:
            self.temp2 = unpack(">H", temp2)[0] / 10
            logger.info('temp2 = %s °C', temp2)

        return True

Louisvdw, 2023-01-09:
Thanks for the testing and update. I'll update it to the driver as soon as I can.
For now the driver assume one serial connection per BMS. Perhaps we can add some settings in future.

mr-manuel, 2023-04-20:
Your fix will be included in the next release.

Did you connect multiple BMS to one RS485 adapter? https://github.com/Louisvdw/dbus-serialbattery/issues/142

PR #551 (2023-04-25): Some fixes
* Changed: Fix System Switch in IO page
* Changed: Fix #239 by @TimGFoley
* Changed: Fix #311
* Changed: Run code analyse only on code change
* Changed: Separate Time-To-Go and Time-To-SoC activation
* Changed: fix UnicodeDecodeError https://github.com/Louisvdw/dbus-serialbattery/discussions/485#discussioncomment-5694781
commit 520eedc9 (2023-04-25): fix #239
by @TimGFoley
commit d8a5d643 (2023-04-25): Some fixes

* Changed: Fix System Switch in IO page
* Changed: Fix #239 by @TimGFoley
* Changed: Fix #311
* Changed: Run code analyse only on code change
* Changed: Separate Time-To-Go and Time-To-SoC activation
* Changed: fix UnicodeDecodeError https://github.com/Louisvdw/dbus-serialbattery/discussions/485#discussioncomment-5694781