Louisvdw/dbus-serialbattery #247: Editing utils.py does not change CCL accordingly.

MikeEcho0815, 2022-10-10:
**Describe the bug**
Editing the utils.py (MAX_BATTERY_CURRENT) leads to wrong Parameter in Cerbo GX CCL

**To Reproduce**
Steps to reproduce the behavior:
1. Editing the utils.py (MAX_BATTERY_CURRENT)
2. Check Cerbo GX Parameters > CCL

**Expected behavior**
I would expect the same value as edited in the utils.py

**Screenshots**
[image]
[image]

**VenusOS (please complete the following information):**
 - Device type: Cerbo GX
 - Firmware Version : v2.91

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JKBMX 150A
 - Cells: 16
 - Interface: USB-RS485

Louisvdw, 2022-10-10:
See https://github.com/Louisvdw/dbus-serialbattery/wiki/Features#charge-current-control-management
Your SOC % are between 95% - 98% => 1/4 Max charge   (105A/4 = 26,25A)

MikeEcho0815, 2022-10-10:
Thank you for pointing me in the direction!

closed 2022-10-10 by Louisvdw