Louisvdw/dbus-serialbattery #249: add monitoring for individual cell over/under voltage levels

arrow1800, 2022-10-10:
Yes, you can already see the highest and lowest cell voltage. however, those values aren't used to change the charging algorithm when necessary.  more info `[here](https://github.com/Louisvdw/dbus-serialbattery/discussions/246)`

One of the clear benefits of using a smart BMS (one that is able to talk to victron equipment) could be monitoring the voltage of not only the total battery but also the individual cell voltages to prevent **individual cell overload or underload conditions**.

And yes, your battery bank should offcourse be balanced out before you start. however in real life a battery can also get out of balance through other ways, for example; long time no full charge (winter) or excessive discharge and charge cycles

**Describe the solution you'd like**
The charge algorithm should probably lower the current when highest cell voltage goes over the max (configured in utils.py) OR when lowest cell voltage goes under the min (configured in utils.py)

in this image; max cell voltage was set at 3.40v
[image]

i am open to suggestions on howto implement is, and when pointed in the right direction i can give it try myself as well!

pkali1, 2022-10-15:
Hello, I totally agree, that the driver should raise alarm in cell imbalance conditions. I just finished small home diy installation and this was rainy day today. The pack was top balanced, but on lower end one cell is much lower and the serial battery driver was totally silent. Regards and thanks for your effort.

coca7, 2022-10-15:
What do you have your cell voltages set at?  I had a problem with imbalances until I used the default voltages in discussion with Louis. Once I did that I have 0 problems with cell imbalances.  I do have some imbalances between parallel batteries, but I'm hopping the multiple battery function will help with that.

pkali1, 2022-10-16:
Hi, these are settings in utils.py
# Cell min/max voltages - used with the cell count to get the min/max battery voltage
MIN_CELL_VOLTAGE = 2.9
MAX_CELL_VOLTAGE = 3.45
FLOAT_CELL_VOLTAGE = 3.35
MAX_VOLTAGE_TIME_SEC = 15*60
SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 90

The voltage was lower then 2.7V and there was no alarm. I will set alarm for lowest cell in node-red for now,
I hope Louis will try manage that and raise alarms for individual cells.

Louisvdw, 2022-10-16:
If I understand it correct, this ticket is about the charge discharge limits that should also be lowered when looking at cell values.

The low cell voltage alarm only triggers when the BMS raise a low voltage. 
@pkali1 what BMS to you have and what settings are you using inside that BMS?

pkali1, 2022-10-17:
I use DALY 16S 250A with 16 CALB 200Ah Lifepo4
Parameters from .cfg file saved from BMS app. There are two levels of voltages as I see- warning and cutoff for over voltage, but I see only one (cutoff) for single cell under voltage and no imbalance alarm. I use node-red battery monitor node to switch off inverter, when single cell is below 2.9V as workaround. But it would be cleaner way to disable system ATD (allow to discharge). I thought, that this would be done by BMS plus its driver. Is it possible to raise imbalance alarm in driver itself?

Rated capacity: 200
Monoblock rated voltage: 3,2
Equilibrium opening voltage: 3,3
Equilibrium opening differential pressure: 0,05
Total charge amperage: 293
Total discharging amperage: 290
Short-circuit protection current: 2,500
Current sampling resistance: 0,1
Number of acquisition boards: 1
Number of individual boards: 16
Number of acquisition boards 2: 0
Number of acquisition boards 3: 0
Number of acquisition boards 1 temperature: 1
Number of acquisition boards 2 temperature: 0
Number of acquisition board 3 temperatures: 0
Battery operation mode: Long press to switch on/off
Sleep time (S): 65535
Zero drift (A): 0,2
Battery production date - year: 2022
Battery production date - month: 10
Battery production date - day: 11
Battery type: Ternary
Firmware index number: 20220816
Battery code: 20220816                        
SOC.
Single cell overvoltage level 1: 3,55
Monomer overvoltage class II: 3,65
Monoblock undervoltage class I: 2,6
Monoblock undervoltage level 2: 2,5
Total overpressure class I: 56,8
Total overpressure class II: 58,4
Total undervoltage class I: 41,6
Total undervoltage class II: 40
Discharge overcurrent class I: 300
Discharge overcurrent class II: 375
Charging overcurrent class I: 300
Charge overcurrent class II: 375
Excessive differential pressure class I: 0,5
Excessive differential pressure class II: 0,8
Excessive temperature difference primary: 10
Excessive temperature difference class II: 15
Excessive SOC level 1: 100
SOC too high level 2: 102
SOC too low level 1: 20
SOC too low level 2: 10
Charge overtemperature level 1: 55
SOC Over Temperature Level 2: 65
Undercharge Level 1: 7
Undercharged Level 2: 5
Discharge overtemperature level 1: 65
Discharge overtemperature level 2: 70
Discharge under-temperature level 1: -5
Discharge under-temperature level 2: -8
Active equalisation current: none
Number of active equalisation strings: none
Active equalisation turn-on voltage: 3,3V
Active equalisation turn-on differential voltage: 0,05V
Sleep time: 65535S
Single over-voltage: 3,65V
Single undervoltage: 2,5V
Over temperature: 70°C
Over temperature: -8°C

Louisvdw, 2022-10-17:
It's possible to raise alarms in the driver. But up to now the alarms come from the BMS itself and is just relayed.
What you want to do you do not need alarms, but this feature request could do it (just not with alarms)

arrow1800, 2022-10-22:
preventing alarms is what its all about right :)

arrow1800, 2023-01-11:
maybe using a virtual battery driver is the solution: [https://github.com/arrow1800/Victron-VirtualBattery](https://github.com/arrow1800/Victron-VirtualBattery). then you have full control

pkali1, 2023-01-12:
Hi, thans for this. But I think Louis has done the control of charge current by single cell voltage in latest version of battery driver. (see    https://github.com/Louisvdw/dbus-serialbattery/blob/master/etc/dbus-serialbattery/utils.py   )For me is everything working fine (only the bug in node.red - identification USB (disconnection) on restart is persistant). I will not update the Venus OS, because I made some changes to DVCC module, which I do not want to repeate and destroy functioning system. Regards and thanks to Louise for update also.

mr-manuel, 2023-04-14:
This is realized with the next version of the driver.

If you are searching for a virtual battery driver that you can feed over MQTT you can try this [venus-os_dbus-mqtt-battery](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery).

closed 2023-04-14 by mr-manuel