Louisvdw/dbus-serialbattery #253: Self.port

derhier, 2022-10-12:
Hey

I fresh installed dbus-serialbattery.

Getting syntax errors and not sure how to fix this issue. We have been at it for a while:

pastebin.com/HcbvHskj

root@raspberrypi2:~# tail -f /data/log/dbus-serialbattery.ttyUSB0/current @400000006345fda72b47eee4 Traceback (most recent call last): @400000006345fda72b49e6cc File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 18, in <module> @400000006345fda72b49fe3c from dbushelper import DbusHelper @400000006345fda72b4a830c File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 12, in <module> @400000006345fda72b4b03f4 import battery @400000006345fda72b4b84dc File "/opt/victronenergy/dbus-serialbattery/battery.py", line 339 @400000006345fda72b4c2504 logger.info(f'Battery connected to dbus from {self.port}') @400000006345fda72b4c388c ^ @400000006345fda72b4f5d3c SyntaxError: invalid syntax @400000006345fdc601743b04 *** CCGX booted (0) *** root@raspberrypi2:~# tail -f /data/log/serial-starter/current @400000006345fddd3aa40044 INFO: Start service dbus-cgwacs.ttyUSB0 once @400000006345fddf074b5134 INFO: Start service mk2-dbus.ttyUSB1 @400000006345ff0f063cfa2c INFO: Create daemontools service mk2-dbus.ttyUSB0 @400000006345ff150852af14 INFO: Start service mk2-dbus.ttyUSB0 @400000006345ff340401a85c *** CCGX booted (0) *** @400000006345ff34365969fc serstart starting @400000006345ff34383f9304 INFO: loading config file /etc/venus/serial-starter.conf @400000006345ff351a7ca5b4 INFO: loading config file /data/conf/serial-starter.d @400000006345ff450ff11f94 INFO: Create daemontools service mk2-dbus.ttyUSB0 @400000006345ff4b1350446c INFO: Start service mk2-dbus.ttyUSB0

Louisvdw, 2022-10-12:
What firmware version is running on your GX device?
The latest versions require VenusOS 2.80 firmware or newer
https://github.com/Louisvdw/dbus-serialbattery/wiki/How-to-install

derhier, 2022-10-13:
> 

OK thanks. Good to know - didnt find that info on the install page and my german version said 2.66 is the newest version for venus.

closed 2022-10-13 by derhier

Louisvdw, 2022-10-14:
I have seen that if you use the Raspberry Pi as a GX device then it cannot update from the older versions of VenusOS. You have to manually install the new firmware. From the new VenusOS version it seems to work better.