Louisvdw/dbus-serialbattery #254: ECS BMS not found

trinler007, 2022-10-12:
Thanks for the implementation of the ECS System! 

Unfortunately, the ECS components aren't found in my case.

/var/log/dbus-serialbattery.ttyUSB0/current :
@4000000063467fe732b73914 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
@40000000634680091993f74c INFO:SerialBattery:Starting dbus-serialbattery
@400000006346800919a1e99c INFO:SerialBattery:dbus-serialbattery v0.14~1
@400000006346800919f7646c INFO:SerialBattery:Testing LltJbd
@40000000634680092ad9a86c ERROR:SerialBattery:>>> ERROR: No reply - returning
[...]
@400000006346800e3aefb32c INFO:SerialBattery:Testing Renogy
@400000006346800f107bbfdc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006346800f108a9c8c INFO:SerialBattery:Testing Ecs
@400000006346800f30a12a54 INFO:SerialBattery:Testing LltJbd
@4000000063468010070bf28c ERROR:SerialBattery:>>> ERROR: No reply - returning
[...]
@4000000063468012116defdc INFO:SerialBattery:Testing Renogy
@400000006346801221fb360c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063468012220ced34 INFO:SerialBattery:Testing Ecs
@4000000063468013069b7d7c ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
---

R485 cable is the original Victron cable

root@einstein:~# lsusb
Bus 004 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 010: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 005 Device 003: ID 0bda:b720 Realtek Semiconductor Corp.
Bus 005 Device 002: ID 0424:2422 Standard Microsystems Corp.
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Device: Cerbo GX
OS Version : 2.91
ECS Greenview Version 1.10.02
8 cells with LiPro1-6 Active (v1)
RS485 is connected to the slave port on ECS Greenview. The modbus ID Greenview is 50, LiPro1-6 Active IDs 1-8

The Greenview seems to see RX/TX on the bus (is shown by a symbol on the display)

Which additional information can I provide, or what can I do solve the issue?

Louisvdw, 2022-10-12:
The ECS devices need a little bit more manual set up, and that is not well documented yet. I'll update the documentation for this soon.

You need to edit utils.py around line 88 and set the following values to match your GreenMeter and LiPro setup with the Modbus address your set each device with. It works best if your LiPro devices have sequential address. 

> nano /data/etc/dbus-serialbattery/utils.py

> GREENMETER_ADDRESS = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS = 4
LIPRO_CELL_COUNT = 8

Reboot your GX after you have made these changes.

trinler007, 2022-10-12:
Thanks! I've changed the settings and rebooted, but unfortunately no change.

I don't have a green meter. Is this mandatory? My setup consists of the Greenview (ID 50) and 8 LiPro1-6 Active. The currents are measured by the Greenview with two HALL sensors (IN/OUT).

Louisvdw, 2022-10-12:
Currently you need a GreenMeter. I'll have to investigate if it would be possible to get all the required values from a GreenView to know if that would be possible in future.

trinler007, 2022-10-12:
OK, too bad. 

I'm pretty sure, you would also get all the values from the greenview, since I have already read them out for visualisation via modbus tcp. I'm attaching the register overview for you. Unfortunately, this is only available in German, but I'm happy to translate if necessary.

[MODBUS-greenView.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/9764322/MODBUS-greenView.pdf)

Louisvdw, 2022-10-12:
I'm looking at the docs now as well. It looks like the GreenView use different registers that the GreenMeter, so it will not be a case of just adding the device ID for it to work. I'll have to code it spesifically to work for the GreenView as well.

trinler007, 2022-10-12:
Thank you Louis!

The Greenmeter is quite new, which is why I assume that many ECS systems are built with the Greenview and the current sensors, like mine. If I can support you in any way with the implementation, please let me know.

mr-manuel, 2023-04-20:
Issue for GreenMeter https://github.com/Louisvdw/dbus-serialbattery/issues/28.