Louisvdw/dbus-serialbattery #268: Support for Polarium Gen 5 Batteries

cbbstewart, 2022-10-20:
Im looking for support of the Polarium Gen 5 Batteries. 

I can see from another thread someone has got it working. But im not sure what they added to the driver to make it work. 

Work was done by @arnoc-intuvision 

https://github.com/Louisvdw/dbus-serialbattery/issues/8#issuecomment-1164341379

Stewart

Louisvdw, 2022-10-20:
They have not shared their change yet

cbbstewart, 2022-10-21:
[POLARIUM Modbus Interface - Description (2).pdf](https://github.com/Louisvdw/dbus-serialbattery/files/9841008/POLARIUM.Modbus.Interface.-.Description.2.pdf) - i do have the modbus interface info now and it uses 115200 baud rate. I can talk to the battery using the polarium software.