Louisvdw/dbus-serialbattery #273: VenusOS 2.91 -  dbus-serialbattery v0.13 - JK-BMS RS485 not seen in Venus OS

ducloslaurent, 2022-10-25:
Hello Louis,

_**Bug :**_ 
impossible to see the JK BMS model JK-B2A24S20P in the Venus OS

_**Installation :**_ 
Venus OS : v2.91 mounted on a Rpi3B+
Additional content : Package Manager (ShutDownMonitor + RpiDisplaySetup + GuiMod installed)
Victron hardware : Multiplus 2 48/3000 (not a GX version)
-> DVCC activated
BMS : JK-B2A24S20P - SW version : V10.07 - Version V4.8.0
JK-BMS RS485 module plugged into the GPS port of the GPS + using a DSD TECH SH-U10L Câble USB vers RS485 (https://www.amazon.fr/gp/product/B08DXLJF8J/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

_**Dbus-Serialbattery installation :**_ 
made with SSH access using Putty (root access) : 
-> wget https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/installrelease.sh
   NO ISSUE
-> sh installrelease.sh
   2 message reporting "cp: omitting directory '/data/etc/dbus-serialbattery/qml'" and "cp: omitting directory '/data/etc/dbus-serialbattery/service'"
-> reboot
   NO ISSUE

_**Defect retest :**_ 
Rpi Reboot have been done 2 times
RS485/USB converter have been tested on all Rpi USB port and also on an USB extender port

Test made with VenusOS 2.92 -> same result, impossible to see the JK-BMS on the Rpi screen

Installation have already work (JK-BMS shown on the Rpi but I don't remember the Venus OS version using the same JK-BMS RS485 module and RS485/USB converter)

_**Printscreen :**_ 

Dbus-serialbattery installation
[image]

DVCC activation
[image]

VICTRON setup
[image]

Thanks for your support on this issue and great work for the JK-BMS adaptation on Venus OS

Louisvdw, 2022-10-26:
Can you post the log files here please
https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot

ducloslaurent, 2022-10-26:
Hello Louis,

1st log file -> 1st message error
Super user is activated (see 2nd screen shot)
I'm connected through Putty as root and the root entry seams to be OK

[image]

[image]

Marvo2011, 2022-10-28:
@ducloslaurent You have to use `cat`

```cat /data/log/serial-starter/current```

And please hide your IPs for safety reasons...

Louisvdw, 2022-10-28:
@ducloslaurent the problem is you cannot run the file, you need to view it's contents. So either by using cat like @Marvo2011 mentioned, or if you look at the bottom of the troubleshoot you will see the examples for tail 
> tail -f /data/log/serial-starter/current
tail -f /data/log/dbus-serialbattery.ttyUSB0/current

The IP range 10.*.*.* is an private IP range and so no traffic on that will be routed to the internet. So it is save to show
https://en.wikipedia.org/wiki/Private_network

Marvo2011, 2022-10-28:
@Louisvdw I' know 10... is private, I mean the 84... with tunnel port. https://www.victronenergy.com/media/pg/Cerbo_GX/en/faq.html#UUID-9977528a-a1d7-c785-64af-2ab39d915b25
Seams a Victron server IP should be save...

Louisvdw, 2022-10-28:
Ah! I missed that in the picture. You are correct

karrycrasto2008, 2022-10-31:
I had to use combination of following 2 adapters to make it work with JK BMS and Venus OS
USB to RS485([https://robu.in/product/usb-rs485-ttl-serial-converter-adapter-ft232/)=>](https://robu.in/product/usb-rs485-ttl-serial-converter-adapter-ft232/) RS485 to TTL(https://robu.in/product/ttl-to-rs485-power-supply-converter-board-3-3v-5v-hardware-auto-control-module/)

[image]

ducloslaurent, 2022-11-04:
@karrycrasto2008 thanks for your answer, but I can not see how both adapter are plugged to the JK-BMS as only 1 RS485 output on the JK-BMS.
No more use of the JK-BMS RS485 adapter?

[image]

WaldemarFech, 2022-11-15:
If you have the JK-BMS RS485-Adapter, then it already converts TTL (The Rx and Tx Pins in the "GPS" Port) to RS485.
Then you only need a RS485 -> USB Adapter. In best case with a FTDI-Chip.

Please print, as @Louisvdw said, the output of the following command:
`tail -f /data/log/serial-starter/current`

It is possible, that you have an other Chip on your converter - then maybe we need to change a victron file, to use modbus on your chip.

But you can also try to change A and B in your Converter - sometimes it's printed wrong on the china converter.

The Jumpers in the last picture (under the title USB to Rpi) should be all set to supply the rs485 driver on that board. But you don't have to connect anything else than A and B (even not GND)

karrycrasto2008, 2022-11-16:
JK-BMS RS485 adapter was not available for me without a very long wait for it to be imported from China so i had to use the adapter i have mentioned above to convert TTL (The Rx and Tx Pins in the "GPS" Port) to RS485. It has been working fine since last 2 months now. Note that the adapter has max input voltage of 33 volts so you may need a voltage converter to reduce it if your battery bank voltage is higher than that.

ducloslaurent, 2022-11-16:
> If you have the JK-BMS RS485-Adapter, then it already converts TTL (The Rx and Tx Pins in the "GPS" Port) to RS485. Then you only need a RS485 -> USB Adapter. In best case with a FTDI-Chip.
>
> Please [...]

@WaldemarFech , yes I have the JK-BMS adapter
Below the print screen for the commande "tail -f /data/log/serial-starter/current"

[image]

But you can also try to change A and B in your Converter - sometimes it's printed wrong on the china converter.
-> OK I will try it

The Jumpers in the last picture (under the title USB to Rpi) should be all set to supply the rs485 driver on that board. But you don't have to connect anything else than A and B (even not GND)
-> OK

Louisvdw, 2022-11-16:
The dbus-serialbattery driver is running as per the log file you posted. Now we need to see what happens in the dbus-serialbattery.ttyUSB1/current log file, but my guess would be that it does not get a reply. That would indicate problems with the wiring or connections.

ducloslaurent, 2022-11-19:
@Louisvdw see below the result for the command tail -f /data/log/dbus-serialbattery.ttyUSB1/current

[image]

-> wiring problem?

WaldemarFech, 2022-11-19:
Maybe yes. There ist absolute no communication from the BMS.

Would it be an option for you to make a video of your wiring and upload a short video on YouTube or so? Then we could see where you have plugged in the cables and which color and so on.

Louisvdw, 2022-11-19:
@ducloslaurent if you are still using the adapter that you listed in the opening post I think that might be the issue. I'm not sure the SILICON LABS CP2102 chip is part of the drivers in the VenusOS. There is no reply from a BMS.

Try using something like 
https://www.amazon.fr/RS485-C%C3%A2ble-adaptateur-convertisseur-RS485-avec-Usb-rs485-we-1800-bt/dp/B0757N8WPG
https://www.amazon.fr/Convertisseur-Industriel-Adaptateur-Anti-Foudre-Protection/dp/B0B87D9LNC

ducloslaurent, 2022-11-22:
@Louisvdw second link product ordered and received, I will try to make the change and kepp you informed.

ducloslaurent, 2022-11-23:
Hello @Louisvdw , some progress, wiring made with 2nd proposed converter, yellow wire on A pad / white wire on B pad and TXD led is flashing (Power led red continuously)
See below both print screen for command : 

tail -f /data/log/serial-starter/current

[image]

tail -f /data/log/dbus-serialbattery.ttyUSB1/current

[image]

But no BMS signal on the Venus OS

[image]

Louisvdw, 2022-11-23:
@ducloslaurent you will see the serial-starter log that the port is for ttyACM0 which mean you need to get that log file instead of ttyUSB1 for the serialbattery log.

ducloslaurent, 2022-11-24:
@Louisvdw , yes sorry, not seen that the test have not been done on the good USB
Below th test on the ttyUSB1

[image]

Louisvdw, 2022-11-24:
The driver is still not receiving a reply. Can you send us detail pictures of your where your adapter plugs into your GX device, each piece until plugged into the BMS on the GPS port. I need to see each connection please.

ducloslaurent, 2022-11-24:
@Louisvdw 
See below complete wiring from BMS to RPi 3B+ used for Venus OS emulation :

BMS plugged (JK-BMS version JK-B2A24S20P RS485 - According to JK-BMS, the RS485 is on the GPS port)
[image]

[image]

JK-BMS RS-485 adaptor 
[image]

Connection between JK-BMS RS-485 adaptor and RS485 convertor (only yellow and white wire plugged, not the black wire on GND)
[image]

Connection from RS485 convertor to RPi 3B+
[image]

WaldemarFech, 2022-11-24:
> TXD led is flashing

there must be something wrong with the wiring or the jk converter / connector.

If your tx pin or your adapter is blinking, then the serial battery driver is sending messages and is waiting for answers.
But if A and B is over cross, then it don't work.
If the jk rs485/TTL converter is broken or has no good connection in the GPS port, then it don't works, too.

If you have a second USB/rs485 converter, then you can plug it in your computer and run a program like hterm.
choose the com-port, choose 115200 baud, click connect, make a selection in HEX and let it make a new line after 200ms or so. after some receptions you can disconnect. there should be a message in hex (i have to look in the code to tell the exactly message) which asks the different BMSs if they are there.

Then you would know, that both of your usb rs485 converters are working fine and the problem is somewhere in the jk converter or the wiring to the GPS port.

try to crossover A and B and replug the converters.

ducloslaurent, 2022-11-24:
Changing the JK-BMS RS485 adaptor and crossover A & B -> it is alive and visible on the Venus OS

Many, many thanks to @Louisvdw and @WaldemarFech 

[image]

closed 2022-11-24 by Louisvdw