Louisvdw/dbus-serialbattery #279: JK BMS looses connection after battery high voltage alarm.

karrycrasto2008, 2022-10-31:
JK BMS looses connection after battery high voltage alarm. 

[image]

**VenusOS
 - Device type:  Raspberry Pi
 - Firmware Version 292

**Battery/BMS
 - BMS/Battery type: JK
 - Cells: 8 , 150 AH
 - Interface: USB to RS485(https://robu.in/product/usb-rs485-ttl-serial-converter-adapter-ft232/)=> RS485 to TTL(https://robu.in/product/ttl-to-rs485-power-supply-converter-board-3-3v-5v-hardware-auto-control-module/)
 - Active balance is on. 
This has been happening from beginning from setup, I think it happens when the cells reach max voltage and the BMS stops charging the battery.
It resolves once Venus OS is rebooted.

Louisvdw, 2022-10-31:
The JKBMS reboots when it gets a high voltage alarm and then the connection is lost. I have seen that happen. 
Check what the min/max cell voltages are when that happens. (VRM Advance section graphs) 
Most likely some of your cells are imbalanced which raise the alarm.

karrycrasto2008, 2022-10-31:
This is what it looks like in VRM for min/max cells
[image]

Louisvdw, 2022-10-31:
Do you see anything in the log files? https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot

closed 2023-04-20 by mr-manuel