Louisvdw/dbus-serialbattery #282: only 2 of 3 JK-BMS are shown in venus os

FrankSausR, 2022-10-31:
I can't connect to a JK-BMS. I've tried it with JK adapter, it works with two. I tried it via USB-TTL adapter with two it works. i can´t connect to the oldes one.

root@raspberrypi4:/data/log/dbus-serialbattery.ttyUSB1# tail -f /data/log/dbus-serialbattery.ttyUSB1/current 
@40000000635ff3c0222a548c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000635ff3c02233da0c INFO:SerialBattery:Testing Lifepower
@40000000635ff3c032629904 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000635ff3c03262ac8c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000635ff3c03262b844 INFO:SerialBattery:Testing Renogy
@40000000635ff3c106a5c64c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000635ff3c106af4bcc INFO:SerialBattery:Testing Renogy
@40000000635ff3c116d4527c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000635ff3c116dec644 INFO:SerialBattery:Testing Ecs
@40000000635ff3c1383388fc ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB1
@40000000635ff3ec2efd1f14 INFO:SerialBattery:Starting dbus-serialbattery
@40000000635ff3ec2f048984 INFO:SerialBattery:dbus-serialbattery v0.13
@40000000635ff3ec2f1830c4 INFO:SerialBattery:Testing LltJbd
@40000000635ff3ed03f1c5f4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000635ff3ed074d0a9c INFO:SerialBattery:Testing Ant
@40000000635ff3ed17da412c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000635ff3ed1b409dac ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000635ff3ed1b450e64 INFO:SerialBattery:Testing Daly
@40000000635ff3ed2b8ae384 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000635ff3ed2eee752c INFO:SerialBattery:Testing Daly
@40000000635ff3ee0400a2a4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000635ff3ee076bcd9c INFO:SerialBattery:Testing Jkbms
@40000000635ff3ee121a8774 INFO:SerialBattery:Testing Sinowealth
@40000000635ff3ee22f005ec ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000635ff3ee265ae6ac INFO:SerialBattery:Testing Lifepower
@40000000635ff3ee36b0b1bc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000635ff3ee3a10efd4 ERROR:SerialBattery:>>> ERROR: Incorrect Data

root@raspberrypi4:/data/log/dbus-serialbattery.ttyUSB1# tail -f /data/log/serial-starter/current 
@40000000635ff317065d054c INFO: Start service vedirect-interface.ttyUSB1 once
@40000000635ff31c1871ca24 INFO: Start service dbus-serialbattery.ttyUSB1 once
@40000000635ff32733a181a4 INFO: Start service dbus-cgwacs.ttyUSB1 once
@40000000635ff34125ffaf1c INFO: Start service gps-dbus.ttyUSB1 once

ftp://admin@192.168.1.2:21/homes/admin/Camera Uploads/Camera/Camera/1 Motorola/Screenshots/Screenshot_20221030-152758.png
[image]
[image]
[image]

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi 4
 - Firmware Version 2.92
 - dbus-serialbattery 0.13

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JK-BMS 2xBMS JK-B1A24S15P 1xBMS JK-B2A24S15P
 - Cells: 12
 - Interface: 2x USB-RS485 and JK RS485 Adapter, 1x USB-TTL

mrgithub, 2022-11-01:
Can you show a picture of how you've wired two jk bms, adapter and usb together along with a line diagram please?

Louisvdw, 2022-11-01:
Are you saying that it work when connected to the JK-B1A24S15P , but it fails when connected to JK-B2A24S15P?

FrankSausR, 2022-11-01:
> Are you saying that it work when connected to the JK-B1A24S15P , but it fails when connected to JK-B2A24S15P?

it work with the  JK-B1A24S15P Hardware Ver. 8x und wiht the  JK-B2A24S15P Hardware Ver. 8x.
But not with JK-B1A24S15P Hardware Ver. 6x

FrankSausR, 2022-11-01:
> Can you show a picture of how you've wired all your jk bms, battery, rs485 adapter and usb adapter together along with a line diagram please?
[image]

is this okay?

mr-manuel, 2023-04-20:
Probably the oldest hardware revision is using another protocol structure. From what year is the BMS with the 6.x hardware?

closed 2023-04-27 by mr-manuel