Louisvdw/dbus-serialbattery #284: JKBMS disconnects

tdeckers, 2022-11-05:
**Describe the bug**
JK-B2A20S20P initially connects fine after a restart of Venus OS.  I can see it working briefly, but after a short time, it fails with following error in the logs:

```
@4000000063661d0333491104 *** CCGX booted (0) ***
@4000000063661d0f34cd150c INFO:SerialBattery:Starting dbus-serialbattery
@4000000063661d0f34d4e8f4 INFO:SerialBattery:dbus-serialbattery v0.13
@4000000063661d0f34f43c7c INFO:SerialBattery:Testing LltJbd
@4000000063661d100a43097c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063661d100acdbd4c INFO:SerialBattery:Testing Ant
@4000000063661d101b9447a4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063661d101c209984 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000063661d101c27e0cc INFO:SerialBattery:Testing Daly
@4000000063661d102cd74d2c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063661d102d574eb4 INFO:SerialBattery:Testing Daly
@4000000063661d1102529cb4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000063661d1102d5076c INFO:SerialBattery:Testing Jkbms
@4000000063661d110c4f2d2c INFO:SerialBattery:Connection established to Jkbms
@4000000063661d110c58f514 INFO:SerialBattery:Battery connected to dbus from /dev/ttyUSB1
@4000000063661d110c5ee884 INFO:SerialBattery:=== Settings ===
@4000000063661d110c689ce4 INFO:SerialBattery:> Connection voltage 53.33V | current -0.0A | SOC 88%
@4000000063661d110c6e849c INFO:SerialBattery:> Cell count 16 | cells populated 0
@4000000063661d110c759d04 INFO:SerialBattery:> CCL Charge NoneA | DCL Discharge NoneA
@4000000063661d110c7cb184 INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.65V | MAX_CELL_VOLTAGE 3.5V
@4000000063661d110e8e7a84 INFO:SerialBattery:DeviceInstance = 1
@4000000063661d110e969c8c INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
@4000000063661e40327ecfac Traceback (most recent call last):
@4000000063661e40327f0274   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 163, in publish_battery
@4000000063661e40327f1214     success = self.battery.refresh_data()
@4000000063661e40327f19e4   File "/opt/victronenergy/dbus-serialbattery/jkbms.py", line 53, in refresh_data
@4000000063661e40327f2984     result = self.read_status_data()
@4000000063661e40327f2d6c   File "/opt/victronenergy/dbus-serialbattery/jkbms.py", line 66, in read_status_data
@4000000063661e403281cd4c     status_data = self.read_serial_data_jkbms(self.command_status)
@4000000063661e403281dcec   File "/opt/victronenergy/dbus-serialbattery/jkbms.py", line 150, in read_serial_data_jkbms
@4000000063661e403281ec8c     data = read_serial_data(command, self.port, self.baud_rate, self.LENGTH_POS, self.LENGTH_CHECK,None, self.LENGTH_SIZE)
@4000000063661e4032820014   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 84, in read_serial_data
@4000000063661e40328261bc     return read_serialport_data(ser, command, length_pos, length_check, length_fixed, length_size)
@4000000063661e4032827544   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 125, in read_serialport_data
@4000000063661e40328280fc     toread = ser.inWaiting()
@4000000063661e40328288cc   File "/usr/lib/python3.8/site-packages/serial/serialutil.py", line 590, in inWaiting
@4000000063661e4032832124     return self.in_waiting
@4000000063661e4032832cdc   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 467, in in_waiting
@4000000063661e4032833894     s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
@4000000063661e403283444c OSError: [Errno 5] Input/output error
@4000000063661e403b6475cc /opt/victronenergy/serial-starter/run-service.sh: line 15: kill: (1515) - No such process
```

**To Reproduce**
Install dbus-serial as documented.  After a short while JKBMS just disconnects and the Multiplus II goes in passthrough.

**Expected behavior**
Expect the BMS monitor to continue operating.

**Screenshots**
See log above.
[image]

**VenusOS (please complete the following information):**
 - Device type: Multiplus II 48/5000/70-50
 - Firmware Version v2.92

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JK-BMS -- JK-B2A20S20P
 - Cells: 16
 - Interface: TTL->RS485 then RS485->USB

**Additional context**
Add any other context about the problem here.

WaldemarFech, 2022-11-15:
Hi.
I've got no problems. See my version (at the end of this post).
Try to use an other rs485-USB-Converter ... it seems like, that a standard python function is crashing, not the serial-battery-driver, as you can see in your log:

```
@4000000063661e4032832cdc   File "/usr/**lib/python3.8/site-packages/serial/serialposix.py**", line 467, in in_waiting
@4000000063661e4032833894     s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
```

```
root@einstein:~# python -V
Python 3.8.13
```
[image]

tdeckers, 2022-11-16:
Thanks @WaldemarFech. I'll give that a try.  Which adapter are you using?

Louisvdw, 2022-11-16:
Most issues come from loose connections or wires that does not make good contact. So check those.
The adapters based on FTDI or CH340 both work well. Some cheaper adapters could give issues (oven on those chips) so swopping it out for an alternative manytimes solves it as well.

peregrines2, 2022-11-17:
Having the same problem. Using 3 JK BMS with 3 USB TTL Azdelivery adapters. Everything is exactly the same for all three packs/banks. But one of them is disconnecting some minutes after staring Venus OS.
It is still visible via Bluetooth and everything but the connection works perfect.
Very strange!

WaldemarFech, 2022-11-17:
> Which adapter are you using?
I had some of these flying around.
[image]

> But one of them is disconnecting some minutes after staring Venus OS.
Try to change over cross the USB-Converter between one, that is working fine. if the disconnects travel to the new battery, then it's the hardware. when it's still at the same battery: change ttl/rs485 converter between the batteries.

i'm curious, so keep us updated :D

peregrines2, 2022-11-17:
> I had some of these flying around. [image]
>
>
> Try to change over cross the USB-Converter between one, that is working fine. if the disconnects travel to the new battery, then it's the hardware. when [...]

Thank You! That is a very good idea :)
Will try tomorrow and keep you informed about the results!

Louisvdw, 2022-11-18:
If it is not the converter it would be most likely a connection that is not making good contact. Swopping the converter adapters means you will be redoing the wire connections on both which many times fix the issue 😀

WaldemarFech, 2022-11-18:
> If it is not the converter it would be most likely a connection that is not making good contact. Swopping the converter adapters means you will be redoing the wire connections on both which many times fix the issue 😀 

yes, if there would be no problem anymore, then it was the bad connect. but if it still disconnects, then it should be possible possible to locate the faulty device :)

peregrines2, 2022-11-18:
Before beeing able to check again on the cabeling I discovered something else. 
The dropouts only happen, when one of my JK BMS is selected as "Batterymaster". 
As soon as I select Multiplus as the Battery Controlling Instance there will be no more dropouts.
I tested this 3 times now (see Grafana graphs). Whenever I select Multiplus the system seems to stay stable. 

[image]
[image]
[image]
[image]

WaldemarFech, 2022-11-18:
Suggestion:
Use tail like this:
`tail -f /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal`

with ` | tai64nlocal` to get a readable timestamp to compare better with grafana etc.

arndtw69, 2022-12-29:
I have also problems with that disconnection. With no no there seems to be no issues. Now i connected the multiplus to the grid to and the battery is used to compensate grid usage.

I use this adapter with the jkbms
https://www.amazon.de/DSD-TECH-SH-U09C3-Isolierter-Adapter/dp/B07TS3GPQ1

2022-12-29 15:59:25.961458500 INFO:SerialBattery:Testing Jkbms
2022-12-29 15:59:26.107946500 INFO:SerialBattery:Connection established to Jkbms
2022-12-29 15:59:26.108714500 INFO:SerialBattery:Battery connected to dbus from /dev/ttyUSB0
2022-12-29 15:59:26.109208500 INFO:SerialBattery:=== Settings ===
2022-12-29 15:59:26.109822500 INFO:SerialBattery:> Connection voltage 53.18V | current -0.0A | SOC 71%
2022-12-29 15:59:26.110292500 INFO:SerialBattery:> Cell count 16 | cells populated 0
2022-12-29 15:59:26.110857500 INFO:SerialBattery:> CCL Charge NoneA | DCL Discharge NoneA
2022-12-29 15:59:26.111380500 INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.45V
2022-12-29 15:59:26.175296500 INFO:SerialBattery:DeviceInstance = 1
2022-12-29 15:59:26.176087500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2022-12-29 15:59:30.288303500 Traceback (most recent call last):
2022-12-29 15:59:30.288317500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 163, in publish_battery
2022-12-29 15:59:30.288320500     success = self.battery.refresh_data()
2022-12-29 15:59:30.288323500   File "/opt/victronenergy/dbus-serialbattery/jkbms.py", line 53, in refresh_data
2022-12-29 15:59:30.288327500     result = self.read_status_data()
2022-12-29 15:59:30.288329500   File "/opt/victronenergy/dbus-serialbattery/jkbms.py", line 66, in read_status_data
2022-12-29 15:59:30.288544500     status_data = self.read_serial_data_jkbms(self.command_status)
2022-12-29 15:59:30.288550500   File "/opt/victronenergy/dbus-serialbattery/jkbms.py", line 150, in read_serial_data_jkbms
2022-12-29 15:59:30.288555500     data = read_serial_data(command, self.port, self.baud_rate, self.LENGTH_POS, self.LENGTH_CHECK,None, self.LENGTH_SIZE)
2022-12-29 15:59:30.288559500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 84, in read_serial_data
2022-12-29 15:59:30.288595500     return read_serialport_data(ser, command, length_pos, length_check, length_fixed, length_size)
2022-12-29 15:59:30.288599500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 125, in read_serialport_data
2022-12-29 15:59:30.288603500     toread = ser.inWaiting()
2022-12-29 15:59:30.288604500   File "/usr/lib/python3.8/site-packages/serial/serialutil.py", line 590, in inWaiting
2022-12-29 15:59:30.288661500     return self.in_waiting
2022-12-29 15:59:30.288663500   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 467, in in_waiting
2022-12-29 15:59:30.288667500     s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
2022-12-29 15:59:30.288669500 OSError: [Errno 5] Input/output error
2022-12-29 15:59:30.495420500 /opt/victronenergy/serial-starter/run-service.sh: line 15: kill: (28768) - No such process
2022-12-29 15:59:57.714235500 INFO:SerialBattery:Starting dbus-serialbattery

arndtw69, 2022-12-29:
I just tested further. With no load, when the multiplus is in charger Mode (switch in Pos II) there are no issues.
Celle voltages are readable an volating.

mr-manuel, 2023-04-14:
Does the issue still persist?

closed 2023-04-20 by mr-manuel