Louisvdw/dbus-serialbattery #296: alarm from bms in Venus OS that are not showing blue tooth and phone

MS123uk, 2022-11-22:
Have installed the driver for bms comm in Raspberry Pi 4b running  Venus OS  2.91 the from off grid garage video and your all work ok to start with until  a lot alarms  started showing in alarm log on portal but not showing on the phone app ?  e.g.  low temp charge - over current charge and so on you can clear them by rebooting  Raspberry  from the soft ware. having  unplug the comm lead and reset the solar controller  to default to stop (bms comm 67 alarm ) at the solar controller program will run ok  until  with out the bms comm being plug in all starts again to day alarms for the bms but its not plug in   

**VenusOS  2.91
 - Device type: [ Raspberry Pi 4b ]
 - Firmware Version [e.g. 262]

**Battery/BMS (please complete the following information):**
 - BMS/Battery type:  JBD -  liftpo4 
 - Cells: [e.g. 16] 8
 - Interface: [e.g. USB- to UART 

**Additional context**
Add any other context about the problem here.

Louisvdw, 2022-11-23:
Can you please upload the log file for the serialbattery on your bms port after you see this problem.
https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot

MS123uk, 2022-11-23:
Thank  for  prompt reply i had !!!! last night i have remove diy usb to rs232 leads and using the only Victron usb to VE direct  and the bms to usb uart plug in and so far  . if the problem show again  i will upload the logs  i have order  isolated 4 way usb port and  Ve direct  leads as show on off grid garage video link  finger cross

Louisvdw, 2022-11-23:
OK. I'll close the issue for now. You can come here and reopen it again if you need too

closed 2022-11-23 by Louisvdw

MS123uk, 2022-12-10:
Diy  usb to uart  leads  are not worth  cash saved only go  Victron usb to VE direct  or  isolated 4 way usb port and Ve direct leads as show on off grid garage  wills save you hole load grief

MS123uk, 2022-12-15:
isolated 4 way usb port and Ve direct leads as show on off grid garage try this to day  and same problem shown high battery temp- low temp - and so on driver for bms that`s not plug in