Louisvdw/dbus-serialbattery #297: CCL/DCL depending on cell-voltage does not turn off charging completely, still overvoltage alarm

chaerle, 2022-11-24:
When using CCL/DCL from https://github.com/Louisvdw/dbus-serialbattery/pull/293 with mode "WaldemarFech" the code correctly sets the charge current limitation (CCL) to zero, if the highest cell voltage gets over 3.55V. The Victron Multipli 48/3000 however does not honor it precisely, and still keeps charging at 0.4A to 0.6A, thus still driving the battery into over-voltage-alarm.

The charge current in the Multiplus seems to be always higher than the CCL, which is not a big problem as long the CCL ist above zero. But if the charge limit is zero it should definitively have zero current to the battery to avoid over charging. 

I also observed that the LED in on the Multiplus keeps in "Bulk" mode and does not change into the "absorbtion" or "tickle". I don't know if this is correct or if switching the mode could solve the problem, because the Multiplus might use different charge current limitation algorithms in different modes. 

The code should ensure that the charge current is really zero, either by switching off the charger, or changing the charging mode or altering the charge voltage. There might be also energy savings when the charger is completly off.

WaldemarFech, 2022-11-24:
hm.
so you see the CCL in the remote console set to zero in the serial battery - parameters menu?

first we have to find out weather it's a bug in our code or if it's a strange behavior of the internal Victron charge control.

i could change the code to print some debug messages in a log file, because I don't think, that i can reproduce it. here it's working fine.

if the code is not the problem, then we can think about workarounds. It's no problem to lower the max charge voltage with the CVL.

But first we have to figure out the reason for the continuous charge current.

Let me change a bit - see you here again.

chaerle, 2022-11-24:
> so you see the CCL in the remote console set to zero in the serial battery - parameters menu?>

Yes. Zero in CCL, but 0,6A still loading into the battery.

Louisvdw, 2022-11-25:
It won't work like you expect. There always will be a bit of current flowing into and out of the battery. I've done many tests on this question and talked to Victron people.

Firstly the chargers cannot accuratly limit/set the charge under 0.6A. Anything under that will be ignored except for the smallest inverters which might have a bit smaller value.
Secondly (and the more important reason) is that the ESS system use the battery as a buffer. The inverter has a ramp up/down cycle to adjust to different power use. It cannot magicly switch from 3000W production to 20W production in an instant. The MPPTs also do the same to supply PV power when the inverter requests it. During these cycles the battery is used for bufferring by either charging or discharging power for a very short time. For backup/UPS mode it does something similar but the battery use is even more because the grid is not there to help buffer as well.

There is only 1 option you have. You need to lower your battery full capacity so that there is a bit more buffer available. 
In DVCC override your battery charge voltage and lower it by a few millivolts until you don't see this happening.
It also happen more often if you have imballanced cells. Fixing imballanced cells will also help.

chaerle, 2022-11-25:
Any battery charger must be able to stop loading a battery which cannot be loaded any more. This can happen either because the battery is full or the BMS signals any condition that the battery must not be loaded any more (e.g. temperature, imbalanced cells, overvoltage etc). A battery charger which is not able to stop charging could destroy a battery.

Maybe there is a a measurement inaccuracy in the Multiplus if the charger is on (also in combination with the charging voltage), so that small currents cannot be controlled exactly, but then the code has to switch off the charger completely to archive zero charging or maybe lower the charging voltate dynamically. And this is definitely possible as you can manually switch the charger and inverter on and off via the Venus GUI. 

Also the default charging algorithm of the Multiplus seems to alter the charging voltage depending on the state of the battery. There are three charging state in the multiplus: Bulk, Absorption, Float and each uses different voltages as you can see from the documentation picture:

[image]

With the serial battery code activated, the Multiplus always keeps the state of "Bulk". Maybe the this should be changed?

Also I strongly disagree with your suggestion constantly changing the charing parameters manually to prevent overcharge and damage to the battery. This must be done by the code and it is possible. 

Waldemars solution it already much better then the original algorithm, because it handels the problem with imbalanced cells correctly in theory, which the original solution could not. The only problem is how to implement the complete stop of charging.

chaerle, 2022-11-25:
It seems to be a known problem, that the Victron cannot reliably stop charging by setting CCL to zero and keeping CVL at the original level. E.g. see here https://community.victronenergy.com/questions/136013/multiplus-48500-still-charging-when-ccl-from-bms-i.html

Someone from Victron Energy explains that with "supported" batteries the charge voltage will be lowered by the BMS.

_"Hi, sorry but no its not: we decided as a policy to not work with CCL=0.
Such change you’re suggesting should not be necessary for all supported batteries.
The way to stop charging is make the bms tell our system to work on a lower voltage.
Lowering the voltage stops the charge(current). When that works properly, there are no is."_
https://community.victronenergy.com/questions/150732/multiplusess-battery-current-doesnt-go-to-zero-whe.html

ALso another discussion about changing also CVL, which Pylontech BMS seems to do.
https://community.victronenergy.com/questions/106290/bug-in-ess-dc-feed-in-enabled-charge-current-limit.html

WaldemarFech, 2022-11-25:
Ok,
so the Victron System itself cannot detect such a small current.
And because we have set a maximum charge voltage the system wants to keep this battery-full-voltage, but can't hold it exactly constant because of changing loads and sun.

But maybe we can do something anyway:
We see the cell voltage.
I could wait until one cell reaches 3.55V. Then i backup the current battery voltage and set CVL to this value. When the cell voltages (over all) goes over 3.55V eg. 3.57V, then I could calc the difference -0.02V, multiply with a configurable weight and lower the CVL as kind of penalty. 

That would be so nice. We give the balancer time to balance, but if the voltage still raises, then we drain all cells because we discharge the battery a bit.

I'm so excited to test it.
Now I'm motivated to realize it.

Here some examples about my thoughts. To write less first for 4S:
C1: 3.40
C2: 3.45
C3: 3.50
C4: 3.54
resulting CVL: 4 * 3.55V = 14.2V

C1: 3.40
C2: 3.45
C3: 3.50
C4: 3.55
(distanceToMax: 0.3V * weight to add anyway? i think not. If it falls under 3.55V, let it charge with max CVL and higher CCL again)
resulting CVL: sum(c1-c4) = 13.90V

C1: 3.40
C2: 3.45
C3: 3.54
C4: 3.55
resulting CVL: sum(c1-c4) = 13.94V

C1: 3.40
C2: 3.45
C3: 3.54
C4: 3.56
resulting ov: 0.01V
penalty: (ov^2 * weight (= -100)) could be: 0.01V * 0.01V * -100 = 0.01V
resulting CVL: sum(c1-c4) + penalty = 13.95 + -0.01V = 13.94V

C1: 3.40
C2: 3.45
C3: 3.54
C4: 3.57
resulting ov: 0.02V
penalty: (ov^2 *  -100) = 0.02V * 0.02V * -100 = 0.04V
resulting CVL: sum(c1-c4) + penalty = 13.96 + -0.04V = 13.92V

C1: 3.40
C2: 3.45
C3: 3.56
C4: 3.58
resulting ov: 0.04V
penalty: (ov^2 *  -100) = 0.04V * 0.04V * -100 = 0.16V
resulting CVL: sum(c1-c4) + penalty = 13.99 + -0.16V = 13.83V

we could then adjust the weight.

or i don't make a quadratic penalty function, but another mapping function like:
PENALTY_CV: [3.55,   3.6]
PENALTY_PV: [-0.01, -2.0]

maybe the easiest way?

Louisvdw, 2022-11-25:
I suggest the easy way. Some GX device already cannot cope if you add more than one BMS and they have other devices connected. A mapping should be good enough resolution

chaerle, 2022-11-25:
The easiest way is just to to ensure, that charge is really zero when CCL=0. I.e. in case if CCL=0, lower the CVL by some constant or lower it until charge = 0. With Waldemars solution everything works perfectly, but with the only exception the ensure really zero charging current at CCL=0.

A slightly more general way would be to check the real charge value and dynamically lower the CVL until the real charge load it matches CCL.

chaerle, 2022-11-26:
I just saw, that the seems to be already some code in battery trying to disallow charging in case of charge_current=0:

```
       if self.control_charge_current == 0:
            self.control_allow_charge = False
        else:
            self.control_allow_charge = True

```

WaldemarFech, 2022-11-27:
> seems to be already some code in battery trying to disallow charging in case of charge_current=0:

yeah, i did xD

here is a first foretaste:
```
# if the cell voltage reaches 3.55V, then reduce current battery-voltage by 0.01V
    # if the cell voltage goes over 3.6V, then the maximum penalty will not be exceeded
    # there will be a sum of all penalties for each cell, which exceeds the limits
    PENALTY_AT_CELL_VOLTAGE  = [3.55, 3.6]
    PENALTY_BATTERY_VOLTAGE  = [0.01, 2.0]  # this voltage will be subtracted
```
```
Starte Test
Mode:  WaldemarFech
c1-c4 in V: 3.50 | 3.45 | 3.45 | 3.45 - bat.voltage=13.85 - CCL:   2.00A - DCL: 200.00A - CVL:  14.20V - Penalty:   0.00
c1-c4 in V: 3.51 | 3.45 | 3.45 | 3.45 - bat.voltage=13.86 - CCL:   1.60A - DCL: 200.00A - CVL:  14.20V - Penalty:   0.00
c1-c4 in V: 3.52 | 3.45 | 3.45 | 3.45 - bat.voltage=13.87 - CCL:   1.20A - DCL: 200.00A - CVL:  14.20V - Penalty:   0.00
c1-c4 in V: 3.53 | 3.45 | 3.45 | 3.45 - bat.voltage=13.88 - CCL:   0.80A - DCL: 200.00A - CVL:  14.20V - Penalty:   0.00
c1-c4 in V: 3.54 | 3.45 | 3.45 | 3.45 - bat.voltage=13.89 - CCL:   0.40A - DCL: 200.00A - CVL:  14.20V - Penalty:   0.00
c1-c4 in V: 3.55 | 3.45 | 3.45 | 3.45 - bat.voltage=13.90 - CCL:   0.00A - DCL: 200.00A - CVL:  13.89V - Penalty:   0.01
c1-c4 in V: 3.56 | 3.45 | 3.45 | 3.45 - bat.voltage=13.91 - CCL:   0.00A - DCL: 200.00A - CVL:  13.50V - Penalty:   0.41
c1-c4 in V: 3.57 | 3.45 | 3.45 | 3.45 - bat.voltage=13.92 - CCL:   0.00A - DCL: 200.00A - CVL:  13.11V - Penalty:   0.81
c1-c4 in V: 3.58 | 3.45 | 3.45 | 3.45 - bat.voltage=13.93 - CCL:   0.00A - DCL: 200.00A - CVL:  12.73V - Penalty:   1.20
c1-c4 in V: 3.59 | 3.45 | 3.45 | 3.45 - bat.voltage=13.94 - CCL:   0.00A - DCL: 200.00A - CVL:  12.34V - Penalty:   1.60
c1-c4 in V: 3.60 | 3.45 | 3.45 | 3.45 - bat.voltage=13.95 - CCL:   0.00A - DCL: 200.00A - CVL:  11.95V - Penalty:   2.00
c1-c4 in V: 3.61 | 3.45 | 3.45 | 3.45 - bat.voltage=13.96 - CCL:   0.00A - DCL: 200.00A - CVL:  11.96V - Penalty:   2.00
c1-c4 in V: 3.62 | 3.45 | 3.45 | 3.45 - bat.voltage=13.97 - CCL:   0.00A - DCL: 200.00A - CVL:  11.97V - Penalty:   2.00
c1-c4 in V: 3.63 | 3.45 | 3.45 | 3.45 - bat.voltage=13.98 - CCL:   0.00A - DCL: 200.00A - CVL:  11.98V - Penalty:   2.00
c1-c4 in V: 3.64 | 3.45 | 3.45 | 3.45 - bat.voltage=13.99 - CCL:   0.00A - DCL: 200.00A - CVL:  11.99V - Penalty:   2.00
c1-c4 in V: 3.64 | 3.54 | 3.45 | 3.45 - bat.voltage=14.08 - CCL:   0.00A - DCL: 200.00A - CVL:  12.08V - Penalty:   2.00
c1-c4 in V: 3.64 | 3.55 | 3.45 | 3.45 - bat.voltage=14.09 - CCL:   0.00A - DCL: 200.00A - CVL:  12.08V - Penalty:   2.01
c1-c4 in V: 3.64 | 3.56 | 3.45 | 3.45 - bat.voltage=14.10 - CCL:   0.00A - DCL: 200.00A - CVL:  11.69V - Penalty:   2.41
c1-c4 in V: 3.64 | 3.57 | 3.45 | 3.45 - bat.voltage=14.11 - CCL:   0.00A - DCL: 200.00A - CVL:  11.30V - Penalty:   2.81
c1-c4 in V: 3.64 | 3.58 | 3.45 | 3.45 - bat.voltage=14.12 - CCL:   0.00A - DCL: 200.00A - CVL:  10.92V - Penalty:   3.20
c1-c4 in V: 3.64 | 3.59 | 3.45 | 3.45 - bat.voltage=14.13 - CCL:   0.00A - DCL: 200.00A - CVL:  10.53V - Penalty:   3.60
c1-c4 in V: 3.64 | 3.60 | 3.45 | 3.45 - bat.voltage=14.14 - CCL:   0.00A - DCL: 200.00A - CVL:  10.14V - Penalty:   4.00
```

WaldemarFech, 2022-11-27:
@chaerle 
Here you can try it out:
#https://github.com/WaldemarFech/dbus-serialbattery/releases/tag/v0.14beta3_WFech

closed 2022-12-14 by Louisvdw

rcrc1978, 2023-01-13:
I tried this linear setting and get a CVL of 13.4v that I can't seem to get rid of, when I set it back to false my CVL goes back to what I put in the max cell voltage setting. Been using 3.45/13.8. Tried bumping to 3.55/14.2 while using the linear setting but it still does not change from 13.4v. Also tried changing to what you have above in your post. 
 PENALTY_AT_CELL_VOLTAGE  = [3.55, 3.6]
    PENALTY_BATTERY_VOLTAGE  = [0.01, 2.0] 
My battery voltage was 13.3 to 13.6 pending if I had the charger on or not, it was at 80% SOC when testing. Any suggestions? 
I'm new to putty and have not figure out how to post my file yet so I apologize in advance.

Thanks for your work on this seems like a nice addition.