Louisvdw/dbus-serialbattery #304: Calculating the SOC value at jkbms

Bonzai10, 2022-11-27:
Hi,
it is possible to calculate the SOC value, available amps and consumed amps with the driver.

Reason: The values of the JKBMS deviate quite a bit from reality over time due to the "Abnormal release of current sensor" error.

So far I had the problem that the BMS had e.g. 98% SOC, then it was only charged with 5 amps, although the voltage was only 53.6. With the version of WaldemarFech the loading/unloading problem is already fixed. Thanks.

Now it would be nice to see a real SOC value.
It would be nice to calculate like a shunt which synchronizes itself again and again.
Likewise, the integration of the float mode in the WaldemarFech version would be very nice

Bonzai10, 2022-11-27:
[image]

Louisvdw, 2022-11-27:
Hi
It will not be easy to calculate SOC in the driver.

Bonzai10, 2022-11-28:
OK. Thank´s for the info

closed 2022-11-28 by Bonzai10

Dr-Gigavolt, 2023-01-15:
Perhaps this can help you:
https://github.com/Dr-Gigavolt/dbus-aggregate-batteries