Louisvdw/dbus-serialbattery #311: Midpoint deviation

merlep, 2022-11-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**VenusOS (please complete the following information):**
 - Device type: [e.g. VenusGX, Cerbo, Raspberry Pi]
 - Firmware Version [e.g. 262]

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: [e.g. LTT, Daly]
 - Cells: [e.g. 16]
 - Interface: [e.g. USB-RS485, USB-RS232, Ve.Direct]

**Additional context**
Add any other context about the problem here.

Louisvdw, 2022-11-29:
Is this a new feature request or possible bug? I don't see any details except the heading.

merlep, 2022-11-29:
It seems that Midpoint deviation as implemented on a BMV 702 is an unsigned number ie always above 0 . With the ddbus implementation ( which I guess is supposed to mimic the BMV implementation )  I see both positive and negative numbers reported to VenusOS remote console.  I say "seems"  as the Maths given in Victron datasheet does not indicate its an absolute number , however their description on  setting up an alarm in the handbook indicates its can only be a positive number. Furthermore I have only ever seen a positive deviation on my system having had several battery batteries in the past

merlep, 2022-11-29:
I made a statement , not sure if you can see it on not. Basically how Victron implements deviation and how its done on dbusSerialdriver is slightly different . Victron implementation is unsigned whilst in dBusSerialdriver it is signed. Not a big deal , just how one implements an Alarm Rule for deviation  on VRM is slightly different

mr-manuel, 2023-04-20:
Sorry, but I can't understand it neither.

1. You are saying that basically it's working, but it should be implemented like Victron does?
2. You saw, that the midpoint deviation sometimes was calculated negative by `dbus-serialbattery` and it should always be positive?

I could add `abs()` to the midpoint deviation to unsign the value.

merlep, 2023-04-21:
You cannot have a negative deviation by definition is always a positive
number - just but an ABS function after your calculation

mr-manuel, 2023-04-21:
A fix will be released with the next version.

PR #551 (2023-04-25): Some fixes
* Changed: Fix System Switch in IO page
* Changed: Fix #239 by @TimGFoley
* Changed: Fix #311
* Changed: Run code analyse only on code change
* Changed: Separate Time-To-Go and Time-To-SoC activation
* Changed: fix UnicodeDecodeError https://github.com/Louisvdw/dbus-serialbattery/discussions/485#discussioncomment-5694781
commit a4da321e (2023-04-25): fix #311
commit d8a5d643 (2023-04-25): Some fixes

* Changed: Fix System Switch in IO page
* Changed: Fix #239 by @TimGFoley
* Changed: Fix #311
* Changed: Run code analyse only on code change
* Changed: Separate Time-To-Go and Time-To-SoC activation
* Changed: fix UnicodeDecodeError https://github.com/Louisvdw/dbus-serialbattery/discussions/485#discussioncomment-5694781
closed 2023-04-21 by mr-manuel