Louisvdw/dbus-serialbattery #313: trouble installing v13 or v14 update

havnjero, 2022-12-02:
odd thing happening when trying to install v13:

following upgrade steps from the following, but v14 wasn't working.
https://github.com/Louisvdw/dbus-serialbattery/discussions/261
https://github.com/Louisvdw/dbus-serialbattery/blob/master/etc/dbus-serialbattery/installrelease.sh#L2

root@einstein:~# wget https://github.com/Louisvdw/dbus-serialbattery/releases/download/v0.13/venus-data.tar.gz

--2022-12-02 12:07:35--  https://github.com/Louisvdw/dbus-serialbattery/releases/download/v0.13/venus-data.tar.gz
Resolving github.com... 140.82.121.3
Connecting to github.com|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/288771147/bf8863f0-8490-4dfb-a

(I truncated this for readability)

2022-12-02 12:07:37 (245 KB/s) - ‘venus-data.tar.gz’ saved [58408/58408]

root@einstein:~# 
root@einstein:~# tar -zxf venus-data.tar.gz -C /data
root@einstein:~# curl -s https://api.github.com/repos/Louisvdw/dbus-serialbattery/releases/latest | grep "https://github.com/Louisvdw/dbus-serialbattery/releases/download/v0.13/venus-data.tar.gz" | cut -d : -f 2,3 | tr -d \" | wget -O venus-data.tar.gz -qi - 

root@einstein:~# 
root@einstein:~# tar -zxf ./venus-data.tar.gz -C /data
root@einstein:~# sh /data/etc/dbus-serialbattery/reinstalllocal.sh
cp: omitting directory '/data/etc/dbus-serialbattery/__pycache__'
cp: omitting directory '/data/etc/dbus-serialbattery/dbus-serialbattery'
cp: omitting directory '/data/etc/dbus-serialbattery/qml'
cp: omitting directory '/data/etc/dbus-serialbattery/service'
root@einstein:~# 

cerbo gx with venus v3.00-8
jkbms

Louisvdw, 2022-12-02:
Hi
Just reboot and check that the new version is loaded. The omitting directories are just sub folders that is copyied elsewhere and those warnings can be ignored

havnjero, 2022-12-02:
ok! thank you. 

and the same goes for v.14 as welll?

Louisvdw, 2022-12-02:
The v0.13 betas has the warning suppressed, so you should not see it, but it works the same.

closed 2022-12-14 by Louisvdw