Louisvdw/dbus-serialbattery #317: v14 not running correctly

Bonzai10, 2022-12-04:
i have problems with v14 when switching to linear:
no CVL limit is displayed there and the driver is no longer recognized under DVCC.
I have the feeling that the driver is not working properly.

During a comparison I noticed that in the new utils.py there is no longer an if LIMITATION_MODE query.

how does the driver know which settings to use for linear and step?

Louisvdw, 2022-12-05:
I see now that I left the Limitation mode setting a bit inconsistant to how the other settings are set. You switch between Step and Linear by commenting out line 37 and removing the comment from line 38. 

So for the normal Step mode you should have 
```
LIMITATION_MODE = "Step"
# LIMITATION_MODE = "Linear"
```
[image]

while if you want to use Linear mode you should have
```
# LIMITATION_MODE = "Step"
LIMITATION_MODE = "Linear"
```
[image]

All the Limit Attributes that are Enabled for Charge/Discharge will then be applied. So all their settings must be populated.

If something is nor working, can you please provide the serial battery log files

DirkReudenbach, 2022-12-05:
Hi Louis,

thanks - this is exactly how I did it and the result is like described above. Even the MPPT switch of due to missing BMS.
[image]
[image]

How can I retrieve the serial battery log file?

Bonzai10, 2022-12-05:
Identical to me. made the change. The only thing I didn't do was delete the comment after # 

Please also instruct me how to get the log file.

Possible it is the firmware? i use 3.00 ~8

DirkReudenbach, 2022-12-05:
same Firmware on my site 3.00 ~8

Louisvdw, 2022-12-05:
It could be the firmware (I tested on 2.92). But might also be something else.

For the log file instructions see https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot#check-the-log-files-on-your-gx-device

DirkReudenbach, 2022-12-05:
Hi Louis,
pls find attached the log files

„Serialbattery #14 (Step)“

root@einstein:~# tail -f /data/log/dbus-serialbattery.ttyUSB0/current
@40000000638e151e13254c6c INFO:SerialBattery:Testing Jkbms
@40000000638e151e1aeb903c INFO:SerialBattery:Connection established to Jkbms
@40000000638e151e1aebb364 INFO:SerialBattery:Battery connected to dbus from /dev                                                                                        /ttyUSB0
@40000000638e151e1b7040fc INFO:SerialBattery:=== Settings ===
@40000000638e151e1b81cd2c INFO:SerialBattery:> Connection voltage 50.68V | curre                                                                                        nt -0.62A | SOC 7%
@40000000638e151e1bf08104 INFO:SerialBattery:> Cell count 16 | cells populated 0
@40000000638e151e1c16f8ac INFO:SerialBattery:> CCL Charge NoneA | DCL Discharge                                                                                         NoneA
@40000000638e151e1c73f944 INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_                                                                                        VOLTAGE 3.5V
@40000000638e151e238a9c24 INFO:SerialBattery:DeviceInstance = 1
@40000000638e151e239955ac INFO:SerialBattery:com.victronenergy.battery.ttyUSB0

„Serialbattery #14 Linear“

root@einstein:~# tail -f /data/log/dbus-serialbattery.ttyUSB0/current
@40000000638e18f814a5e95c INFO:SerialBattery:Testing Daly
@40000000638e18f82665db5c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000638e18f82676179c INFO:SerialBattery:Testing Daly
@40000000638e18f838351f3c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000638e18f83849dbd4 INFO:SerialBattery:Testing Jkbms
@40000000638e18f90f33c304 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000638e18f90f483d34 INFO:SerialBattery:Testing Lifepower
@40000000638e18f921a0aa5c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000638e18f921b14074 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@40000000638e18f921c2f79c INFO:SerialBattery:Testing Renogy
@40000000638e18f933cd96cc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000638e18f933f22a14 INFO:SerialBattery:Testing Renogy
@40000000638e18fa09d612f4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000638e18fa09e6b8ac INFO:SerialBattery:Testing Ecs
@40000000638e18fa2b81e2d4 ERROR:SerialBattery:ERROR >>> No battery connection at                    /dev/ttyUSB0

root@einstein:~# tail -f /data/log/serial-starter/current
@40000000638e18ef0a1834a4 INFO: Create daemontools service vedirect-interface.ttyUSB1
@40000000638e18f52781b36c INFO: Start service vedirect-interface.ttyUSB1 once
@40000000638e18f93ae051ac INFO: Create daemontools service dbus-serialbattery.ttyUSB1
@40000000638e18fe075484ac INFO: Create daemontools service gps-dbus.ttyUSB0
@40000000638e19000ad644e4 INFO: Start service dbus-serialbattery.ttyUSB1 once
@40000000638e1904111c41b4 INFO: Start service gps-dbus.ttyUSB0 once
@40000000638e190e319c6aa4 INFO: Create daemontools service vedirect-interface.ttyUSB0
@40000000638e191437184944 INFO: Start service vedirect-interface.ttyUSB0 once
@40000000638e19181b4fdfec INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@40000000638e191e2503b734 INFO: Start service dbus-cgwacs.ttyUSB0 once

„Switch to USB1“
root@einstein:~# tail -f /data/log/dbus-serialbattery.ttyUSB1/current
@40000000638e19001e529c84 INFO:SerialBattery:Testing Jkbms
@40000000638e190026366304 INFO:SerialBattery:Connection established to Jkbms
@40000000638e19002659871c INFO:SerialBattery:Battery connected to dbus from /dev/ttyUSB1
@40000000638e1900266c0964 INFO:SerialBattery:=== Settings ===
@40000000638e19002687a3cc INFO:SerialBattery:> Connection voltage 49.3V | current -0.2A | SOC 5%
@40000000638e190026aa9134 INFO:SerialBattery:> Cell count 16 | cells populated 0
@40000000638e190026bf8864 INFO:SerialBattery:> CCL Charge NoneA | DCL Discharge NoneA
@40000000638e190026e004cc INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.5V
@40000000638e19002d4a1014 INFO:SerialBattery:DeviceInstance = 3
@40000000638e19002d6aa004 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1

Bonzai10, 2022-12-05:
i hope thats rigt

[image]

[image]

DirkReudenbach, 2022-12-05:
Hi Louis,

found another topic which is not clear for me. Changing the DCL to another value stays at the same level 
Could you check this as well, please?

Thx

Dirk

Louisvdw, 2022-12-05:
@Bonzai10 you log show that the driver is running, so not sure what is happening there.
For @DirkReudenbach the log show that the driver is running normally when in Step and that it does not even pick up the JKBMS (or if there is a error that may be earlier in the logfile?) If the logfile show and error with the line numbers then I can check that.

For the DCL question you need to check each of the 3 limiting attributes. Between them which ever limits to the smallest value will be used. So if your temperature gives your a limit of 40A and your cell voltage gives you a limit of 50A then the resulting limit will be the 40A from the temeratures. You need to disable them one by one to see which one is the limiting value currenlty for you, or you can calculate it from the attributes and the limiting setting for those. It should be much easier when I have the GUI finished.

Louisvdw, 2022-12-05:
Can you show me the log file with the error when your run Linear please.

lennycb, 2022-12-05:
Cool work, thanks for that! I'm looking forward to setting all the magic in the UI. :-)

Now my problem:
Since I keep having problems with the internal SoC calculation of the Daly, I now open the lines in the "SOC limitation" section
```
CCCM_SOC_ENABLE = False
DCCM_SOC_ENABLE = False
```
set. The battery is now charged again up to 55.2V (3.45V*16). In places, the SoC had already jumped to 100% at 52.5V and charging was stopped.

2 questions:
- can I leave the setting without "SOC limitation"?
- what effects does the "System Setup/Battery Monitor" setting on Shunt instead of Daly have on your program?

Louisvdw, 2022-12-05:
You can disable the SOC attribute limitation, yes. That is no problem. 
It is also no problem if you disable all the attribute limits. It will just mean that the charge currents your battery specify to be charge with will be a fixed value. Many batteries work like that. 

All these limits do is that if possible problems arise then it will limit the current and hopefully reduce/prevent the problem. 
For instance if your battery temperature becomes to high, then the reduced current would mean there would also be less heat and the battery will cool down.  

The driver is not effrected if you do not choose it as your Battery Monitor. The GX system is effected by not using the data from your battery for it's control. The easiest explanation would be that when you choose as the battery monitor would be the device the the SOC for your system is being read from.

Bonzai10, 2022-12-05:
Thats above are the logs from linear

lennycb, 2022-12-05:
OK thanks for the answer.
Then I'll leave it like that and watch.
I think it works best for my setup.

Louisvdw, 2022-12-06:
There is a new patch release that fix a few issues and add more details to the log file. See if it helps.
https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.14.1

Bonzai10, 2022-12-06:
with the new patch nothing is running. no step, no linear

[image]

[image]

Bonzai10, 2022-12-06:
Hi Louis, you do a very, very good job. many thanks.
but as i stay not at home this week, i have made a rollback to v0.14beta1.

Louisvdw, 2022-12-07:
@Bonzai10 it seems that the serialbattery has move to another USB port. You 3rd screenshot show it on ttyUSB1. That is why the log file on ttyUSB0 show nothing. 
I see that CVL show empty. I'll see if I can replicate that.

kakariki1, 2022-12-10:
I think some of the issues here have to do with https://github.com/Louisvdw/dbus-serialbattery/issues/334

Louisvdw, 2022-12-15:
try fix in #353 - build V0.14.2

Bonzai10, 2023-01-02:
Hi, sorry it took me so long.
Now it works, but under the linear function it takes the value as CVL that is set as FLOAT_CELL_VOLTAGE under "Step"?!
does anything else need to be changed here?

[image]

[image]

[image]

Bonzai10, 2023-01-08:
hi, since v 014.3 doesn't work for me in linear mode either (after testing for a long time, I found that it still uses the Step settings). I'm going back to 014.1 and waiting for new updates. Thank you for the excellent driver!

closed 2023-01-08 by Bonzai10

rcrc1978, 2023-01-13:
> Hi, sorry it took me so long. Now it works, but under the linear function it takes the value as CVL that is set as FLOAT_CELL_VOLTAGE under "Step"?! does anything else need to be changed here?
> 

I seem to be having the same issue, when linear is enabled i'm stuck with a CVL of 3.35v/13.4v and that's my float setting. Tried upping the max cell voltage in all the other areas with no luck. Disabled linear mode and CVL is back to 3.45/13.8v. I'm using V14.3.