Louisvdw/dbus-serialbattery #318: v14 upgrade: Charge limited to 50A, dont understand why

pos-ei-don, 2022-12-05:
Since the upgrade, charge and discharge speed is limited to 50A.
I cannot figure out, why, these values are not present in the utils.py.

I need to increase them to 450, to charge faster.
I rolled back to v13 for today.
Thank you for this great driver!!

Louisvdw, 2022-12-06:
You need to have 450 in the value grid, so change it to 

> MAX_CHARGE_CURRENT_CV = [ 0, 2, 160, 450]

pos-ei-don, 2022-12-10:
Ok, thank you! This helps! 50 was never mentioned there, so i didn't find the cause for it. 
it should have been higher, but something got substracted so the result was 50A.
I stillt don't understand exactly, how the result gets calculated, but i understand that i can raise it by setting these values always "a bit higher than wanted".

closed 2022-12-10 by pos-ei-don