Louisvdw/dbus-serialbattery #324: Cerbo GX -death after trying to upload the driver.

entropia1ac, 2022-12-07:
Cerbo GX -death after trying to upload the driver. Milky white display, LAN SSH connection OK.
Any ideas?

LOG:
Using username "root".
root@192.168.1.172's password:
Last login: Wed Dec  7 16:26:39 2022 from 192.168.1.171
root@einstein:~# wget https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/installrelease.sh
--2022-12-07 16:27:37--  https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/installrelease.sh
Resolving raw.githubusercontent.com... 185.199.111.133, 185.199.109.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 277 [text/plain]
Saving to: ‘installrelease.sh’

installrelease.sh   100%[===================>]     277  --.-KB/s    in 0s

2022-12-07 16:27:38 (1.19 MB/s) - ‘installrelease.sh’ saved [277/277]

root@einstein:~# sh installrelease.sh
sh: /opt/victronenergy/swupdate-scripts/remount-rw.sh: No such file or directory
mkdir: can't create directory '/opt/victronenergy/service-templates/dbus-serialbattery': No such file or directory
cp: can't create directory '/opt/victronenergy/service-templates/dbus-serialbattery/log': No such file or directory
cp: can't create '/opt/victronenergy/service-templates/dbus-serialbattery/run':       No such file or directory
root@einstein:~#

Bonzai10, 2022-12-07:
Did you made a reboot out of ssh? The errors you can ignore.

entropia1ac, 2022-12-07:
yes - white screen after reset. Cerbo soft 2.77

Bonzai10, 2022-12-07:
You should make a Update to 2.92. I belive the driver is not for so old version

entropia1ac, 2022-12-07:
I managed to recover Cerbo after upgrading via SSH to version 2.91

entropia1ac, 2022-12-07:
Works with Cerbo V2.92 and DALY BMS

closed 2022-12-07 by entropia1ac