Louisvdw/dbus-serialbattery #326: Support for Orion JR2 BMS

Gijsvm71, 2022-12-08:
I just add succesfully my DALY bms to my CCGX, THANKS for it !!!
 would it be possible to add my orion JR2 bms too ?

https://www.orionbms.com/products/orion-jr2-bms/

Louisvdw, 2022-12-09:
The Orion JR2 does have a RS232 serial interface, but there is no reference on their website for a communications protocol that it use. If you can find that then it should be possible to add.

Gijsvm71, 2022-12-11:
I have connected the Orion via CAN bus to my CCGX, but the information I receive is very limited, only see the battery percentage, the current and the voltage, no further information about the cell voltage. It would be nice if this could also be visible too.