Louisvdw/dbus-serialbattery #327: Support BMS DR-JC03?

andkirby, 2022-12-09:
**Describe the bug**
I'm trying gather all things together for:
Victron MultiPlus 48/5000/70 + Cerbo GX + Battery LiFePO4 with BMS DR-JC03.

I've installed your app, and connected 
`Cerbo GX-USB` <-> `USB-RS485` Adapter with 3 pins (A/B/Ground) <-> `RS485-RJ-45` on battery

On RJ-45(RS485):
1- RS485B
2- RS485B
3- GND
(I made this according to [this manual](https://jakiperbattery.com/2022/05/17/mpp-solar-lv6548/). It's not so suitable, but didn't find better.)

Am I wrong somewhere?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Cerbo GX terminal
2. Install dbus-serialbattery
3. Connect the battery
4. See logs: `tail -f /data/log/dbus-serialbattery.ttyUSB0/current`
5. See error
```
ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
```

**Expected behavior**
Logs shall show success connection.

**Screenshots**
BMS
[image]
[image]
[image]

**VenusOS (please complete the following information):**
 - Device type: Cerbo GX
 - Firmware Version: 2.92 (latest firmware for 2022-12-09)

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: ??? / LFP
 - Cells: 16
 - Interface: 2xRJ45-RS485, 1xRJ45-RS232

**Additional context**

Louisvdw, 2022-12-09:
This is not a BMS that is supported by die driver. I cant's find any communication protocol information on their website, so I can't say if it would be possible to add.

andkirby, 2022-12-09:
Hi @Louisvdw ,
Thanks for your quick response.

But how these 3 wires to RJ45 (RS495) shall be connected to a battery via RS485/RJ45?
Is there some standard or something?

As the solution 1-`A+` / 2-`B-` / 3-`GND` does not work, I found this adapter USB-RS485/RJ45
https://www.amazon.com/RS-485-Adapter-Serial-BCBL-01-Communication/dp/B08GLFBTZW

According to the picture: Green-`GND`, Red-`A+`, Black-`B-`.
[image]

So, on RJ-45(RS485):
1) GND
6) RS485 A+
7) RS485 B-

Error `No battery connection at /dev/ttyUSB0`
<details>
  <summary> Full log</summary>
```
@400000006393172b34f8ebb4 INFO:SerialBattery:Starting dbus-serialbattery
@400000006393172b35055f34 INFO:SerialBattery:dbus-serialbattery v0.141
@400000006393172b350f80f4 INFO:SerialBattery:Testing LltJbd
@400000006393172c0a5dffd4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006393172c0adb0b8c INFO:SerialBattery:Testing Ant
@400000006393172c1bc2d59c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006393172c1c403744 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@400000006393172c1c4a2254 INFO:SerialBattery:Testing Daly
@400000006393172c2e399094 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006393172c2ebfd3ac INFO:SerialBattery:Testing Daly
@400000006393172d0421ced4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006393172d0496df94 INFO:SerialBattery:Testing Jkbms
@400000006393172d157be2b4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006393172d15f0d434 INFO:SerialBattery:Testing Lifepower
@400000006393172d276c182c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006393172d27e00fac ERROR:SerialBattery:>>> ERROR: Incorrect Data
@400000006393172d27ece8bc INFO:SerialBattery:Testing Renogy
@400000006393172d38d95a34 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006393172d3957487c INFO:SerialBattery:Testing Renogy
@400000006393172e0e88f474 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006393172e0f071d54 INFO:SerialBattery:Testing Ecs
[...]
@400000006393173419ad9db4 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
```
</details>

That's a web page of my battery: [512KWH-48V-lifepo4](https://www.xdbattery.com/512KWH-48V-lifepo4-Rack-Mounted-battery-module-309.html)

Louisvdw, 2022-12-09:
The wires does not matter if the driver does not talk the language of that battery. 
If this battery can do CANbus you might not need the serial battery driver.

andkirby, 2022-12-09:
I got no CANbus input, unfortunately.
[image]

> The wires does not matter
But I have to establish right connection first, right?

Louisvdw, 2022-12-09:
Yes, you do need to get the correct wire connection first. But there are no standard for that. You will have to check your battery manual which pins are used. 
But even if the correct pins are used and the connection is valid, the driver will not know how to talk to this battery unless it use the same BMS as one of the supported BMS.

andkirby, 2022-12-09:
> But there are no standard for that.

That's a shame :(

> You will have to check your battery manual which pins are used.

No manual from XD battery vendor. :( 🤦 

Thanks for this note, L!

andkirby, 2022-12-11:
Contacted with the XD Battery company. I was told, it can be connected with their battery (w/o programming).
They promised to help to make it work. My hands are itching to see the manual.
I'll bring updates...

joggl, 2022-12-28:
Diese Batterie funktioniert mit dem Treiber nicht. Du musst folgende Dinge ausführen. https://we.tl/t-pgNDHMLJ7y (the link for the battery manual, pc software, BIN code , upgrade video)
1.upgrade the battery by the BIN code first.
2.battery connect victron by CAN bus (the RJ45 cables PIN change for CAN)
3.master battery dip 5 and 6 on 
4.victron setting pylon working model

andkirby, 2023-01-24:
@joggl, thanks for the answer. Why it's not in english (completely)? :-/
What this link about? I see "Introducing portraits". I guess, you added wrong link.

1) Didn't get it. What does "BIN code" mean?
2,3) Yes, 5 & 6 pins must be up. For some reason it works only via CAN bus which is configured as BMS CAN.

That's the answer from the battery vendor. This manual helped me to connect my battery.
[image]

Probably, will add detailed manual later.

joggl, 2023-01-24:
Hello undkirby, i am from Germany, i try to write in english. I have a battery with the same BMS from an other battery maker. The program from louis doesn't work with this BMS. In the delivered version of my battery the BMS can not comunicate with victron and I have to install an Update on my battery ( this means bin-code). In the link are the updatefile, a PC-software to bring the update on the battery and choose some settings, a movie who show the way of update and settings and a manual (PDF). I had to go this way and no my Raspberry Pi with venus os comunicate with the battery. Best Regards

andkirby, 2023-01-29:
Hello @joggl,

it looks like you did a piece of work!

So, the link either is broken or invalid or not allowed to see from my side. I see the following:
> Transfer expired
> Sorry, this transfer has expired and is not available any more

I assume, something went wrong... =) Maybe you have another source of this video.

My questions to your comment:

1. Where this update (installed on your battery) is coming from?
2. How to did you install it?
3. Which version do you have now?

> BMS can not communicate with victron
4) It somehow communicates with my Cerbo. Does it work via BMS Can on your side?
I have "Settings > Services > VE.Can port > Can Bus profile" = "CAN-bus BMS (500 kb/s)".

joggl, 2023-01-29:
This is the manual of my battery. The update for my battery I get from my seller. I only connect 2 cables on my can-hat.
[image]
[NEW-48V LIFEPO4 BATTERY USER MANUAL.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/10530753/NEW-48V.LIFEPO4.BATTERY.USER.MANUAL.pdf)

joggl, 2023-01-29:
I don't understand how i can do the Video in GitHub. So i do it this way . 
   
  
    Best Regards

andkirby, 2023-01-30:
> I don't understand how i can do the Video in GitHub. So i do it this way .
I assume, youtube would be the best place ;)

LeoWinterDE, 2023-02-11:
We started to collect all available information (manuals, firmware version and so on) over here https://github.com/LeoWinterDE/BMS. Feel free commit any new firmware version or guide, thanks.

> This is the manual of my battery. The update for my battery I get from my seller. I only connect 2 cables on my can-hat (CANH and CANL -> Pin 4+5). [image] [NEW-48V LIFEPO4 BATTERY USER MANUAL.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/10530753/NEW-48V.LIFEPO4.BATTERY.USER.MANUAL.pdf)

> Hello @joggl,
>
> it looks like you did a piece of work!
>
> So, the link either is broken or invalid or not allowed to see from my side. I see the following:
>
>
> I assume, something went wrong... =) Maybe [...]

ChristophPech, 2023-02-14:
> We started to collect all available information (manuals, firmware version and so on) over here https://github.com/LeoWinterDE/BMS. Feel free commit any new firmware version or guide, thanks.

I'm not allowed to commit directly, but this here seems to be an older version of the protocol: https://www.photovoltaikforum.com/core/attachment/77134-pylon-lfp-battery-communication-protocol-rs485-v2-8-20161216-pdf/

MarzaBogdan, 2023-03-14:
> We started to collect all available information (manuals, firmware version and so on) over here https://github.com/LeoWinterDE/BMS. Feel free commit any new firmware version or guide, thanks.

Hi. After all of this information i still have a question :) I am using a raspberry pi with venus OS. So this type of bms works only with waveshare can pi hat ? I have a USB -Rs485 adapter and i failed to connect this bms with raspberry pi. So i have to buy a waveshare can pi hat ? The adapter i use right now is the one in the picture, please see the link down below. Thanks.

[https://user-images.githubusercontent.com/127730319/224959954-41155586-4ca1-456b-9dff-6ebe6ba1ebe7.jpg]

LeoWinterDE, 2023-03-14:
@MarzaBogdan i have only been able to run it with a pi Hat on the venus os, but technically it should also work with a RS485 USB adapter - but I haven't seen this yet.

MarzaBogdan, 2023-03-14:
> @MarzaBogdan i have only been able to run it with a pi Hat on the venus os, but technically it should also work with a RS485 USB adapter - but I haven't seen this yet.

I have to install or activate any driver on venus os ?

LeoWinterDE, 2023-03-14:
> > @MarzaBogdan i have only been able to run it with a pi Hat on the venus os, but technically it should also work with a RS485 USB adapter - but I haven't seen this yet.
> 
> I have to install or activate any driver on venus os ?

This [pi hat from waveshare](https://www.amazon.de/gp/product/B09JKJCMHN/) runs without any additional drivers on venus os.

MarzaBogdan, 2023-03-14:
> This [pi hat from waveshare](https://www.amazon.de/gp/product/B09JKJCMHN/) runs without any additional drivers on venus os.

Ok thanks, I will try again with my rs485 adapter and if it still not working i will buy that pi hat waveshare. The thing about my rs485 adapter is i have only 3 ping A B and GND. On wave share i see H L B A.

Thanks for your quickly response

joggl, 2023-03-14:
For me it only works the can hat too. I have to Insert a few words at config.txt. See on waveshare wiki.
https://www.waveshare.com/wiki/RS485_CAN_HAT