Louisvdw/dbus-serialbattery #339: JK BMS Temperature problem below 0°

Knedox, 2022-12-10:
Hello,
now that we have negativ temperature in germany i have seen that my battery reports 0°C then 50°C then 100°C and then 0°C

So it seems like the negative temperature calculation here is wrong.
https://github.com/Louisvdw/dbus-serialbattery/blob/4157bd5dbe44c0bb4876f206af9ad810fdc339b2/etc/dbus-serialbattery/jkbms.py#L88

I was already able to find this out:

0x02 = 0b10      -> 2°C
0x01 = 0b1       -> 1°C
0x00 = 0b0       -> 0 °C
0x32 = 0b0110010 -> 0 ?
0x64 = 0b1100100 -> -1 °C
0x65 = 0b1100101 -> -2 °C
0x66 = 0b1100110 -> -3 °C

does anyone have the same problem? i am not really sure how to handle the 0x32 byte. Seems a little bit out of order

kakariki1, 2022-12-12:
According to the protocol definition here (https://github.com/jblance/mpp-solar/issues/112#issuecomment-850585642) the part exceeding 100 is negative temperature, such as 101 is negative 1 degree.

So I think the problem is here:
https://github.com/Louisvdw/dbus-serialbattery/blob/4157bd5dbe44c0bb4876f206af9ad810fdc339b2/etc/dbus-serialbattery/jkbms.py#L88

This should be changed to:
self.to_temp(1, temp1 if temp1 <= 100 else temp1 - 100)

Best regards,
lakeroe

wormsman, 2022-12-12:
Many thanks for you fix and the protocol definition. I have changed it online on my installation. Currently my battery is at +1°C, so maybe I can see if the temperature reading is Ok with the change.

kakariki1, 2022-12-13:
I made a mistake above. If it helps at all, the change has to be made to:
self.to_temp(1, temp1 if temp1 <= 100 else -(temp1 - 100))

In theory there should be no difference to the original code though ...

MS123uk, 2022-12-15:
how do you make this change to the scraped ?  I'm new programming rasp and vinos os

kakariki1, 2022-12-15:
Sorry, I'm no native english speaker.
What do you mean by "change to the scraped" ?

MS123uk, 2022-12-15:
i have same problem with  jk bms showing the wrong temp below 0 c or 32f  i.e. 100 c  when bms probe go below 0c 
you 
This should be changed to:
self.to_temp(1, temp1 if temp1 <= 100 else temp1 - 100)

how  i  change this line

kakariki1, 2022-12-15:
No I understand, just read here (How to edit utils.py):
https://github.com/Louisvdw/dbus-serialbattery/wiki/How-to-install#how-to-edit-utilspy

MS123uk, 2022-12-15:
ok  can not find  line 88  self.to_temp(1, temp1 if temp1 <= 100 else 100 - temp1)  ?

closed 2022-12-15 by Louisvdw

Knedox, 2022-12-15:
Well, now there is a copy & paste mistake in the code. Please check temp2, line 89

wormsman, 2022-12-15:
Also the fix was not working for me. Via Bluetooth I got -0°C as current temperature and with the code in version 14.2 I got 100°C.

For me 

        self.to_temp(1, temp1 if temp1 < 99 else (100 - temp1))
        self.to_temp(2, temp2 if temp2 < 99 else (100 - temp2))

seems to work. I now get 0°C and hopefully can check this night some values below 0°C.

Knedox, 2022-12-29:
> Also the fix was not working for me. Via Bluetooth I got -0°C as current temperature and with the code in version 14.2 I got 100°C.
> 
> For me
> 
> ```
>     self.to_temp(1, temp1 if temp1 < 99 else (100 - temp1))
>     self.to_temp(2, temp2 if temp2 < 99 else (100 - temp2))
> ```
> 
> seems to work. I now get 0°C and hopefully can check this night some values below 0°C.
> 
> I have tested the temperature sensors now with and salt ice bath and my fingers. In the range from -16°C up to 30°C the measurement with the formula above is correct and corresponds the data in the JK-BMS android software. In the android software I could see -0°C and also 0°C and booth were translated to 0°C.

i have tested your code. Seems to work:

[image]

commit e85023e5 (2022-12-15): fix #339 negative temp
commit 7b239007 (2022-12-15): Cvl fix (#353)

* fix #334 missing CVL

* fix #325 restore GUI

* fix #339 negative temp

* Deprecate Revov driver - replaced by LifePower