Louisvdw/dbus-serialbattery #350: Please add support for ABC-BMS (for SOK LiFePO4 Bluetooth batteries)

scottrosendahl, 2022-12-14:
Please add support for ABC-BMS (the BMS for SOK LiFePO4 Bluetooth batteries).

Thanks in advance!

zuccaro, 2022-12-27:
I have reverse engineered the ABC-BMS (SOK-BMS) protocol from tearing down their Android app and capturing bluetooth between mobile phone and battery. I'll share working PoC with @Louisvdw.

scottrosendahl, 2022-12-28:
> I have reverse engineered the ABC-BMS (SOK-BMS) protocol from tearing down their Android app and capturing bluetooth between mobile phone and battery. I'll share working PoC with @Louisvdw.

Awesome, thanks much!  Looking forward to trying it out.

scottrosendahl, 2023-02-14:
Hello, how's it coming along for this?

mdshw5, 2023-04-04:
I would love to see this implemented! Currently I have a home assistant instance running in my RV, with ESPHome pulling in BLE sensor data. If the spec for the ABC-BMS were known it would enable many other projects to take advantage of the BMS values.

mdshw5, 2023-04-07:
@zuccaro I'm about to start reverse engineering the SOK bluetooth services, but if you've already done this would you mind sharing with me as well?

zuccaro, 2023-04-07:
I don't have a full understanding of their protocol but it should help you get started!

zuccaro, 2023-04-07:
```py
#!/usr/bin/python3

# communicates with SOK BMS via BLE commands

import BLE_GATT
import logging
import argparse
import struct
import sys
import threading
import time
import statistics

mac_address = '00:00:01:AA:EE:DD'

svc_id    = '0000ffe0-0000-1000-8000-00805f9b34fb'
notify_id = '0000ffe1-0000-1000-8000-00805f9b34fb'
tx_id     = '0000ffe2-0000-1000-8000-00805f9b34fb'

cmd_name       = [ 0xee, 0xc0, 0x00, 0x00, 0x00 ]
cmd_info       = [ 0xee, 0xc1, 0x00, 0x00, 0x00 ]
cmd_detail     = [ 0xee, 0xc2, 0x00, 0x00, 0x00 ]
cmd_setting    = [ 0xee, 0xc3, 0x00, 0x00, 0x00 ]
cmd_protection = [ 0xee, 0xc4, 0x00, 0x00, 0x00 ]
cmd_break      = [ 0xdd, 0xc0, 0x00, 0x00, 0x00 ]

def get_str(ubit, uuid):
    """ reads utf8 string from specified bluetooth uuid """
    return ''.join(bytes(ubit.char_read(uuid)).decode('UTF-8'))

def unpack(pattern,data):
    """ slightly simpler unpack call """ 
    return struct.unpack(pattern, data)[0]

def getBeUint4(data, offset):
    """ gets big-endian unsigned integer """
    return unpack('>I',bytes(data[offset:offset+4]))

def getBeUint3(data, offset):
    """ reads 3b big-endian unsigned int  """
    return unpack('>I',bytes( [0]+data[offset:offset+3]))

def getLeInt3(data, offset):
    """ reads 3b little-endian signed int """
    return unpack('<i',bytes( [0] + data[offset:offset+3]))

def getLeShort(data, offset):
    """ reads little-endian signed short """
    return unpack('<h',bytes(data[offset:offset+2]))

def getLeUShort(data, offset):
    """ reads little-endian unsigned short """
    return unpack('<H',bytes(data[offset:offset+2]))

def send(device, data):
    """ sends data (appends crc - don't send in your own) """
    data2 = data + [minicrc(data)]
    logging.debug(f'Sending [{bytes(data2).hex().upper()}]')
    device.char_write(tx_id, data2)

def minicrc(data):
    """ computes crc8 of specified data"""
    i = 0;
    for b in data:
        i ^= b & 255;
        for i2 in range(0, 8):
            i =  (i >> 1) ^ 140 if (i & 1) != 0 else i >> 1;
    return i

[... 137 lines cut ...]

if __name__=='__main__':

    parser = argparse.ArgumentParser(
                    prog = 'SOK BMS Reader',
                    description =  'Communicates with weird BMS on SOK battery',
                    epilog = 'Reverse engineered from ABC BMS Android app (com.sjty.sbs_bms)')

    parser.add_argument('-a', '--address', type=str, default=mac_address)
    parser.add_argument('-d', '--debug', action='store_true', default=False)
    args = parser.parse_args()

    stream_handler = logging.StreamHandler(stream=sys.stderr)
    lx = logging.DEBUG if args.debug else logging.INFO
    stream_handler.setLevel(lx)
    logging.basicConfig(level=lx,format='%(asctime)s %(levelname)s %(message)s', handlers=[stream_handler])

    device = SokBms()
    device.connect(args.address)
    logging.info(device)

    logging.debug('Data collection complete, exiting')
    device.central.cleanup()

```

mdshw5, 2023-04-08:
@zuccaro This is amazing! Do you mind if I build on this and contribute this BMS to https://github.com/fl4p/batmon-ha? I'd be glad to credit you in the pull request.