Louisvdw/dbus-serialbattery #351: Support for a third temperature sensor

iLeeeZi, 2022-12-15:
I have a JBD-SP04S034 4S 200A bms and there is three temperature sensors, one internal and two external. Currently the driver uses only first two temperatures which are one internal and one external in my setup.

I did some testing and lltjbd.py seems to read the data correctly for all sensors but there is no calculations for self.temp3 in battery.py

mr-manuel, 2022-12-29:
Same thing for JK-BMS. Is it correct, that in Venus OS we see only one temperature?

vincegod, 2023-01-11:
This is something I have noticed too and as the temperature is an average on the internal sensor and first external sensor it skews the battery temperature giving a higher value.

I was thinking of seeing if I can make this configurable so you can pick the sensors to use.

mr-manuel, 2023-01-11:
Would it be possible, somehow, to display the temperatures like the cell voltages?

[image]

Obviously without `Sum`

mr-manuel, 2023-02-15:
https://github.com/Louisvdw/dbus-serialbattery/pull/440 adds the MOSFET temperature. So you have one external temperature sensor in `Minimum cell temperature` and the other external in `Maximum cell temperature`. Unfortunately you cannot difference which external temperature is linked to which external sensor.

flyer9999, 2023-04-15:
I also stumbled across this problem today and would like to follow up here again. My BMS is a JBD/AP21S002, the first sensor (Temp-1) is internal for the BMS, NTC2 (Temp-2) and NTC3 (Temp-3), which are connected externally to the batteries. Temp-1 and Temp-2 are currently being calculated, but this does not reflect the actual temperature in the cells..
This then also distorts the settings for "TEMPERATURE_LIMITS_WHILE_CHARGING" and "TEMPERATURE_LIMITS_WHILE_DISCHARGING" in 'utils.py'.
Shouldn't a simple setting in the 'utils.py' of the TempSensors be possible? For example. "Temp-1" -> n.a./BMS ; "Temp-2" n.a./Battery ; "Temp-3" n.a./Battery

DRIVER_VERSION = 0.14
DRIVER_SUBVERSION = '.3'
BMS = JBD/AP21S002 21S 200A
venusOS on Raspberry Pi3 Model B V1.2 = v2.94

pos-ei-don, 2023-04-16:
Using the mqtt driver from Manuel you can use any temp sensor and sent its value to the virtual mqtt-bms.

mr-manuel, 2023-04-16:
> Shouldn't a simple setting in the 'utils.py' of the TempSensors be possible? For example. "Temp-1" -> n.a./BMS ; "Temp-2" n.a./Battery ; "Temp-3" n.a./Battery

In my eyes this would complicate things a lot more. What do you understand under external sensors?

1. Only BMS hardware external, but battery internal
2. BMS and battery external
3. Maybe even other...

One solution could be to change the driver of the JBD to display the Temp-1 as MOS Temperature and the other two as cell temperatures. What do you think?

flyer9999, 2023-04-16:
> One solution could be to change the driver of the JBD to display the Temp-1 as MOS Temperature and the other two as cell temperatures. What do you think? 

Yes, perfect, that is what i mean. The "external" Sensors you can plug in and with these you go to the cells, these are Temp2 and Temp3. The Temp1 is the MOS/internal Temperature.

mr-manuel, 2023-04-16:
Could you try this file and check the log after some minutes? There should appear some informations.

To try it, you need first to install the `jkbms_ble` branch, check if it's working and if yes then swap the file.

[lltjbd.zip](https://github.com/Louisvdw/dbus-serialbattery/files/11242442/lltjbd.zip)

flyer9999, 2023-04-17:
Oh Hah

Sorry for my questions regarding the procedure, I'm not a programmer ...

You say:
> To try it, you need first to install the jkbms_ble branch, check if it's working and if yes then swap the file.

Means for me:
1. Install jkbms_ble
`cd /tmp
wget https://github.com/Louisvdw/dbus-serialbattery/archive/refs/heads/jkbms_ble.zip
unzip jkbms_ble.zip
cp /tmp/dbus-serialbattery-jkbms_ble/etc/dbus-serialbattery/ /data/etc
chmod +x /data/etc/dbus-serialbattery/*.sh
chmod +x /data/etc/dbus-serialbattery/*.py
chmod +x /data/etc/dbus-serialbattery/service/run
chmod +x /data/etc/dbus-serialbattery/service/log/run
`
bash /data/etc/dbus-serialbattery/installlocal.sh

3. Then --> Adjust Mac address .. Even if I'm using a JBD-BMS?
4. In utility.py under "battery_types" I have deactivated all BMS's except for the JBD that I am using, do I have to reactivate the Jkbms here? Have I currently deactivated because I don't have a Jkbms but a LltJbd?
5. I understand the subsequent replacement of the IItjbd.py

Where do I read the log file from??

sorry for the questions, but I don't want to destroy anything on my running system. I'm actually a quick learner and have already realized things in python...

mr-manuel, 2023-04-17:
Please use tripe backticks in the line before and after multiline code.

1. Yes
2. Not needed
3. Not needed. You can specify your BMS in the `config.ini`, if you want that only one BMS is tried
4. Yes

You can rename the existing folder to .old or something like that, so you can revert back, if something is not working. Just rename back and run `reinstalllocal.sh` in this case.

flyer9999, 2023-04-18:
Sorry for my late reply, just tried the update (almost 4 hours)...

Here are my Steps and what was happend:

`cd /tmp`
`wget https://github.com/Louisvdw/dbus-serialbattery/archive/refs/heads/jkbms_ble.zip`
`unzip jkbms_ble.zip`
`cp /tmp/dbus-serialbattery-jkbms_ble/etc/dbus-serialbattery/ /data/etc`
`chmod +x /data/etc/dbus-serialbattery/*.sh`
`chmod +x /data/etc/dbus-serialbattery/*.py`
`chmod +x /data/etc/dbus-serialbattery/service/run`
`chmod +x /data/etc/dbus-serialbattery/service/log/run`
`bash /data/etc/dbus-serialbattery/installlocal.sh`          --> ERROR

tar (child): ./venus-data.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now

Then i do the following:
`wget venus-data-URL `https://github.com/Louisvdw/dbus-serialbattery/releases/download/v0.14b3ble/venus-data.tar.gz`

Then:
`bash /data/etc/dbus-serialbattery/installlocal.sh`  -->> Kein Error

REBOOT
No connection from VenusOS to the BMS, the LogFile \data\log\dbus-serialbattery.ttyACM0 says:
@40000000643dbc3016d9c11c INFO:SerialBattery:=== Settings ===
@40000000643dbc3016e0a6bc INFO:SerialBattery:> Connection voltage NoneV | current NoneA | SOC None%
@40000000643dbc3016e6f404 INFO:SerialBattery:> Cell count None | cells populated 0
@40000000643dbc3016ed7be4 INFO:SerialBattery:> CCCM SOC False | DCCM SOC False
@40000000643dbc3016f3ffdc INFO:SerialBattery:> CCCM CV True | DCCM CV True
@40000000643dbc3016fa58dc INFO:SerialBattery:> CCCM T True | DCCM T True
@40000000643dbc301701658c INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.45V
@40000000643dbc30195655f4 INFO:SerialBattery:DeviceInstance = 1
@40000000643dbc30195dc44c INFO:SerialBattery:com.victronenergy.battery.ttyACM0
@40000000643dc242181d87d4 Traceback (most recent call last):
@40000000643dc242181da32c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 24, in <module>
@40000000643dc242181dba9c     from jkbms_ble import Jkbms_Ble
@40000000643dc242181dc26c   File "/opt/victronenergy/dbus-serialbattery/jkbms_ble.py", line 4, in <module>
@40000000643dc242181dd5f4     from jkbms_brn import JkBmsBle
@40000000643dc242181de1ac   File "/opt/victronenergy/dbus-serialbattery/jkbms_brn.py", line 2, in <module>
@40000000643dc242181fba54     from bleak import BleakScanner, BleakClient
@40000000643dc242181fc60c ModuleNotFoundError: No module named 'bleak'
@40000000643dc25132910f8c Traceback (most recent call last):
@40000000643dc25132912ae4   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 24, in <module>
@40000000643dc25132913e6c     from jkbms_ble import Jkbms_Ble
@40000000643dc25132914a24   File "/opt/victronenergy/dbus-serialbattery/jkbms_ble.py", line 4, in <module>
@40000000643dc251329159c4     from jkbms_brn import JkBmsBle
@40000000643dc2513291657c   File "/opt/victronenergy/dbus-serialbattery/jkbms_brn.py", line 2, in <module>
@40000000643dc2513305f554     from bleak import BleakScanner, BleakClient
@40000000643dc251330610ac ModuleNotFoundError: No module named 'bleak'

Now try to install python-bleak ...
`opkg install python3-pip python3-modules`  -->> ERROR  "opkg_prepare_url_for_install: Couldn't find anything to satisfy 
pyton3-"

Get some Information and do the following:
`/opt/victronenergy/swupdate-scripts/set-feed.sh candidate`
`opkg update`
`opkg install python3-pip python3-modules`
`pip3 install bleak`

REBOOT

No BMS connection
LogFile \data\log\dbus-serialbattery.ttyACM0 says:
@40000000643dcf8902a256dc INFO:SerialBattery:Starting dbus-serialbattery
@40000000643dcf8902ae7854 INFO:SerialBattery:dbus-serialbattery v0.14.3ble
@40000000643dcf8902b76d4c INFO:SerialBattery:Testing LltJbd
[... 65 lines cut ...]
@40000000643dd318198a528c INFO:SerialBattery:> CCCM CV True | DCCM CV True
@40000000643dd31819909804 INFO:SerialBattery:> CCCM T True | DCCM T True
@40000000643dd31819978574 INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.45V
@40000000643dd31828f01464 INFO:SerialBattery:DeviceInstance = 1
@40000000643dd31828f027ec INFO:SerialBattery:com.victronenergy.battery.ttyACM0

But i cannot see any Changes on VenusOS ???

Pictures befor the Update:
[image]

[image]

[image]

Pictures after the Update:
[image]

[image]

[image]

I'm not sure if that was all correct and if the information will help you. I backed up the log folder completely after each reboot, if you need more information just let me know. That was a 4 hour number for me :-) .....

PS: Just as an interim question, how can I uninstall the dbus-serialbattery cleanly and completely for a new installation without BLE?

mr-manuel, 2023-04-18:
I saw it right now and changed it. You don‘t have to run `installlocal.sh` but `bash /data/etc/dbus-serialbattery/reinstalllocal.sh`, else you are overwriting the data from the zip.

You don‘t need bleak netiher.

You can uninstall with the uninstall script.

flyer9999, 2023-04-19:
Where is an uninstall script for dbus-serialbattery ?? I can't find one ....

Today first i removed ...
`pip3 uninstall bleak`
`opkg remove python3-pip python3-modules`

Then I deleted the folder `/data/etc/dbus-serialbattery` and `/data/conf/serial-starter.d`

Then I make e new install with
`wget https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/installrelease.sh`
`sh installrelease.sh`
`reboot`

... and my JBD is online on VenusOS ... then I run your script:
`wget -O /tmp/install-nightly.sh https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/jkbms_ble/etc/dbus-serialbattery/install-nightly.sh && bash /tmp/install-nightly.sh`

and changed the `lltjbd.py`

Reboot ...

NO BMS on VenusOS ....

Logfile dbus-serialbattery.ttyACM0 :
@40000000643f1463117b3264 ModuleNotFoundError: No module named 'bleak'
@40000000643f14723279804c Traceback (most recent call last):
@40000000643f1472327997bc   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 26, in <module>
@40000000643f14723279ab44     from jkbms_ble import Jkbms_Ble
@40000000643f14723279b6fc   File "/opt/victronenergy/dbus-serialbattery/jkbms_ble.py", line 4, in <module>
@40000000643f14723279ca84     from jkbms_brn import JkBmsBle
@40000000643f14723279d63c   File "/opt/victronenergy/dbus-serialbattery/jkbms_brn.py", line 2, in <module>
@40000000643f1472327bb6b4     from bleak import BleakScanner, BleakClient
@40000000643f1472327bc654 ModuleNotFoundError: No module named 'bleak'

Logfile serial-starter :
@40000000643f139605906a64 *** CCGX booted (0) ***
@40000000643f13962e3c8a4c *** starting serial-starter ***
@40000000643f139712e53a14 serstart starting
@40000000643f139718c54924 INFO: loading config file /etc/venus/serial-starter.conf
@40000000643f1398119d4124 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@40000000643f1399223e421c INFO: Create daemontools service gps-dbus.ttyACM0
@40000000643f139c21230264 INFO: Create daemontools service mk2-dbus.ttyUSB0
@40000000643f139f35e3e024 INFO: Start service gps-dbus.ttyACM0 once
@40000000643f13a231de0f54 INFO: Start service mk2-dbus.ttyUSB0
@40000000643f13ab0bde334c INFO: Create daemontools service vedirect-interface.ttyACM0
@40000000643f13b11e5deef4 INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f13b22f3793ec INFO: Create daemontools service dbus-serialbattery.ttyACM0
@40000000643f13c706f8a90c INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f13c81ac591ac INFO: Start service gps-dbus.ttyACM0 once
@40000000643f13d135752c4c INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f13d508f6ab8c INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f13d81b3314d4 INFO: Start service gps-dbus.ttyACM0 once
@40000000643f13e12ae8bfb4 INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f13e43604357c INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f13e80500c8dc INFO: Start service gps-dbus.ttyACM0 once
@40000000643f13f113db7924 INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f13f41ee11144 INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f13f7295f6c4c INFO: Start service gps-dbus.ttyACM0 once
@40000000643f1400397674f4 INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f1404087e4b24 INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f14071385cf74 INFO: Start service gps-dbus.ttyACM0 once
@40000000643f14102277c0dc INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f14132d297854 INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f141639a25b3c INFO: Start service gps-dbus.ttyACM0 once
@40000000643f14200cfae234 INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f142317e9aa7c INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f14262427e9ac INFO: Start service gps-dbus.ttyACM0 once
@40000000643f142f32ce2c14 INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f1433037c4004 INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f14360e6f7904 INFO: Start service gps-dbus.ttyACM0 once
@40000000643f143f1d093464 INFO: Start service vedirect-interface.ttyACM0 once
[... 52 lines cut ...]
@40000000643f1eed0bf74d14 INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f1ef030fae314 INFO: Start service gps-dbus.ttyACM0 once
@40000000643f1ef40cab3f7c INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f1ef72504fb6c INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f1efa36a8939c INFO: Start service gps-dbus.ttyACM0 once
@40000000643f1f040afd8f7c INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f1f0715869114 INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f1f0a21745a9c INFO: Start service gps-dbus.ttyACM0 once
@40000000643f1f13305953b4 INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f1f163aea4c5c INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f1f1a0a831404 INFO: Start service gps-dbus.ttyACM0 once
@40000000643f1f2319b95d84 INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f1f26259957e4 INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f1f293102322c INFO: Start service gps-dbus.ttyACM0 once
@40000000643f1f33044d47bc INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f1f360fc52d94 INFO: Start service dbus-serialbattery.ttyACM0 once
@40000000643f1f391a865dfc INFO: Start service gps-dbus.ttyACM0 once
[... cut ...]
@40000000643f1f490482888c INFO: Start service gps-dbus.ttyACM0 once
@40000000643f1f52141211b4 INFO: Start service vedirect-interface.ttyACM0 once
@40000000643f1f552225b8dc INFO: Start service dbus-serialbattery.ttyACM0 once
`
Somehow that doesn't work... for further tests it would be important for me how I can uninstall dbus-serialbattery cleanly. Not that an error crept in when I proceeded with deleting "/data/etc/dbus-serialbattery" and "/data/conf/serial-starter.d"
Anyway, thanks for your efforts...

mr-manuel, 2023-04-19:
I will reinstall my Venus OS from scratch and test it again, to make sure it works. I will keep you updated.

mr-manuel, 2023-04-19:
@flyer9999 now I found the error and fixed it. A function name was wrong.

Thank you for all the testing. Uninstalling is not neccessary, if you are "updating" the driver. A reboot is anyway recomended.

Now it should run without errors, I tested it on a fresh system. Please redo this steps:

> And now i run your script
> `wget -O /tmp/install-nightly.sh https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/jkbms_ble/etc/dbus-serialbattery/install-nightly.sh && bash /tmp/install-nightly.sh`
>
> and changed the lltjbd.py

flyer9999, 2023-04-19:
Good Evening Manuel,
Beginning of another attempt :-)

Ohh Haa, in under 1 Minute everything is done and the Result is: It works !!!
[image]

[image]

[image]

[image]

However, what is still noticeable is that the log file is being shot up ...
Logfile: dbus-serialbattery.ttyACM0
@40000000644066ec33dd88d4 *** CCGX booted (0) ***
@400000006440670520036194 INFO:SerialBattery:Starting dbus-serialbattery
@4000000064406705200380d4 INFO:SerialBattery:dbus-serialbattery v1.0.0-jkbms_ble (20230419)
@400000006440670520039074 INFO:SerialBattery:Testing LltJbd
@4000000064406705271cbbc4 INFO:SerialBattery:Connection established to LltJbd
@40000000644067052728d954 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
@40000000644067052730c4ac INFO:SerialBattery:========== Settings ==========
@4000000064406705273a0bac INFO:SerialBattery:> Connection voltage: NoneV | Current: NoneA | SoC: None%
@4000000064406705274189a4 INFO:SerialBattery:> Cell count: None | Cells populated: 0
@400000006440670527b7c72c INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
@400000006440670527b7e284 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0V | MAX BATTERY DISCHARGE CURRENT: {utils.MAX_BATTERY_DISCHARGE_CURRENT}V
@400000006440670527b7fddc INFO:SerialBattery:> CVCM:     False
@400000006440670527b805ac INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@400000006440670527b81934 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@400000006440670527b824ec INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@400000006440670527b9af74 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@40000000644067052e9c2eac INFO:SerialBattery:DeviceInstance = 1
@40000000644067052ea54e9c INFO:SerialBattery:com.victronenergy.battery.ttyACM0
@400000006440670535ea21b4 ERROR:SerialBattery:type: <class 'int'>
@40000000644067053901122c INFO:SerialBattery:publish config values = 1
@40000000644067070712c88c ERROR:SerialBattery:type: <class 'int'>
@400000006440670808f55f84 ERROR:SerialBattery:type: <class 'int'>
@40000000644067090ab24ddc ERROR:SerialBattery:type: <class 'int'>
@400000006440670a0af93e04 ERROR:SerialBattery:type: <class 'int'>
@400000006440670b09cd4cdc ERROR:SerialBattery:type: <class 'int'>
@400000006440670c0b6eac84 ERROR:SerialBattery:type: <class 'int'>
@400000006440670d09e7742c ERROR:SerialBattery:type: <class 'int'>
.
.
.
.
.

mr-manuel, 2023-04-20:
> However, what is still noticeable is that the log file is being shot up ...

Yes since it publishes some debug output because I was not sure, what type the variable was. Could you warm up the external temperature sensors to make sure that the assignment is correct?

flyer9999, 2023-04-20:
I just tried it, it works!
Temp-1 correct, Temp-2 correct and Temp-3 correct..

mr-manuel, 2023-04-20:
Thanks. The fix was merged with the `jkbms_ble` branch. The PR for the `master` branch will be merged mid May.

closed 2023-04-20 by mr-manuel

AlexdeKairos, 2024-01-20:
> Yes, perfect, that is what i mean. The "external" Sensors you can plug in and with these you go to the cells, these are Temp2 and Temp3. The Temp1 is the MOS/internal Temperature.

Hi Guys, 
Sorry, that's my first post here.
I have a pair of BLE JBD connected batteries. BasenGreen ones. In my particular setup with 1.0.20231117dev. My first and only temperature sensor is connected to battery cells and not MOSFET. So with Basen Green Batteries, this mod does not work properly. 
How can I set the temperature sensor n1 connected to Cells and not to MOS?

Many thanks
Alex

Louisvdw, 2024-01-20:
@AlexdeKairos at the moment the first temp is always used for the MOS and there is no setting to adjust this. But it should be possible to adjust the code to cater for a scenario like this.