Louisvdw/dbus-serialbattery #352: isolated 4 way usb port and Ve direct leads as show on off grid garage try this to day  and same problem shown high battery temp- low temp - and so on driver for bms that`s not plug in shown device list

MS123uk, 2022-12-15:
isolated 4 way usb port and Ve direct leads as show on off grid garage try this to day  and same problem shown high battery temp- low temp - and so on driver for bms that`s not plug in shown device list

_Originally posted by @MS123uk in https://github.com/Louisvdw/dbus-serialbattery/issues/296#issuecomment-1353247020_

Louisvdw, 2022-12-15:
Can you clarify please.
Are you saying there is a problem with the temperature?

MS123uk, 2022-12-15:
No if i use other type comm leads between rarsp 4  and smart shunt - solar controller and inverter all vitron alarm show  high battery temp- low temp - and so on and driver for bms that`s not plug in shown device list . Victron Ve direct usb leads are the only ones will work  but isolated 4 way usb port and Ve direct leads as show on off grid garage work fine on house solar system  Victron and rarsp  to  

only way  clear  the alarms is reboot

MS123uk, 2022-12-17:
[image]

coca7, 2022-12-17:
I have seen similar alarms to MS123uk.  I also use a USB hub.  I mostly ignored it because I wasn't sure how to further isolate the problem and I knew the values were false.

MS123uk, 2022-12-17:
And serial battery ( Sino wealth ) bms driver  will appear in your devices list in remote Console

Louisvdw, 2022-12-29:
If you update to the latest v0.14.* then the Sinowealth driver has been disabled by default. This should prevent it from being picked up and giving false values.

closed 2022-12-30 by MS123uk