Louisvdw/dbus-serialbattery #359: JBD BMS: Balancing indicator not showing on correct cells

iLeeeZi, 2022-12-16:
Balancing indicators on Cell Voltages page are not showing on correct cells. The red indicators jump between cells showing 1-3 cells balancing and after a while it shows all cells balancing:
[image]

Checked with dbus-spy and it shows the same:
[image]

Switched to bluetooth adapter and xiaoxiang app is showing only two cells balancing as it should:
[image]

I have a JBD-SP04S034 4S 200A BMS with three temperature sensors, so there might be an issue with unusual message length. I need to check if disabling one temperature sensor in xiaoxiang app makes any difference.

mr-manuel, 2023-04-24:
Have others with a JBD BMS the same issue?

mr-manuel, 2023-05-04:
@iLeeeZi can you organize the protocol documentation so we can fix this? Please ask your seller or the manufacturer.

iLeeeZi, 2023-05-04:
Did some testing and looks like that it is not a protocol issue.

I added logging after L123 in lltjbd.py
https://github.com/Louisvdw/dbus-serialbattery/blob/jkbms_ble/etc/dbus-serialbattery/bms/lltjbd.py#L123
```
logger.info(str(tmp))
``` 
And the data seems correct, but with data showing only two of the cells balancing, the UI indicated after a few seconds that three of the cells are constantly balancing
```
2023-05-04 15:55:04.967329500 INFO:SerialBattery:0001
2023-05-04 15:55:05.974664500 INFO:SerialBattery:0001
2023-05-04 15:55:06.965833500 INFO:SerialBattery:0001
2023-05-04 15:55:07.972370500 INFO:SerialBattery:0101
2023-05-04 15:55:08.978049500 INFO:SerialBattery:0001
2023-05-04 15:55:09.969268500 INFO:SerialBattery:0001
2023-05-04 15:55:10.976891500 INFO:SerialBattery:0101
2023-05-04 15:55:11.982067500 INFO:SerialBattery:0000
2023-05-04 15:55:12.973059500 INFO:SerialBattery:0001
2023-05-04 15:55:13.979577500 INFO:SerialBattery:0101
2023-05-04 15:55:14.986797500 INFO:SerialBattery:0001
2023-05-04 15:55:16.028883500 INFO:SerialBattery:0001
2023-05-04 15:55:17.018241500 INFO:SerialBattery:0101
2023-05-04 15:55:18.024716500 INFO:SerialBattery:0001
2023-05-04 15:55:19.015430500 INFO:SerialBattery:0001
2023-05-04 15:55:20.021196500 INFO:SerialBattery:0001
2023-05-04 15:55:21.012090500 INFO:SerialBattery:0001
2023-05-04 15:55:22.017786500 INFO:SerialBattery:0001
2023-05-04 15:55:23.025228500 INFO:SerialBattery:0101
```

mr-manuel, 2023-05-09:
So the log data is correct, but the displayed data not?

iLeeeZi, 2023-05-09:
Yes. The bms seems to send correct data, and the driver reads it correctly but dbus and gui is incorrect.

mr-manuel, 2023-05-09:
Could you write me on Discord?

mr-manuel, 2023-05-09:
Could you try to replace

https://github.com/Louisvdw/dbus-serialbattery/blob/88e5eea93331ae99dc28fcf8b5a40cb866e432e2/etc/dbus-serialbattery/bms/lltjbd.py#L118-L130

with

```python
    def to_cell_bits(self, byte_data, byte_data_high):
        try:
            # get up to the first 16 cells
            tmp = bin(byte_data)[2:].rjust(min(self.cell_count, 16), utils.zero_char)
            logger.info("tmp: type: " + type(tmp) + " - value: " + tmp)
            # tmp = 0101
            tmp_reversed = reversed(tmp)
            # tmp_reversed = 1010
            logger.info("tmp_reversed: type: " + type(tmp_reversed) + " - value: " + tmp_reversed)

            test = tmp_reversed + reversed(tmp)
            logger.info("test: type: " + type(test) + " - value: " + test)

            if self.cell_count > 16:
                tmp2 = bin(byte_data_high)[2:].rjust(self.cell_count - 16, utils.zero_char)
                logger.info("tmp2: type: " + type(tmp2) + " - value: " + tmp2)
                tmp_reversed = tmp_reversed + reversed(tmp2)
                logger.info("tmp_reversed: type: " + type(tmp_reversed) + " - value: " + tmp_reversed)
            # tmp = 1010
            for c in range(self.cell_count):
                if is_bit_set(tmp_reversed[c]):
                    logger.info("balancing cell " + c)
                    self.cells[c].balance = True
                else:
                    self.cells[c].balance = False

        except Exception as err:
            logger.error(f"Unexpected {err=}, {type(err)=}")
```

?

It could be that nothing works, since I'm not sure about the var types and I have no JBD BMS to test. But the driver shouldn't crash.

iLeeeZi, 2023-05-10:
I got errors on reversed() so changed it to [::-1]:
```python
    def to_cell_bits(self, byte_data, byte_data_high):
        try:
            # get up to the first 16 cells
            tmp = bin(byte_data)[2:].rjust(min(self.cell_count, 16), utils.zero_char)
            logger.info("tmp: type: " + str(type(tmp)) + " - value: " + tmp)
            tmp_reversed = tmp[::-1]
            logger.info("tmp_reversed: type: " + str(type(tmp_reversed)) + " - value: " + tmp_reversed)

            test = tmp_reversed + tmp[::-1]
            logger.info("test: type: " + str(type(test)) + " - value: " + str(test))

            if self.cell_count > 16:
                tmp2 = bin(byte_data_high)[2:].rjust(self.cell_count - 16, utils.zero_char)
                logger.info("tmp2: type: " + str(type(tmp2)) + " - value: " + tmp2)
                tmp_reversed = tmp_reversed + tmp2[::-1]
                logger.info("tmp_reversed: type: " + str(type(tmp_reversed)) + " - value: " + tmp_reversed)

            for c in range(self.cell_count):
                if is_bit_set(tmp_reversed[c]):
                    logger.info("balancing cell " + str(c+1))
                    #self.cells.append(Cell(True))
                    self.cells[c].balance = True
                else:
                    #self.cells.append(Cell(False))
                    self.cells[c].balance = False

        except Exception as err:
            logger.error(f"Unexpected {err=}, {type(err)=}")
``` 
But without self.cells.append(Cell()) I get 
```
2023-05-10 15:53:14.552053500 ERROR:SerialBattery:Unexpected err=IndexError('list index out of range'), type(err)=<class 'IndexError'>
``` 
In some other function

mr-manuel, 2023-05-10:
Can you post the error logs or give me access to the system? Second would save me a lot of time writing back and forth. Please contact me on Discord for this. You find the link in my profile.

ngr58, 2023-06-16:
Thank you @mr-manuel for fixing this. I have a JBD-AP20S003S BMS with 18 cells. I suspected that the red balancing indicators in SerialBattery were wrong ever since cell voltages became available but never had a chance to compare them to the xiaoxiang app. Saw this note in the changelog, upgraded to v1.0.20230531 and now they look much more normal! Thank you.