Louisvdw/dbus-serialbattery #361: Current venus-data.tar.gz file is missing revov.py

nc1066, 2022-12-19:
**Describe the bug**
Current venus-data.tar.gz is missing revov.py

**To Reproduce**
1. Do a clean install of dbus-serialbattery using installrelease.sh
2. Check log file for driver (in my case: /var/log/dbus-serialbattery.ttyUSB0/current), 
3. Log file contains error messages: failing to import revov package, 
4. No revov.py in install directory

**Expected behavior**
tar file contains all needed files

**Screenshots**
NA

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi 3 Model B v1.2
 - Firmware Version 2.92

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: EG4 Lifepower
 - Cells: 8
 - Interface: USB-RS485

**Additional context**
NA

soluga, 2022-12-20:
Workaround:
`cd /data/etc/dbus-serialbattery/
wget https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/revov.py
reboot`

Gir4ff33, 2022-12-20:
confirmed - soluga's workaround worked for me, same device and firmware version, different battery (JBD 16s)a

Greature8, 2022-12-22:
Yup, worked for me too. Many thanks to soluga for leading me to this workaround.

chaerle, 2022-12-28:
See pull request https://github.com/Louisvdw/dbus-serialbattery/pull/369 for fix.

Louisvdw, 2022-12-29:
Fixed in new release v0.14.3

closed 2022-12-29 by Louisvdw