Louisvdw/dbus-serialbattery #362: dvcc: Add support for BMS Can with SOK SK48V100 battery.

dexter12353, 2022-12-20:
Hello,

I am requesting for the SOK 48v 100ah using a PACE BMS (Pylontech-CAN protocol) to have support added.

We already have a pull request with the solution to make this work - it takes 2 lines of code to make this work.  See pull request here: https://github.com/victronenergy/dbus-systemcalc-py/pull/9

We are already manually implementing the fix manually, but if this fix were added, it would make many battery owners very happy!

Louisvdw, 2022-12-28:
@dexter12353  I see this is now included in the VenusOS firmware. It is a better place to add it. 
Should we close this request, or is there a reason to try and add this to the serialbattery driver in future?

dexter12353, 2022-12-28:
You can close the request.  We posted here because it was posted elsewhere but appeared to be unseen for months. 

Thank you for all your hard work!

closed 2022-12-29 by Louisvdw