Louisvdw/dbus-serialbattery #364: Revov.py not installed by installscript via SSH on a Cerbo GX

pmcgn, 2022-12-21:
**Describe the bug**
Revov.py is not downloaded/installed by the installation script via SSH.  (Tried to run the installer a second time - no change)

I see the following error in /data/log/dbus-serialbattery.ttyUSB0/current :

Traceback (most recent call last):
  File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 25, in <module>
    from revov import Revov
ModuleNotFoundError: No module named 'revov'

After downloading it manually to /opt/victronenergy/dbus-serialbattery/, the issue is resolved and I see a cuccess message in the log and the BMS appears in the UI.

I noticed that the last commit (7b239007bd87bc3fdfd6a3a3726937f1210997f2) removed the file Revov.py from another file. Didn't look more into the details - sorry.

**To Reproduce**
Steps to reproduce the behavior:
1. Run installation routine as described in Wiki on a Cerbo GX via SSH. 
2. Check if File /opt/victronenergy/dbus-serialbattery/Revov.py exists
3. Check log /data/log/dbus-serialbattery.ttyUSB0/current (should indicate that the file is missing)

**Expected behavior**
All dependencies (including Revov.py) should be available and the python script should not throw any errors.

**Screenshots**
[image]

**VenusOS (please complete the following information):**
 - Device type: Cerbo GX
 - Firmware Version: 2.91

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JKBMS

 - Cells: 16
 - Interface: USB-RS485

**Additional context**
Thanks for doing all the hard work!

NickDolfin, 2022-12-23:
Same here.
I get the same error.
device raspberry pi 3b 
firmWARE 2.92
using ftdi rs485 cable
it is attacjed to usb3

chaerle, 2022-12-28:
See Pull request https://github.com/Louisvdw/dbus-serialbattery/pull/369 for fix.

Louisvdw, 2022-12-29:
Fixed in new release v0.14.3

closed 2022-12-29 by Louisvdw