Louisvdw/dbus-serialbattery #365: Charge Discharge Limits sticks at 60A after change

wetty, 2022-12-22:
**Describe the bug**
I changed both values for Carge/Discharge Current several times in the source  /data/etc/dbus-serialbattery/utils.py
Reboot the system after each change, but the values display in Venus OS are always 60A for both. So nothing changed.
I want to change the values as my Victron MultiPlus II 5000 has 70A Charge Current

**To Reproduce**
Steps to reproduce the behavior:
1. Edit /data/etc/dbus-serialbattery/utils.py and change MAX_BATTERY_CHARGE_CURRENT = 70 and MAX_BATTERY_DISCHARGE_CURRENT = 90
2. Save the changes
3. Reboot the system
4. In Venus OS got to SerialBattery(JKBMS) then Parameter. Charge Current Limit is 60A and Discharge Current Limit is 60A 

**Expected behavior**
The SerialBattery(JKBMS) Parameter should be the same as set in the utils.py. 
=> Charge Current Limit is 70A 
=> Discharge Current Limit is 90A 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi
 - Firmware Version dbus-serialbattery 0.142 (with separate download of revov.py)
 - Venus OS Version 2.92

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JKBMS
 - Cells: 16
 - Interface: USB-RS485

**Additional context**
[image]
[image]

corneel27, 2022-12-22:
You have CCCM_T_ENABLE and DCCM _T_ENABLE = True and there you have 60 A as the max .
Or set them to False of you change that 60 to an higher value.

wetty, 2022-12-22:
As mention I have already fixed ist by setting the values mention in the forum to false.