Louisvdw/dbus-serialbattery #366: VenusOS no web interface if # INVERT_CURRENT_MEAUREMENT = 1

e-i-k-e, 2022-12-22:
Hello everyone,

I just updated to the current 0.14.2
VenusOS 3.00~8
DalyBMS
Raspberry Pi 3

If I set "INVERT_CURRENT_MEASUREMENT = -1" back to = 1, I don't get any graphical interface from VenusOS. As soon as I go back to =-1, everything is okay again.

huskyte, 2022-12-25:
I just add myself as I think this the solution to my issue. My measured charge to the battery adds to DC loads. Shouldn't it be the athoer way around? Couldn't find anything in the FAQ. DalyBMS 4S.

Louisvdw, 2022-12-29:
@e-i-k-e can you post the serialbattery log file for when it fails with the -1 value.
https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot#check-the-log-files-on-your-gx-device

mr-manuel, 2023-04-14:
Is fixed with the next release.

closed 2023-04-14 by mr-manuel