Louisvdw/dbus-serialbattery #368: Victron beta v3.00–14

DirkReudenbach, 2022-12-24:
Victron released yesterday a new beta version which supports much more 3rd party batteries.
Updating to this version showed much more details in the JK BMS fields in Victron but all solar chargers showed "Error 67"

I'm sure you are working already on a further development on this new release 

Regards

Dirk

testpaul999, 2023-01-02:
@DirkReudenbach 
Is this issue persistent or did you find a solution?
I‘ll update to 3.00-14 but at the moment I wait.
Regards

DirkReudenbach, 2023-01-02:
on my site v3.00-14 doesn't work and I could not find a solution on my site

testpaul999, 2023-01-03:
Have done the update to 3.00-15
There are many new switches like switch „on/standby“ if I activated all it runs fine.

e-i-k-e, 2023-01-15:
So do I understand this correctly, if the switch is activated on/standby, does the driver work as usual?
Can anybody confirm this? (which version?)

testpaul999, 2023-01-17:
@e-i-k-e 
Sure have run the 3.00~14 and also the ~15 with the last public version of serialbattery.

mr-manuel, 2023-03-09:
@e-i-k-e does the issue still persist?

e-i-k-e, 2023-03-09:
> @e-i-k-e does the issue still persist?

No, it Work.

mr-manuel, 2023-03-09:
@DirkReudenbach is it's also working for you please close this issue

closed 2023-04-14 by mr-manuel