Louisvdw/dbus-serialbattery #371: DCL is "jumping" - should be 0 ?! - JKBMS

MisterX1000, 2022-12-28:
**Describe the bug**
In the utils.py (file is attatched) I set the DCCM current settings to "0A" when cell voltage reached 2.90V. 
The "SOC limitation" is disabled (False)

Now I have the problem that when this limit is reached and one cell is <= 2.90V, the discharge current starts jumping. And instead of switching the discharge off, the MP2-GX begins with charging the battery, then goes to "idle", then to "discharge" and so on.

While this happens there are many, many "Low battery: Alarm" messages from the VE.Bus System 275.
All the MP2-GX settings which could cause this are below this voltage levels and should not trigger any alarm. 
The DVVC is switched ON.

Maybe I did somenthing wrong in the serialbattery config file or I missunderstand something.

**Expected behavior**
I would expect that the current goes down to 0A when the voltage limit of <= 2.9V is reached. I don´t expect that the battery is reloaded with AC-grid power in the middle of the night when there is no sun.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**VenusOS (please complete the following information):**
 - Device type: MP2-GX
 - Firmware Version: 3.00 ~14, (v498)
 - Serialbattery version 0.14.2

**Battery/BMS**
 - BMS/Battery type: JKBMS / CALB LiFePo4 cells
 - Cells: 16
 - Interface: RS485 -->USB (MP2-GX)

**Additional context**

**MP2 & ESS settings; VRM graphs:** [all settings.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/10313886/all.settings.pdf)
utils.py: [utils.txt](https://github.com/Louisvdw/dbus-serialbattery/files/10313795/utils.txt)

Alarm log: [image]

VRM DCL graph: [image]

[image]
[image]
[image]

Louisvdw, 2022-12-29:
We'll have to improve the algorithm by adding a release voltage as well. 
What is happening is that as soon as no more current is drawn the cell values starts to recover  and the BMS release 
(the charger should only kick in when you are below the Min Soc value you have set in the ESS settings)

MisterX1000, 2022-12-30:
Hey Louis,
thank you very much for your reply and for your great work here.

Maybe I can set the "sustain voltage" in ESS a little bit higher than the smaltest "DCCM_CV" cell voltage x16. Maybe this prevents the cells from discharging before the 0A voltage is reached and activates the sustain voltage like set in ESS.

Can you tell me if all the voltages which I set in the ESS (with VE Configure) are still counting, when your driver is installed?
I wonder which of the ESS settings have a higher prioroty then the settings in the utils.py?!

Thanks again for your help an great work. I wish you a happy new year.
Best regards
Alex

Louisvdw, 2022-12-30:
The Limit values you set in DVCC always override everything. 
If those are not set then the CVL, CCL, DCL that you see under the Parameters section of your battery in the GX is derived from the settings in utils.py and will apply.

MisterX1000, 2022-12-30:
Hey Louis,

I will try this and increase the "sustain voltage" - but this is not a setting in the DVCC, this is a "normal" setting in the VE configure. I attach a screenshot.

Best regards
Alex

[image]

MisterX1000, 2023-01-02:
Hey Louis,
unfortunately it didn't work to reduce the "Sustain-Voltage" value to 48V in the ESS assistant of the MP2.

[image]

In the DVCC section I cant set anything which could help me here. There is no setting for a minimal voltage, only in the ESS assistant.

[image]
[image]

The battery voltage is still dropping to the value which is set in the utils.py (2.9V) and then it begins to jump between charge-discharge-idle and so on.

I there something more I could do? Is this am JKBMS problem (or MP2-GX) only?

Thank you very much for your help!
Alex

[image]

MisterX1000, 2023-01-09:
> Hey Louis,
> 
> I will try this and increase the "sustain voltage" - but this is not a setting in the DVCC, this is a "normal" setting in the VE configure. I attach a screenshot.
> 
> Best regards Alex
> 
> [image]

Just for information. 

What helped me here was reducing the Cut-off voltages in the ESS asisstant (there are 4 voltages for different currents) - when the battery voltage dropped to this cut-off voltage, the discharging will stop and the "sustain voltage" will jump in until the battery voltage rises up over the "above-cut-off" value.

Important!!! - When the ESS asisstant is activated, all the 3 low-voltage voltage settings under the "Inverter" tab are completely ignored by the MP2.

Best regards
Alex

mr-manuel, 2023-04-20:
@MisterX1000 do I understand correctly, that your issue was solved? If yes, then I will add this to the FAQ.

MisterX1000, 2023-04-21:
> @MisterX1000 do I understand correctly, that your issue was solved? If yes, then I will add this to the FAQ.

Hey mr-manuel,
for me the issue is solved. I think there are 2 ways to prevent this jumipng:
1. Current setting >0 (but I did not test this)
2. Reducing the ESS "Cut-off voltage" like I mentioned above

Thank you for your great support here!

closed 2023-04-21 by MisterX1000