Louisvdw/dbus-serialbattery #373: white screen after update from 0.12 to 0.14

gerd-git, 2022-12-28:
**Describe the bug**
Just a white screen after update from 0.12 to 0.14.
You can still see the hotkeys in the web interface, but only white in the display area and on the display. No menu or other symbols.

**To Reproduce**
Run v0.12 then update to 0.14:
```
root@raspberrypi2:~# wget https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/installrelease.sh
--2022-12-28 17:18:45--  https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/installrelease.sh
Resolving raw.githubusercontent.com... 2606:50c0:8003::154, 2606:50c0:8001::154, 2606:50c0:8000::154, ...
Connecting to raw.githubusercontent.com|2606:50c0:8003::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 277 [text/plain]
Saving to: ‘installrelease.sh’

installrelease.sh                                           100%[========================================================================================================================================>]     277  --.-KB/s    in 0s

2022-12-28 17:18:46 (1.28 MB/s) - ‘installrelease.sh’ saved [277/277]

root@raspberrypi2:~# sh installrelease.sh
root@raspberrypi2:~# reboot
```

**Screenshots**
[image]

**VenusOS:**
 - Device type: Raspberry Pi2
 - Firmware Version
```
 root@raspberrypi2:~# cat /etc/version
20220715122828
```

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JKBMS
 - Cells: 8
 - Interface: USB-RS485

**Additional context**
Please help!!!!

Louisvdw, 2022-12-29:
Hi @gerd-git 
Normally this will happen if you were on an older firmware for your GX.
You can remove the GUI changes or update your GX firmware to solve this.

Remove GUI changes
> cp -f /opt/victronenergy/gui/qml/PageBattery.qml.backup /opt/victronenergy/gui/qml/PageBattery.qml
reboot

Update firmware run
> /opt/victronenergy/swupdate-scripts/check-updates.sh

kr0815, 2022-12-29:
sadly, nothing helps

copying the backup PagenBattery - still the white screen 

I can´t do an update either - my version is to new...

`
installed: 20220715122828 v2.89
available: 20220715122828 v2.89
`

any other ideas how to fix the blank screen?

Louisvdw, 2022-12-29:
You should be on v2.92 for the driver which is newer than 2.89, so there should be an update available. 
What GX device are you using?

kr0815, 2022-12-29:
Venus is running on a Raspi

I´m not that much in Details about Venus, nut as far as i read, update could only be done with a USN Stick because the actual Version is to new?

I have fear i´ll lose all my settings then?

Louisvdw, 2022-12-29:
Ah yes. You will manually need to update from that older version on the Raspberry Pi.

Run these commands for the GUI
> cp -f /opt/victronenergy/gui/qml/PageBattery.qml.backup /opt/victronenergy/gui/qml/PageBattery.qml
svc -d /service/gui
svc -u /service/gui

kr0815, 2022-12-29:
also after these 3 commands - still the white Page
I can´t even do an Update by USB-Stick, as i can´t get to the gui-point to accept it

Any other ideas?  maybe download PageBattery.qml from somewhere?
Mine has 8832 bytes

kr0815, 2022-12-29:
so i reinstalled the complete system to 2-92

White Screen is gone 

But next problem, max current settings in utils.py have no effect at all

Everything really strange...

Louisvdw, 2022-12-30:
reboot you GX device a second time if you feel your settings does not apply. 
But when you say the settings has no effect, can you give more details of what you changed, what you see and what you expect. 
Also ready through the CCCM feature description to see if that helps with what you are expecting to see https://github.com/Louisvdw/dbus-serialbattery/wiki/Features#charge-current-control-management

kr0815, 2022-12-30:
in utils.py i have 

`# battery Current limits
MAX_BATTERY_CHARGE_CURRENT = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 70.0
`
Battery is at 92 SOC at the moment

min / max cell voltage 3,30 / 3,31 Volt

in serialbattery - Parameters  i see
Charge  Limit/ CCL 35A
Discharge Limit / DCL 60A

I´d expect 35A / 70A?

But i checked again your code now, think i also have to change the other lines like
`MAX_DISCHARGE_CURRENT_CV             = [   0,    5,  30,  60]`
to
`MAX_DISCHARGE_CURRENT_CV             = [   0,    5,  35,  70]`

?

Louisvdw, 2022-12-30:
yes you need to change the CV (cell voltage) settings as well as the temp values as well. The lowest value will apply. That set up is a bit confusing and I am planning to make it easier to set.

kr0815, 2022-12-30:
thanks a lot, i assumed all other settings are calculated from this one line with min/max charge current

Maybe you should change the documentation a little bit - under "Installation" only this one line is explained, so i thought that´s all i have to change
I also have to confess, i didn´t expect to find the detailed explanation under "Features"

IamSierraCharlie, 2022-12-31:
Hello, Is there a way to recover from the white screen without having to reinstall?  I have followed the steps in this thread and have had no success. I note that in this thread, a reimage was completed.  I am not in a position to do that (at least not immediately) although I still have remote access to the Raspberry Pi.

IamSierraCharlie, 2022-12-31:
**Please read all of this post before running any of the commands mentioned here**

`/opt/victronenergy/swupdate-scripts/check-updates.sh` doesn't seem to work because the expected/default version on the website for a release candidate appears incorrect.  The script expects a link => https://updates.victronenergy.com/feeds/venus/candidate/images/raspberrypi2/venus-swu-2-raspberrypi2.swu 
which is not available.  
you can specifiy the version you wish to update to by running the command (for example)
`/opt/victronenergy/swupdate-scripts/check-updates.sh -swu https://updates.victronenergy.com/feeds/venus/candidate/images/raspberrypi2/venus-swu-3-raspberrypi2.swu'

The link above is correct and not broken.

Whilst this addresses the aforementioned issue, it is NOT recommended - at the time of writing, the command ran and it also updated my software but never recovered.  I am 4 hours away from my install so this will have to wait until next year :-(

Also note that in order to select a candidate update (rather than release), you may need to run the command:
`/opt/victronenergy/swupdate-scripts/set-feed.sh candidate`

I guess I lack a little bit of understanding of how these scripts are meant to function, but what appears to be clear is that there appears to be a version mismatch.  I'll look into this a bit more when I have a moment.

IamSierraCharlie, 2023-01-01:
Just an update to the above - I have built a new pi this morning and installed version V2.92 and when I run the check-updates.sh script, it actually calls for the correct version now.  (https://updates.victronenergy.com/feeds/venus/candidate/images/raspberrypi2/venus-swu-3-raspberrypi2.swu)
This may suggest that perhaps on an older version of the Venus OS there was a bug looking for the incorrect version to update to - hard to know for sure, but I'm seeing different results now.

kr0815, 2023-01-01:
the Problem is that Victron doesn´t allow an update from 2.7 to 2.8 or from 2.8 to 2.9
You need to reinstall it manually

mr-manuel, 2023-02-17:
Recently I had the same problem. If you have a remote installation far away and you are already on the latest version you can reinstall Venus OS with `/opt/victronenergy/swupdate-scripts/check-updates.sh -update -force`. Be aware that you may loose some settings, but at least you don't have to be on premise.

closed 2023-04-14 by mr-manuel

Cyberb29, 2023-05-07:
Good day, I have finally found a fix to this problem without having to update your OS whatsoever. 

You require SSH access to do this, if you dont SSH access please perform the Blind Uninstall as instructed by @kwindrem on the main SetupHelper page. (Please note after this procedure your screen will go white again, however it will give you a short period to enable SSH on LAN and set a root password.) 

After its done and you have access please type the following commands as instructed by @Louisvdw 

cp -f /opt/victronenergy/gui/qml/PageBattery.qml.backup /opt/victronenergy/gui/qml/PageBattery.qml
svc -d /service/gui
svc -u /service/gui

Do not reboot yet as your not finished, for those who have tried the above, you would have noticed that it works for a short while then screen goes blank again. This is because in the /data/rc.local file there is a command to reinstall the dbus-serial driver after a reboot, this is to prevent the user from having to redo the setup if a firmware update was done. 

You can remove this command by using nano /data/rc.local and then removing the reinstall command. Use CTRL + O to save and CTRL + X to backout, after this command is removed then perform a reboot. After the reboot you will have access to your GUI. 

Kind regards