Louisvdw/dbus-serialbattery #374: ModuleNotFoundError: No module named 'revov'

gerd-git, 2022-12-28:
**Describe the bug**
after a completely new venus installation:
no jkbms

Log:
```
ModuleNotFoundError: No module named 'revov'
root@raspberrypi2:~# tail -f /data/log/dbus-serialbattery.ttyUSB3/current
@4000000063ac9d9d04dafd14     from revov import Revov
@4000000063ac9d9d04db04e4 ModuleNotFoundError: No module named 'revov'
@4000000063ac9dc20a0456b4 Traceback (most recent call last):
@4000000063ac9dc20a04720c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 25, in <module>
@4000000063ac9dc20a04897c     from revov import Revov
@4000000063ac9dc20a04914c ModuleNotFoundError: No module named 'revov'
@4000000063ac9de718e7c92c Traceback (most recent call last):
@4000000063ac9de718e7dcb4   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 25, in <module>
@4000000063ac9de718e7f424     from revov import Revov
@4000000063ac9de718e7fbf4 ModuleNotFoundError: No module named 'revov'
@4000000063ac9e0c1649dd2c Traceback (most recent call last):
@4000000063ac9e0c1649f49c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 25, in <module>
@4000000063ac9e0c164a0c0c     from revov import Revov
@4000000063ac9e0c164a13dc ModuleNotFoundError: No module named 'revov'
@4000000063ac9e311c030f04 Traceback (most recent call last):
@4000000063ac9e311c032674   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 25, in <module>
@4000000063ac9e311c033de4     from revov import Revov
@4000000063ac9e311c0345b4 ModuleNotFoundError: No module named 'revov'
```

***Reproduce:***
Install venus os,
wget https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/installrelease.sh
sh installrelease.sh
reboot

**VenusOS :**
 - Device type: Raspberry Pi 2
 - Firmware Version v2.92

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: jkbms
 - Cells: 8
 - Interface: USB-RS485

mr-manuel, 2022-12-28:
There are already opened some issues regarding this. There is also already a pull request https://github.com/Louisvdw/dbus-serialbattery/pull/369 for fix.

gerd-git, 2022-12-28:
Thank you. I comment it out. Now it works for me.
```
root@raspberrypi2:~# nano /data/etc/dbus-serialbattery/dbus-serialbattery.py
(...)
#from revov import Revov
(...)
```

closed 2022-12-28 by gerd-git