Louisvdw/dbus-serialbattery #377: Cerbo GX JK BMS - ModuleNotFoundError: No module named 'spidev'

leinadsims, 2022-12-29:
I am fighting with the installtion for hours now, after solving this REVOV issue with help of: "https://github.com/Louisvdw/dbus-serialbattery/issues/364" I now have a other issue for what I can´t find any solution. Starting "tail -f /data/log/dbus-serialbattery.ttyUSB1/current" I receive an error and have no idea how to solve this. The JK BMS is connected to the Battery and the Cerbo and semm to be connected the right way.
This is the error hope somone has an idea how to solve:
@4000000063ad4bce3a76d9ac     import spidev
@4000000063ad4bce3a76e564 ModuleNotFoundError: No module named 'spidev'
@4000000063ad4bf419475dec Traceback (most recent call last):
@4000000063ad4bf419478114   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 28, in <module>
@4000000063ad4bf4194790b4     from mnb import MNB
@4000000063ad4bf419479884   File "/opt/victronenergy/dbus-serialbattery/mnb.py", line 5, in <module>
@4000000063ad4bf41947a43c     from util_max17853 import * #{exclusive. use test for testing}
@4000000063ad4bf41947aff4   File "/opt/victronenergy/dbus-serialbattery/util_max17853.py", line 1, in <module>
@4000000063ad4bf4194a57a4     import spidev
@4000000063ad4bf4194a6744 ModuleNotFoundError: No module named 'spidev'
@4000000063ad4c192dcf8fdc Traceback (most recent call last):
@4000000063ad4c192dcfb6ec   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 28, in <module>
@4000000063ad4c192dcfc68c     from mnb import MNB
@4000000063ad4c192dcfca74   File "/opt/victronenergy/dbus-serialbattery/mnb.py", line 5, in <module>
@4000000063ad4c192dcfd62c     from util_max17853 import * #{exclusive. use test for testing}
@4000000063ad4c192dcfe1e4   File "/opt/victronenergy/dbus-serialbattery/util_max17853.py", line 1, in <module>
@4000000063ad4c192dd27ddc     import spidev
@4000000063ad4c192dd285ac ModuleNotFoundError: No module named 'spidev'
@4000000063ad4c3f07023e2c Traceback (most recent call last):
@4000000063ad4c3f0702653c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 28, in <module>
@4000000063ad4c3f070274dc     from mnb import MNB
@4000000063ad4c3f070278c4   File "/opt/victronenergy/dbus-serialbattery/mnb.py", line 5, in <module>
@4000000063ad4c3f0702847c     from util_max17853 import * #{exclusive. use test for testing}
@4000000063ad4c3f07029034   File "/opt/victronenergy/dbus-serialbattery/util_max17853.py", line 1, in <module>
@4000000063ad4c3f0705439c     import spidev
@4000000063ad4c3f07054f54 ModuleNotFoundError: No module named 'spidev'
@4000000063ad4c64204e418c Traceback (most recent call last):
@4000000063ad4c64204e64b4   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 28, in <module>
@4000000063ad4c64204e7454     from mnb import MNB
@4000000063ad4c64204e783c   File "/opt/victronenergy/dbus-serialbattery/mnb.py", line 5, in <module>
@4000000063ad4c64204e83f4     from util_max17853 import * #{exclusive. use test for testing}
@4000000063ad4c64204e8fac   File "/opt/victronenergy/dbus-serialbattery/util_max17853.py", line 1, in <module>
@4000000063ad4c6420513374     import spidev
@4000000063ad4c6420514314 ModuleNotFoundError: No module named 'spidev'

Louisvdw, 2022-12-29:
you must have made some other changes as well. 

The best would be to fix it using this command as the revov reference is already fixed in the code.

>wget https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/dbus-serialbattery.py

leinadsims, 2022-12-29:
Hi, thanks for the ultra fast feedback, will give it a try this evening

Louisvdw, 2022-12-29:
Just update to the new release v0.14.3

closed 2022-12-29 by Louisvdw